espressif/esp-zigbee-sdk #2: HA attribute reporting by trigger

u236, 2023-02-02:
Hello! I've successfully build HA_light_on_off example with my additions, such a led state control with button and it works good. Attribute reporting, configured via "configure reporting" global command, works too, but i can't understand, how to trigger device to report led state attribute when it's changed with button?

likunqiao097304, 2023-02-03:
@u236 Good to hear you are running successfully. According to the specs, if you set the min_interval to 0, it should report immediately when reportable value change. However, there is a issue that it couldn't report as soon as possible. We are working on that to fix it.

closed 2023-02-03 by espressif-bot

u236, 2023-02-03:
I now, how reporting works, but i don't understand how to tell zigbee stack about attrubute was changed