espressif/esp-zigbee-sdk #3: How to rigger ZCL attribute report send when attribute value changed internally?

u236, 2023-02-04:
Hello! Sorry for duplicate issue, but you've close [previous](https://github.com/espressif/esp-zigbee-sdk/issues/2) too fast :)

_I've successfully build `HA_light_on_off` example with some additions, such a led state control with button, and it works good. Attribute reporting, configured via "configure reporting" global command, works too, but i can't understand, how to trigger device to report led state attribute when it's changed with button?_

I need for device send report, when led state changed with button, timer or other reason, but NOT zigbee command. I know how reporting works, but i don't understand how to tell zigbee stack about attrubute value was changed, to trigger report.

Please help me to figure it out :)

jurriaan, 2023-02-04:
Also curious about this, there is an attribute update api, [esp_zb_cluster_update_attr](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/esp_zigbee_attribute.html#_CPPv426esp_zb_cluster_update_attrP23esp_zb_attribute_list_t8uint16_tPv), but it feels like it is to be used when setting up the clusters initially. Or should it also be used to update attributes on the fly?

u236, 2023-02-04:
> Also curious about this, there is an attribute update api, [...]

I've try this api, but nothing changed. As far as I understand, this api just updates pointer to attribute value location.

likunqiao097304, 2023-02-06:
> Hello! Sorry for duplicate issue, but you've close [previous](https://github.com/espressif/esp-zigbee-sdk/issues/2) too fast :)
>
> _I've successfully build `HA_light_on_off` example with some [...]

Sure, I am happy to work with you to figure it out. Correct me if I am wrong for your design: You have created a light-example zigbee local device and it could trigger by itself button, timer or other reason, so you'd like to set the attribute change in the stack and if you bind some other zigbee node with this attribute, it will report the changes to remotely.

For the update the certain attribute on the fly, we haven't provide API yet. We will provide it in the next round.

likunqiao097304, 2023-02-06:
> Also curious about this, there is an attribute update api, [...]

`esp_zb_cluster_update_attr` is used for update the cluster initially for before you the device create. You could see the example:[customized_server](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_server/main/esp_HA_customized_light.c)

u236, 2023-02-06:
> For the update the certain attribute on the fly, we haven't provide API yet. We will provide it in the next round.

Thank you, i'll wait for new features :)

likunqiao097304, 2023-02-13:
> > For the update the certain attribute on the fly, we haven't provide API yet. We will provide it in the next round.
> 
> Thank you, i'll wait for new features :)

The new API has provided in the latest release. `esp_zb_zcl_set_attribute_val` please take a look and close the issue if it works.

u236, 2023-02-13:
> The new API has provided in the latest release. `esp_zb_zcl_set_attribute_val` please take a look and close the issue if it works.

Now attribute reported, when i change led state via button and call `esp_zb_zcl_set_attribute_val`. Thanks a lot!

closed 2023-02-13 by u236