espressif/esp-zigbee-sdk #4: Color control cluster issues

u236, 2023-02-06:
I've build [HA_color_dimmable_light](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_color_dimmable_light) example and ran into few problems:

1. Firmware crashes, when coordinator sends commands to change light color via hue/saturation values (for example, cluster specific command "move to hue (0x00)"). I think it is because no value  pointers set for hue/saturation attributes. If i add this pointers manually, problem disappears.

2. If led color changed via hue/saturation commands, corresponding attributes values updated, but set attribute value callback not triggered, it triggered only if color changed via x/y commands.

Maybe i can contribute your zigbee stack? Please, let me know :)

likunqiao097304, 2023-02-07:
> 1. Firmware crashes, when coordinator sends commands to change led color via hue/saturation commands (for example, cluster specific command "move to hue (0x00)"). I think it is because no value  pointers set for hue/saturation attributes. If i add this pointers manually, problem disappears.

ESP_ZB_DEFAULT_COLOR_DIMMABLE_LIGHT_CONFIG() in the [HA_color_dimmable_light](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_color_dimmable_light/main/esp_zb_light.c) is register a color cluster with following attribute by default. 

'typedef struct esp_zb_color_cluster_cfg_s {
    uint16_t current_x;                         /*!<  The current value of the normalized chromaticity value x */
    uint16_t current_y;                         /*!<  The current value of the normalized chromaticity value y */
    uint8_t  color_mode;                        /*!<  The mode which attribute determines the color of the device */
    uint8_t  options;                           /*!<  The bitmap determines behavior of some cluster commands */
    uint8_t  enhanced_color_mode;               /*!<  The enhanced-mode which attribute determines the color of the device */
    uint16_t color_capabilities;                /*!<  Specifying the color capabilities of the device support the color control cluster */
} esp_zb_color_cluster_cfg_t;'

So you need to add other attributes manually like what you did (I assume you followed the example under [customizd_server)](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_server/main/esp_HA_customized_light.c) If you don't register corresponding attribute, it will cause issue.

> 2\. If led color changed via hue/saturation commands, corresponding attributes values updated, but set attribute value callback not triggered, it triggered only if color changed via x/y commands.

 the hue/saturation command handling is not supported currently. That's why we only provided a few color command in the esp_zigbee_zcl_command.h. Like 'esp_zb_zcl_color_move_to_color_cmd_req'

We will add hue/saturation support pretty soon. Regarding the contribution, unfortunately we can't share the source file of the Zigbee stack. Thanks for your help

u236, 2023-02-07:
> So you need to add other attributes manually like what you did (I assume you followed the example under customizd_server) If you don't register corresponding attribute, it will cause

I think it would be good to add checks for unregistered attributes and empty value pointers to prevent firmware crashes in future :)

likunqiao097304, 2023-03-01:
@u236 The latest commit has been fixed the firmware crashes issue + support hue/saturation attribute callback and related command. Please take a look to see if this issue is fixed.Thanks

chshu, 2023-03-21:
Closing the issue, @u236 feel free to reopen if any follow-up questions.

closed 2023-03-21 by chshu