espressif/esp-zigbee-sdk #5: How to actively report attributes?

lbuque, 2023-02-09:
I want to actively report some attributes to the protocol controller after the device is connected to the network, but I can't find the relevant API, what should I do?

I'm currently using examples/esp_zigbee_customized_devices/customized_client .

likunqiao097304, 2023-02-09:
Currently the `esp_zb_zcl_config_report_cmd_req` could do is reporting attribute if it changes value in min_interval time or report attribute in max_interval time if attribute not change. So for your actively report, is it meaning that you'd like to send report attribute by your action no matter time period or value change?

lbuque, 2023-02-10:
> So for your actively report, is it meaning that you'd like to send report attribute by your action no matter time period or value change?

yeah.

I need an api that can send Report attributes. This api seems to have not been implemented in esp-zigbee-sdk.

[image]

likunqiao097304, 2023-02-16:
> > So for your actively report, is it meaning that you'd like to send report attribute by your action no matter time period or value change?
> 
> yeah.
> 
> I need an api that can send Report attributes. This api seems to have not been implemented in esp-zigbee-sdk.
> 
> [image]

Hi, before I add new API to next round, I'd like to understand little bit more. What scenario you'd like to use this actively send report command, does configure reporting method doesn't meet your case? 

For example, you'd like to send report to certain remote device when local device power back online by your action? Or you'd like to send report without binding and configure report command action? Or you'd like to send report command triggered by your switch button or timer?

lbuque, 2023-02-16:
> For example, you'd like to send report to certain remote device when local device power back online by your action? Or you'd like to send report without binding and configure report command action? Or you'd like to send report command triggered by your switch button or timer?

You have given an example of the usage scenario of the report command here.

However, the Configure Reporting Command cannot report properties whose data types are arrays, structures, collections, or packages.

[image]

likunqiao097304, 2023-02-16:
> You have given an example of the usage scenario of the report command here.
>
> However, the Configure Reporting Command cannot report properties whose data types are arrays, structures, collections, [...]

But as far as I read from the ZCL specs, I didn't see most common cluster have types of structure packages or array.

likunqiao097304, 2023-03-01:
@lbuque The actively send the report attribute command API has been provided in the latest commit. Please take a look to see if it is fixed

chshu, 2023-03-21:
Closing the issue, @lbuque Let us know if any other issues.