espressif/esp-zigbee-sdk #6: Support for more ZCL clusters

jurriaan, 2023-02-25:
I'm trying to create a binary sensor using the zigbee SDK, but noticed that the Cluster/Attribute API is missing support for quite some clusters.

I was trying to create a On/Off Switch Configuration cluster for example:
```c
    esp_zb_attribute_list_t *esp_zb_on_off_configuration_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_ON_OFF_SWITCH_CONFIG);
    uint8_t switch_type = 0x01;
    esp_zb_custom_cluster_add_custom_attr(esp_zb_on_off_configuration_cluster, 0x0000, ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &switch_type);

    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_on_off_configuration_cluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);
```

Which resulted in these error messages:
```
E (422) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
E (422) ESP_ZIGBEE_CLUSTER: Custom cluster id is not within specified range!
```

I also am interested in using the binary input cluster / IAS cluster but these are also missing in the API.

Is it possible to use these clusters as a custom cluster or do I need to wait for official support in the API?

likunqiao097304, 2023-02-27:
@jurriaan Hi, firstly the on/off switch configuration cluster is not supported from our esp-zigbee-sdk yet. Customized cluster and attribute is used for any non-ZCL specs defines clusters. So you **can't** use those custom_cluster API to create a standard ZCL specs cluster.

[image]

So you need to wait for official support for the new APIs. BTW, you mentioned that on_off_switch configuration(0x0007), IAS cluster(0x0500) and BACnet Input, Output and Value Clusters(0x0608) are interested by you right? Is there any other clusters you also interested? We can take a consideration to make plan to implement them.

jurriaan, 2023-03-18:
Yes, these clusters would be really nice to have supported! Not sure yet which would be best for my application (momentary push button / binary state reporting).

JanUmbach, 2023-03-19:
I would also love to see an implementation of something like switch_type in the on_off_switch cluster.

closed 2023-03-21 by espressif-bot