espressif/esp-zigbee-sdk #7: Not getting attribute reports from group

hdavidka, 2023-02-26:
I'm using zigbee2mqtt as a coordinator and I've created a simple switch using the customized client demo. The switch successfully connected to the Z2M's network. I can bound the switch to a group (created in Z2M), and I can control the lamp (simple on/off cluster).
I've configured reporting for the bulb (in Z2M), I can see the state changes in Z2M and also in HA, but I've never receive the attributeReport with the ESP module, "esp_zb_dev_reporting_cb" never called. FYI, I can read the lamp state with "esp_zb_zcl_read_attr_cmd_req".

Can you help me with this?
Is there any debug or verbose mode available for the zigbee sdk?

likunqiao097304, 2023-02-27:
> Is there any debug or verbose mode available for the zigbee sdk?
Currently, there is no verbose mode available for sdk.

> but I've never receive the attributeReport with the ESP module, "esp_zb_dev_reporting_cb" never called
Did you register the `esp_zb_device_add_report_attr_cb`?

hdavidka, 2023-02-28:
I'm using the esp-zigbee-sdk\examples\esp_zigbee_customized_devices\customized_client demo, and I haven't modified the `esp_zb_device_add_report_attr_cb` part.

I haven't got a zigbee sniffer, so I cannot send you the zigbee packet. What I see in the Z2M debug logs is this:
`Received Zigbee message from 'bathlamp', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0`

I get the same debug message if I read this attribute. A verbose or sniffer mode in the sdk could also help in this case...

likunqiao097304, 2023-03-10:
> groupID 0

@hdavidka Hi, I am trying to duplicate your failure from my side. So for your network setup, you bind a switch to a group of lamps? and how do you configure report? for one lamp or group? 

>I've configured reporting for the bulb (in Z2M)

Also It seems like you didn't  use our API to do the configure reporting right?

hdavidka, 2023-03-12:
I've configured the report for each lamp individually, I haven't find an option in the Z2M to configure report for a group.
I didn't use the API to do the report configuration, I'm using the Z2M web interface.

likunqiao097304, 2023-03-13:
> I've configured the report for each lamp individually

Did you bind switch to a group or a lamp before you do the configured report?

hdavidka, 2023-03-19:
I'm created a group in Z2M and added the lamp to it. Z2M automatically configure the reports at that point.
I bind the switch to a group using Z2M. So creating and binding to the group I have only using Z2M.

If I read from the lamp with the esp API I got the same log message in Z2M (with groupID 0). But reading is ok, I can read the attribute state with ESP API.

It would be nice, if we can turn on a verbose mode in the SDK, so we can speed up this...

likunqiao097304, 2023-03-20:
> I bind the switch to a group using Z2M

Binding from switch to group means that switch one on-off command will send to all group member to control them (binding is one-direction). In order to receive the report from light, you may have to bind certain light to switch in order to make report successfully send. I am not sure there is a way to bind a group light to a single switch.

>  we can turn on a verbose mode in the SDK

We will definitely consider that

likunqiao097304, 2023-03-24:
@hdavidka Hi, have you tried to bind a light to a switch to see if there is report attribute command received from switch side?

closed 2023-04-06 by espressif-bot