espressif/esp-zigbee-sdk #10: Support for more ZCL clusters

JanUmbach, 2023-03-23:
I am interested to a Multistate Input, an Analog Input and a Power Configuration cluster, but these aren't supported yet.

Would be really cool if you could also implement this. (It is related to #6 but I don't know if I should reopen this issue or create a new one for a “cluster request”)

likunqiao097304, 2023-03-27:
@JanUmbach It is better to create a new one issue for your new ZCL clusters support. We will consider to add those clusters in the next round release

likunqiao097304, 2023-03-28:
@JanUmbach Just for curiosity, do you have any specific Zigbee devices that you need to those three clusters

likunqiao097304, 2023-04-04:
@JanUmbach We will support power configure cluster and release this week. But for the rest of two are not mandatory for the base device, so we will put into the future plan but not this time. For details, you can refer to CSA document to check this list : 16-02867-021-PRO_BDB-ZigBee-3.0-Cluster-List.xlsx.

closed 2023-04-06 by espressif-bot