espressif/esp-zigbee-sdk #12: Callback for receiving packets and sending packets directly

renzenicolai, 2023-03-25:
I'd like to be able to use the ESP32-C6 as a coordinator for an already existing Zigbee device which implements only the bare minimum of the Zigbee protocol.

These devices send out custom packets, how can I register a callback that gets called when a packet is received?

How can I send a custom Zigbee packet?

If such a feature is not available yet I request you to add support for directly receiving and sending Zigbee packets.

likunqiao097304, 2023-03-27:
@renzenicolai What kind of custom packets you'd like to send and receive? Could you explain little bit more? If you are talking about a custom ZCL command with customized cluster(0xfc00-0xffff), command ID. You could refer to `esp_zb_zcl_custom_cluster_cmd_req`, `esp_zb_add_custom_cluster_command_cb`

renzenicolai, 2023-03-27:
Thank you for the quick response! I've captured a packet as it appears over the air using a packet sniffer and Wireshark.

`Frame 544: 64 bytes on wire (512 bits), 64 bytes captured (512 bits) on interface -, id 0
IEEE 802.15.4 Data, Dst: 02:38:a5:76:3b:1e:ff:ff, Src: TexasIns_00:14:d9:49:35
    Frame Control Field: 0xcc41, Frame Type: Data, PAN ID Compression, Destination Addressing Mode: Long/64-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Long/64-bit
        .... .... .... .001 = Frame Type: Data (0x1)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..0. .... = Acknowledge Request: False
        .... .... .1.. .... = PAN ID Compression: True
        .... .... 0... .... = Reserved: False
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 11.. .... .... = Destination Addressing Mode: Long/64-bit (0x3)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        11.. .... .... .... = Source Addressing Mode: Long/64-bit (0x3)
    Sequence Number: 98
    Destination PAN: 0x4447
    Destination: 02:38:a5:76:3b:1e:ff:ff (02:38:a5:76:3b:1e:ff:ff)
    Extended Source: TexasIns_00:14:d9:49:35 (00:12:4b:00:14:d9:49:35)
    FCS: 0xde49 (Correct)
Data (41 bytes)
    Data: 55233183087d18c61df6373f445c4e80881848f5a36b587de00a4f69f6bee21661f67992…
    [Length: 41]
`

The data contained in this packet is an encrypted custom protocol. I'd like to send and receive the raw data contained in these packets from my application running on an ESP32-C6.

How can I send and receive these custom 802.15.4 packets?

For completeness here is a hex dump of a full packet:

`0000   41 cc 62 47 44 ff ff 1e 3b 76 a5 38 02 35 49 d9
0010   14 00 4b 12 00 55 23 31 83 08 7d 18 c6 1d f6 37
0020   3f 44 5c 4e 80 88 18 48 f5 a3 6b 58 7d e0 0a 4f
0030   69 f6 be e2 16 61 f6 79 92 be 30 df 21 64 49 de`

likunqiao097304, 2023-03-28:
our esp-zigbee-sdk doesn't provide sending / receiving 802.15.4 packet. However, you could refer to those API from [802.15.4 components](https://github.com/espressif/esp-idf/blob/master/components/ieee802154/include/esp_ieee802154.h) and search `esp_ieee802154_receive_done` and `esp_ieee802154_transmit`. Example you could refer[154 command](https://github.com/espressif/esp-idf/blob/master/components/ieee802154/test_apps/test_ieee802154/main/src/ieee802154_cmd.c)

renzenicolai, 2023-03-31:
Thank you for explaining, I've gotten it to work using the esp_ieee802154 component.

(And for other people finding this thread via google, have a look at https://github.com/badgeteam/esp32c6-firmware-esl-station for my solution)

closed 2023-03-31 by renzenicolai