espressif/esp-zigbee-sdk #13: ESP32C6 cannot join network

Tropaion, 2023-04-19:
Hello,
I'm new to ZigBee developement and I'm a bit stumbed.
I'm trying to develop a simple temperature sensor zigbee application and already implemented the library for it.

Now, currently I'm trying to understand the HA_ON_OFF_LIGHT example and I think I have a general understanding how it works.
Only the commissioning part leaves me stumbed.

My current test setup is Home Assistant with an Sonoff Gateway USB Stick.
I'm trying for a while now, for the example code to join the HA Network, but I always get the "ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)" error.

Not sure why this happends, since the commission mode is set to "SP_ZB_BDB_MODE_NETWORK_STEERING" and as far as I understand, this should work.

I hope someone can help me. Sadly there isn't a good guide for zigbee novices.

Thanks for your help :)

closed 2023-04-19 by Tropaion

jayavanth, 2023-06-07:
@Tropaion did you figure out a solution for this? Experiencing the same issue

Tropaion, 2023-06-07:
@jayavanth For me it was the problem that in the example there where only a few channels allowed, so it couldn't my coordinator.

jayavanth, 2023-06-08:
Wow that worked! Thank you @Tropaion

xieqinan, 2023-06-08:
Hi @Tropaion @jayavanth,
The channel of ESP-Zigbee is limited from 11 to 26，and you can set this range using the `#define ESP_ZB_PRIMARY_CHANNEL_MASK (1l << 13)` directive in every example header file. 
The log message "ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)" may occur due to existing network information. In other words, if a Zigbee device, such as an "End Device," forms a network with "Coordinator A," by default, the network information about "Coordinator A" will be stored in the NVRAM. This restricts the "End Device" from joining networks other than "Coordinator A," resulting in a failed join attempt. If it is about this issue, you can call esp_zb_nvram_erase_at_start(true); before esp_zb_start(false) to erase the network information from the device's NVRAM.

jayavanth, 2023-06-08:
@xieqinan that's very helpful. Thank you!