espressif/esp-zigbee-sdk #14: Missing defines

Tropaion, 2023-04-21:
I'm currently trying to implement/test the preconfigured temperature device.
```
    // For test, use preconfigured temperature device
    esp_zb_temperature_sensor_cfg_t temperature_cfg = ESP_ZB_DEFAULT_TEMPERATURE_SENSOR_CONFIG();

    // Create endpoint list for device
    esp_zb_ep_list_t *temperature_ep = esp_zb_temperature_sensor_ep_create(HA_ESP_TEMPERATURE_ENDPOINT, &temperature_cfg);
```
But it seems that in ```ESP_ZB_DEFAULT_TEMPERATURE_SENSOR_CONFIG()``` are missing defines.

In file "esp_zigbee_ha_standard.h", column 289:
```
/**
 * @brief Zigbee HA standard temperature sensor device default config value.
 *
 */
#define ESP_ZB_DEFAULT_TEMPERATURE_SENSOR_CONFIG()                                                  \
    {                                                                                               \
        .basic_cfg =                                                                                \
            {                                                                                       \
                .zcl_version = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE,                          \
                .power_source = ESP_ZB_ZCL_BASIC_POWER_SOURCE_DEFAULT_VALUE,                        \
            },                                                                                      \
        .identify_cfg =                                                                             \
            {                                                                                       \
                .identify_time = ESP_ZB_ZCL_IDENTIFY_IDENTIFY_TIME_DEFAULT_VALUE,                   \
            },                                                                                      \
        .temp_meas_cfg =                                                                            \
            {                                                                                       \
                .measured_value = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_UNKNOWN,                   \
                .min_value = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MIN_VALUE_INVALID,                    \
                .max_value = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MAX_VALUE_INVALID,                    \
            },                                                                                      \
    }
```
The defines "ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_UNKNOWN", "ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MIN_VALUE_INVALID", "ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MAX_VALUE_INVALID" are missing.

likunqiao097304, 2023-04-21:
@Tropaion Thanks for pointing out. We will fixed the name issue in next  release, but for work around you could use those defines:
`/** @brief Default value for Value attribute */
#define ESP_ZB_ZCL_TEMP_MEASUREMENT_VALUE_DEFAULT_VALUE ((int16_t)0x8000)

/** @brief Default value for MinValue attribute */
#define ESP_ZB_ZCL_TEMP_MEASUREMENT_MIN_VALUE_DEFAULT_VALUE ((int16_t)0x8000)

/** @brief Default value for MaxValue attribute */
#define ESP_ZB_ZCL_TEMP_MEASUREMENT_MAX_VALUE_DEFAULT_VALUE ((int16_t)0x8000)`

Tropaion, 2023-04-21:
@likunqiao097304 Cool, thanks for your fast response.
One off-topic question, I configured a timer which start temperature sensor callback, so that it measures temperature every minute, but I still haven't found out, what is the correct/recommended way to update the temperature attribute?
And same question for setting vendor id and device id.

likunqiao097304, 2023-04-21:
@Tropaion are you looking for update the temperature attribute value? It might be this `esp_zb_zcl_set_attribute_val`

closed 2023-04-21 by Tropaion