espressif/esp-zigbee-sdk #15: Unknown error, not sure what causes it

Tropaion, 2023-04-21:
Currently I'm trying to modify the "HA_on_off_light" example to work with the preconfigured temperature device.
I didn't change anything in the "esp_zb_init()" parameters and the code before but suddenly throws error:
```
assert failed: zb_osif_nvram_init zb_esp_nvram.c:70 (zb_partition != NULL)
Core  0 register dump:
MEPC    : 0x4080062e  RA      : 0x4080862e  SP      : 0x4087e120  GP      : 0x4080e050
0x4080062e: panic_abort at C:/Users/Tropaion/esp/esp-idf/components/esp_system/panic.c:451

0x4080862e: __ubsan_include at C:/Users/Tropaion/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x40873708  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x0000007d  S1      : 0x00000001  A0      : 0x4087e15c  A1      : 0x4080e245
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x40815000
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x4087e27a
S4      : 0x4080e244  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4087e120: 0xa5a5a5a5 0x00000004 0x420541bc 0x4080c88a 0x4080e860 0x420541bc 0x4080eb48 0x42054181
0x4080c88a: newlib_include_assert_impl at C:/Users/Tropaion/esp/esp-idf/components/newlib/assert.c:93

4087e140: 0x4080e870 0x4087e154 0x4080e874 0x42054130 0x4080e244 0x00003037 0x4087e19c 0x65737361
4087e160: 0x66207472 0x656c6961 0x7a203a64 0x736f5f62 0x6e5f6669 0x6d617276 0x696e695f 0x627a2074
4087e180: 0x7073655f 0x72766e5f 0x632e6d61 0x2030373a 0x5f627a28 0x74726170 0x6f697469 0x3d21206e
4087e1a0: 0x4c554e20 0x0000294c 0x4087cf00 0x40808cb8 0x00000000 0x00000000 0x408171d0 0x4080c590
0x40808cb8: xQueueGenericSend at C:/Users/Tropaion/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:999

0x4080c590: _multi_heap_unlock at C:/Users/Tropaion/esp/esp-idf/components/heap/multi_heap.c:94 (discriminator 1)

4087e1c0: 0x00000000 0x00000000 0x00000000 0x40022498 0x40880000 0x4087cfcc 0x4087c610 0x400225ca
4087e1e0: 0x42054124 0x00000000 0x4087cfcc 0x40800ac8 0x00000001 0x00000014 0x4087cfcc 0x4080c8b0
0x40800ac8: heap_caps_free at C:/Users/Tropaion/esp/esp-idf/components/heap/heap_caps.c:391

0x4080c8b0: free at C:/Users/Tropaion/esp/esp-idf/components/newlib/heap.c:40

4087e200: 0x42054124 0x00000000 0x4087cfcc 0x4203cebe 0xffffffff 0xffffffff 0x00000000 0x4203cf78
0x4203cebe: esp_partition_iterator_release at C:/Users/Tropaion/esp/esp-idf/components/esp_partition/partition.c:357

0x4203cf78: esp_partition_next at C:/Users/Tropaion/esp/esp-idf/components/esp_partition/partition.c:333

4087e220: 0x42054124 0x00000000 0x00000000 0x42000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e240: 0x00000000 0x00000000 0x4080fd18 0x420363ba 0x4080a9b8 0x00000000 0x4087e2a4 0x4200f0ae
0x420363ba: zb_osif_nvram_init at ??:?

0x4080a9b8: vPortTaskWrapper at C:/Users/Tropaion/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:198

0x4200f0ae: zb_init at ??:?

4087e260: 0x40873708 0x00000000 0x4087e2a4 0x42007624 0x00000000 0x00000000 0x00000000 0x000001ce
0x42007624: esp_zb_init at ??:?

4087e280: 0x420533bc 0x000001ba 0x42053000 0x42007166 0x00000000 0x00000000 0x00000000 0x00000000
0x42007166: zb_hte501_task at C:/Users/Tropaion/Desktop/ZB_HTE501/components/zb_hte501/zb_hte501.c:105

4087e2a0: 0x00000000 0x00000002 0x00060000 0x00000bb8 0x00000000 0x00000000 0x00000000 0x4080a9c4
0x4080a9c4: vPortTaskWrapper at C:/Users/Tropaion/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205

4087e2c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e2e0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154
4087e300: 0x4087e250 0x4087c624 0x4080f28c 0x4080f28c 0x4087e300 0x4080f284 0x00000014 0x00076270
4087e320: 0x00000000 0x4087e300 0x00000000 0x00000005 0x4087d2fc 0x4d5f425a 0x004e4941 0x00000000
4087e340: 0x00000000 0x00000000 0x4087e2f0 0x00000005 0x00000000 0x00000000 0x00000000 0x00000000
4087e360: 0x40815cd8 0x40815d40 0x40815da8 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
4087e380: 0x00000000 0x4204b43a 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0x4204b43a: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229
[... 86 lines cut ...]
}

esp_err_t zb_hte501_init()
{
    // Initialize non-volatile flash
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_LOGI(TAG, "FLASH INIT DONE!");

    // Operating mode (single processor or co-processor)
    esp_zb_platform_config_t config = {
        .radio_config = ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ZB_DEFAULT_HOST_CONFIG(),
    };

    // Set operating mode
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    ESP_LOGI(TAG, "CONFIG DONE");

    // TODO: Maybe initialize sensor here?

    // Create zigbee background task
    xTaskCreate(zb_hte501_task, "ZB_MAIN", 4096, NULL, 5, NULL);
    return ESP_OK;
}
```

likunqiao097304, 2023-04-21:
@Tropaion I haven't read all code, but at least I found that: `esp_zb_temperature_meas_cluster_create()` is return `esp_zb_attribute_list_t` not `esp_zb_ep_list_t`

Please refer to the customized_client/server example. Also what I suggest you do is:

1.  **create a empty cluster list** -> esp_zb_zcl_cluster_list_create()
2. **create a basic, identify, temperature, humdity cluster** -> esp_zb_basic_cluster_create(), esp_zb_identify_cluster_create(), esp_zb_temperature_meas_cluster_create(), esp_zb_humidity_meas_cluster_create()
3. **add them into cluster_list** -> esp_zb_cluster_list_add_xxxx_cluster()
4. **create an empty endpoint  list** ->esp_zb_ep_list_create()
5. **add this cluster_list(endpoing) into this endpoint list** -> esp_zb_ep_list_add_ep()
6. **then register it** ->esp_zb_device_register();

Tropaion, 2023-04-21:
@likunqiao097304 Thanks for the information, that good to know but sadly not the reason for the error.
The programm fails at "esp_zb_init()" and gives an nvram assert error. I doesn't even make it to the cluster creation.

Tropaion, 2023-04-21:
@likunqiao097304 Could the reason be that I haven't defined a custom partition table?
Not sure if it is needed for zigbee sdk or not, but in the example, there is one defined.

likunqiao097304, 2023-04-23:
> Could the reason be that I haven't defined a custom partition table?
Not sure if it is needed for zigbee sdk or not, but in the example, there is one defined.

If you're talking about the "partitions.csv"? Yes, you need a file to allocate each of the zone in the memory. From the error log, the `zb_osif_nvram_init()` it looks for the "zb_storage" partition, but it seems like it can't find it in the "partition.csv"

likunqiao097304, 2023-04-23:
> EDIT: another off-topic question. Is a attribute callback function needed?

It is not mandatory needed. But if user wants to do some actions when any of the attribute changes, then it needs to set this callback for this purpose. Like light on/off.

Tropaion, 2023-04-23:
@likunqiao097304 Thanks for your help, I'm very grateful.
I hope a can bother you with another question, whichs preoccupies me the whole weekend now.
I'm not sure if it is a bug or I'm making some mistakes (probably).
I created my own endpoint device: Basic Cluster (Server), Identity Cluster (Server + Client), Temperature Cluster (Server) and Humidity Cluster (Server).
Creating the enpoint list and registering the device works and I don't get any errors either.
But when I try joining a network, it says "Network joined" and then stops, nothing happens.
On Coordinator side, nothing happens either, no device is created, the interview doesn't, maybe a device signature problem?
I'm searching for hours now and I'm completely clueless, I hope you can enlighten me.
[image]
[code_snippet.zip](https://github.com/espressif/esp-zigbee-sdk/files/11303496/code_snippet.zip)

likunqiao097304, 2023-04-23:
@Tropaion I briefly checked your register device part of code, looks good to me. It seems like your zigbee end-device(humidity sensor) is scanned a Zigbee network on channel 11 and joined successfully. The rest of work should be done from the coordinator side but I can't see them. Es tut mir Leid. By the way, did you enable the install code policy from coordinator side? by default our zigbee example didn't enable them, so once a device request to join it should be join without needing any install code go through.

Tropaion, 2023-04-23:
@likunqiao097304 Not sure but I don't think it is needed, since the "customized_client" example worked (sometimes).

But I tried enabling it and now steering failes continiously:
```
I (222962) zb_hte501: Network steering was not successful (status: -1)        
I (232742) zb_hte501: ZDO signal: 3, status: 0
I (232742) zb_hte501: Network steering was not successful (status: -1)        
I (242512) zb_hte501: ZDO signal: 3, status: 0
I (242512) zb_hte501: Network steering was not successful (status: -1)        
I (252292) zb_hte501: ZDO signal: 3, status: 0
I (252292) zb_hte501: Network steering was not successful (status: -1) 
```
But the coordinator shows some new logs repeatedly:
```
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=38, profile_id=0, cluster_id=32822, data=Serialized[b'&\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
Device 0x3860 (ff:fe:c0:28:40:ca:4c:40) joined the network
[0x3860] Scheduling initialization
[0x3860] Canceling old initialize call
Tries remaining: 3
[0x3860] Discovering endpoints
Tries remaining: 3
[0x3860] Extending timeout for 0xf9 request
Device 0x3860 (ff:fe:c0:28:40:ca:4c:40) joined the network
[0x3860] Scheduling initialization
[0x3860] Canceling old initialize call
Tries remaining: 3
[0x3860] Discovering endpoints
Tries remaining: 3
[0x3860] Extending timeout for 0xfa request
Device 0x3860 (ff:fe:c0:28:40:ca:4c:40) joined the network
[0x3860] Scheduling initialization
[0x3860] Canceling old initialize call
Tries remaining: 3
[0x3860] Discovering endpoints
Tries remaining: 3
[0x3860] Extending timeout for 0xfb request
Device 0x3860 (ff:fe:c0:28:40:ca:4c:40) joined the network
[0x3860] Scheduling initialization
[0x3860] Canceling old initialize call
Tries remaining: 3
[0x3860] Discovering endpoints
Tries remaining: 3
[0x3860] Extending timeout for 0xfc request
Device 0x3860 (ff:fe:c0:28:40:ca:4c:40) joined the network
[0x3860] Scheduling initialization
[0x3860] Canceling old initialize call
Tries remaining: 3
[0x3860] Discovering endpoints
Tries remaining: 3
[0x3860] Extending timeout for 0xfd request
Device 0x3860 (ff:fe:c0:28:40:ca:4c:40) joined the network
[0x3860] Scheduling initialization
[0x3860] Canceling old initialize call
Tries remaining: 3
[0x3860] Discovering endpoints
Tries remaining: 3
[0x3860] Extending timeout for 0xfe request
```

Tropaion, 2023-04-23:
I have the feeling it has something to do with these two messages:
```
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
[0x3860:zdo] No handler for ZDO request:ZDOCmd.Active_EP_rsp([<Status.DEVICE_NOT_FOUND: 129>, 0x3860, []])
```

likunqiao097304, 2023-04-25:
> "manuf" and "model" are empty even though I have set them. Maybe some bug there?

Sorry I have noticed that before, It is a bug there. You created a function `zb_hte501_ep_create` with local variables, and then you leave this function and back to `zb_main_task`. so your local variables are gone. Suggest you created as static value.

Tropaion, 2023-04-25:
> Your are right I haven't noticed that before, It is a bug there. You created a function `zb_hte501_ep_create` with local variables `model_identifier`, and then you leave this function and back to `zb_main_task`. so your local variables are gone. Suggest you created as static value for your attribute and clusters. Or move your code to the `zb_main_task`

Good to know, I thought the function will copy the values and set them internally. Sadly there isn't a nice solution then.
Because I will have multiple endpoints in the future, I wanted to separate them in their own library.

Tropaion, 2023-04-25:
@likunqiao097304 I now set everything to static, didn't change anything.
Then I moved everything to ```zb_main_task```, didn't change anything either, I'm completely out of ideas, can you maybe take a closer look at it?

likunqiao097304, 2023-04-25:
>  is there a limit how often I can update an attribute?

No limitation. 

> For example manuf = "mymanuf", the stack only sends "ymanuf".

Not sure how you did this. Did you send the attribute or read this attribute? I'd like to see if I can duplicate from my side

Tropaion, 2023-04-25:
> Not sure how you did this. Did you send the attribute or read this attribute? I'd like to see if I can duplicate from my side

Basically, on the esp i created an periodic timer with an callback function which is called every 5 seconds.
The callback functions gets temperature and humidity from the sensor and sets the values with
```
    esp_zb_zcl_status_t state = esp_zb_zcl_set_attribute_val(HA_HTE501_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, &attr_temp);

state = esp_zb_zcl_set_attribute_val(HA_HTE501_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_REL_HUMIDITY_MEASUREMENT_VALUE_ID, &attr_hum);```

But the values are not send every 5 seconds, only sometimes.

likunqiao097304, 2023-04-26:
> But the values are not send every 5 seconds, only sometimes.

You set your device attribute every 5 seconds, it is only attribute change on your local node. how can you send this value to other node? By reporting? or binding? That is why I'd like to know the behavior of other side

Tropaion, 2023-04-26:
@likunqiao097304 Not sure what you mean. When I change the attribute on my local node doesn't the stack automatically send cluster attribute update?
I didn't do anything special, just create ep on node, join coordinator (home assistant) and set_val every 5 seconds with timer callback.

Tropaion, 2023-04-26:
@likunqiao097304 Ok, good to know, thanks. Which way would you recommend for an temp/hum device? Is there a example/docu about report mechanism/device binding? I don't really know how to do this.

likunqiao097304, 2023-04-26:
@Tropaion For easy demo or just report your attribute whenever the moment you may use `esp_zb_zcl_report_attr_cmd_req`. 
For delicate automatically report attribute, you could refer the bind -report mechanism in the example[customized_switch](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c)

likunqiao097304, 2023-04-27:
For our API design, we should separate them for single functionality purpose. For now, we just do `esp_zb_zcl_set_attribute_val` and then call `esp_zb_zcl_set_attribute_val` to send your changed attribute to remote node.

Tropaion, 2023-04-27:
@likunqiao097304 Thanks again.
Another question 😄. The sensor I use also provides the dew point, which I want to send.
According to the ZB Cluster Library, the Analog Input (Basic) Cluster provides that attribute. Now I have seen that the identifier is already defined in ```esp_zigbee_zcl_common.h```, but there is no ```esp_zb_analog_input_add_attr``` function available. Is this still in developement or is there another way to use the cluster?

likunqiao097304, 2023-05-04:
@Tropaion The analog input has not supported yet, currently is in development. We will provide it in the future release. Please refer this to see our supported ZCL:[sdk docs](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/introduction.html).

Tropaion, 2023-05-05:
Ok, thank you.