espressif/esp-zigbee-sdk #16: Setting attribute before start not working

Tropaion, 2023-04-21:
I want/tried to set the attribute before zigbee starts working, so that it's available while joining.
I thought it should be possible, since "ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG()" also sets values.
I tried this:
```
esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
esp_zb_device_register(esp_zb_on_off_light_ep);

char manufacturer[] = "mycompany";
esp_zb_zcl_set_attribute_val(HA_ESP_LIGHT_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_BASIC, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, (void *)&manufacturer);

esp_zb_device_add_set_attr_value_cb(attr_cb);
esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
ESP_ERROR_CHECK(esp_zb_start(false));

esp_zb_main_loop_iteration();
```
What am I doing wrong? Thanks for your help :)

likunqiao097304, 2023-04-23:
@Tropaion  In `ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG` I only create two mandatory attributes (zcl_version, power_source) in basic cluster.  You can't set/update a attribute that  is not defined yet, like your new attribute `ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID`. 

My suggestion is add new/more attribute before you registered your device `esp_zb_device_register`. Then you could use 'esp_zb_zcl_set_attribute_val' when you wants to change it.

Tropaion, 2023-04-23:
@likunqiao097304 Thanks for your help.

closed 2023-04-23 by Tropaion