espressif/esp-zigbee-sdk #17: rcp protocol documentation

xppt, 2023-04-23:
Hello!
1. What is the protocol used to communicate between gateway and rcp? Is there any spec describing it?
2. Is there any client implementation of rcp protocol for linux?

likunqiao097304, 2023-04-23:
@xppt we are using UART communication with our own format package between gateway and rcp. No specs describing it. 

> Is there any client implementation of rcp protocol for linux?

Are you talking about  using rcp as client and directly connect to linux like PC? Could you explain little bit more?

xppt, 2023-04-23:
> Are you talking about using rcp as client and directly connect to linux like PC? Could you explain little bit more?

Sure, I need to use Zigbee stack from my linux soc via UART.
Connection is made like this `[linux-soc] <=> [esp32c6]`.
Currently it seems impossible to use esp rcp server on esp32c6, and I need to implement my own high level protocol inside uart connection.
But it would be easier to be able to connect using some transparent protocol like spinel, and only have my business logic on client side (inside linux soc).

likunqiao097304, 2023-04-24:
@xppt Correct me if I am understand right. You have your own Zigbee stack available from linux soc side, but you need a 802.15.4 driver like RCP to handle. The protocol spinel is used in our other 802.15.4 example thread-bourder router (soc+rcp). 
For Zigbee example, we are planning to provide other solution like you want

chshu, 2023-06-15:
Closing since the question is answered.

closed 2023-06-15 by chshu