espressif/esp-zigbee-sdk #18: Rejoin after connection is lost?

Tropaion, 2023-05-02:
Hello,

currently I'm trying to improve signal handling and I want to handle the case, when the network connection was lost, for example because the zigbee coordinator had a power out.

I didn't find the correct signal for it. When I tried to simulate the error, I get, at first, a few times the ZDO Signal "50" and after that the ZDO Signal "60". These signal values are not defined, even in the zboss stack.

Is there another way to handle this error?

likunqiao097304, 2023-05-04:
@Tropaion Are you talking about the rejoin for end-device/router or Zigbee coordinator? "60“ means device unavailable (we will provide this define later). If you lose connection, there is no signal shows you lose connection. Only you are trying to rejoin the network or re-form the network it has some signal.

Tropaion, 2023-05-05:
@likunqiao097304 I meant rejoin of the end-device I programmed.
Case: Coordinator loses power supply, my device falls of. After Coordinator is only again, device does not rejoin network.

energizer91, 2023-05-15:
I have the same issue. It's really hard to catch the moment when device has to rejoin or been reset. Is there any specific place in callbacks or so where I can handle such case?

xieqinan, 2023-05-29:
Hi @Tropaion @energizer91 ,
There are several methods that can help you handle the problem of a lost connection. Assuming that the **Zigbee Coordinator** and **End Device** have already formed a Zigbee network, the following methods can be applied:

Handling Long-Term Power Loss:
If the **Zigbee Coordinator** loses power supply for a long time (exceeding 10 minutes), the **End Device** will send the `ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT` signal to the application layer. You can utilize this signal to obtain the current state of the End Device.

Handling Short-Term Power Loss:
If the Zigbee Coordinator experiences a brief power loss, it will perform one of two behaviors based on the following configuration:

If the function `esp_zb_nvram_erase_at_start(true)` is called before `esp_zb_start()`, the Coordinator will not read information from the non-volatile random-access memory (nvram). As a result, a new Zigbee network will be formed by the Coordinator, and the End Device will not join this new network. After a delay of 5 minutes, the End Device will also send the `ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT` signal to the application layer.

If the function `esp_zb_nvram_erase_at_start(false)` is called before esp_zb_start(), the Coordinator will read the network information from the nvram. In this case, the original Zigbee network will remain intact.

By implementing these methods maybe solve the `rejoin` problem, you can effectively handle the problem of a lost connection in a Zigbee network.

Tropaion, 2023-05-30:
So basically we can only know of a lost network after 10 minutes?
Also, is there a rejoin command or do I have to reboot the device every 10 minutes until network is found again?

xieqinan, 2023-05-31:
In order to establish a Zigbee network and synchronize network information, both the coordinator and end device need to implement processes such as `beacon request`, `association`, `transport key` and `permit join`.  Additionally, the end device should utilize a periodic "data request" (with a customizable interval) to maintain its connection with the coordinator. If the end device sends ten consecutive data requests without receiving a response, it assumes that the coordinator is offline.

If the coordinator powers off and restarts with the `esp_zb_nvram_erase_at_start(true)` configuration, it will erase the original network information and form a new network. In this case, the end device will not be able to find the original network unless it restarts itself to search for a new network.

On the other hand, if the coordinator powers off and restarts with the `esp_zb_nvram_erase_at_start(false)` configuration, the original Zigbee network will remain active, and there will be no need to rejoin the network.

closed 2023-06-19 by Tropaion