espressif/esp-zigbee-sdk #19: Cannot join existing network using examples

patryksiwek, 2023-05-04:
I'm trying to join existing Zigbee mesh using examples provided in the SDK but nothing works.
Board: ESP32-H2-DevKitM-1
I've already switched to newest masters:
esp-idf: `56123c52aaa08f1b53350c7af30c91320b352ef4`
esp-zigbee-sdk: `34d4b8db9ed4e3432aafcbdd734254f1ab99dc31`

I'm trying to introduce as little changes to examples as possible to find problem so now I'm using "HA_on_off_light" with only changes regarding esp flash size, flash speed and channel that my network is using (15). Zigbee mesh is used in HomeAssistant under ZHA (I've also reconfigured it for test under Zigbee2MQTT but got same results). Other devices in mesh are communicating correctly with each other. 
ESP logs looks like this:
```
...
I (355) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (385) phy: phy_version: 101,0, 76d28a9, Mar 30 2023, 17:12:12
I (395) main_task: Returned from app_main()
I (405) ESP_ZB_ON_OFF_LIGHT: ZDO signal: 23, status: -1
I (405) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (405) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (3065) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (6735) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (10405) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (14075) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (17745) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
```
In the same time I have pairing enabled in HA but steering is constantly failing. Each retry ESP is sending Beacon Request and coordinator replies with Beacon but that's all and cycle starts from the beginning.
[image]
Do you have any ideas where the problem could be?

closed 2023-05-06 by espressif-bot

abdulrehmanee12, 2023-05-06:
@patryksiwek  the bug has been fixed , please use the latest  esp-zigbee-sdk commit  and try again.

patryksiwek, 2023-05-06:
It is working now, thanks for fix.