espressif/esp-zigbee-sdk #20: [Feature request] Support more ZCL clusters

rfortunatov, 2023-05-10:
Hi, I'm trying to build a multifunctional sensor that would include MMWave radar, Pir sensors, climate and air quality sensor along with illuminance sensor. For this I need an ability to create ESP_ZB_ZCL_CLUSTER_ID_ILLUMINANCE_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_ID_PRESSURE_MEASUREMENT and ESP_ZB_ZCL_CLUSTER_ID_OCCUPANCY_SENSING. Would you be so kind to add them to SDK, or maybe there's some way I can contribute and add them myself?

likunqiao097304, 2023-05-10:
@rfortunatov Thanks for your wiliness to contribute. Currently the source code of the esp-zigbee-sdk is not exposed to the public.  We will consider to add them(cluster id: 0x0400, 0x0403, 0x0406) in later release

rfortunatov, 2023-05-10:
@likunqiao097304  Thanks, is there a public roadmap I can keep track of?

likunqiao097304, 2023-05-10:
@rfortunatov Currently no road map. For new clusters support, our plan is check/implement those new clusters if someone need.  For your request,  I think we will add them pretty fast.

rfortunatov, 2023-05-11:
@likunqiao097304  Can you please also add clusters 0x040c and 0x042A as per https://csa-iot.org/wp-content/uploads/2022/01/07-5123-08-Zigbee-Cluster-Library-1.pdf

embts, 2023-05-13:
Hi,
can you also add these clusters?

- ESP_ZB_ZCL_CLUSTER_ID_DEVICE_TEMP_CONFIG
- ESP_ZB_ZCL_CLUSTER_ID_ANALOG_VALUE
- ESP_ZB_ZCL_CLUSTER_ID_MULTI_VALUE
- ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING

rfortunatov, 2023-05-17:
@likunqiao097304  Don't mean any pressure, but are there some estimates on this? I'm really eager to put together my sensor :)

xieqinan, 2023-05-29:
Hi @rfortunatov @Tropaion 
These clusters will be added as soon as possible.

closed 2023-06-09 by espressif-bot

rfortunatov, 2023-06-09:
Not quite done from what I see. Following clusters were not added by closing commit.
 
Smart Energy Metering (0x0702)
Analog Input Clustering (0x000C)
Carbon Dioxide Measurement (0x040c)
PM2.5 Measurement (0x042A) 
Multi State Value (0x0014)

xieqinan, 2023-06-12:
Yes, these clusters are not supported by the current version. You can reopen this issue until all clusters are supported.

> Not quite done from what I see. Following clusters were not added by closing commit.
> 
> Smart Energy Metering (0x0702) Analog Input Clustering (0x000C) Carbon Dioxide Measurement (0x040c) PM2.5 Measurement (0x042A) Multi State Value (0x0014)