espressif/esp-zigbee-sdk #21: Coordinator sends "leave" command after the association

Ravenen, 2023-05-10:
I am using `esp-zigbee-gateway` example on the ESPC6. 
I added some communications after the device end device association: ask active endpoints list, ask simple descriptor on each endpoint.
After that, the `ESP_ZB_ZDO_SIGNAL_LEAVE_INDICATION` signal is received, and I see in the WireShark that coordinator sends the `leave` command to that device.
[image]
[image]
[image]
[image]

[zigbee sdk pairing pcapng.zip](https://github.com/espressif/esp-zigbee-sdk/files/11442216/zigbee.sdk.pairing.pcapng.zip)

likunqiao097304, 2023-05-11:
It seems like coordinator is trying to send the nwk key(No.14) but no further action from end-device to request  trust center link key.  Is end-device also our Zigbee stack or our board? Also by default we disable using install code policy to do the network commissioning, so please check both size have the same security mechanism.

Ravenen, 2023-05-11:
Thank you for your response!
End Device is a Sonoff temperature sensor. I didn't change any security configs in the example. So, again, what are the recommendations to fix that? Or should I give some more information?

likunqiao097304, 2023-05-12:
Does this Sonoff temperature sensor use Zigbee 3.0 standard way to do the commission? Like it has install code and MAC on the label. I'd like to know how is process from your side to let your sensor join the network. Have you tried to join other Zigbee coordinator successfully?

From the log you provided, our Zigbee coordinator did send the network key successfully (a351a85....). And later encrypted ZDO active endpoint, simple descriptor request, your sensor did decrypte correctly with response. However, end-device(sensor) didn't do response/action with the network key. 

Here is a picture that we see as successfully join. Highlight them in blue:
[image]

Ravenen, 2023-05-12:
Yes, I have tried to join it to other coordinator with CC ZigBee stick and ZigBee2Mqtt software. I also have the association process captured with the WireShark. There is also no response to the network key, but the coordinator keeps the device in the network.  

[zigbee switch pairing.zip](https://github.com/espressif/esp-zigbee-sdk/files/11464739/zigbee.switch.pairing.zip)

[image]
[image]
[image]

Ravenen, 2023-05-12:
And yes, the manual of the device does say that it supports ZigBee 3.0.  
[image]

Ravenen, 2023-05-18:
So... Any suggestions here? What can I possibly do to help with my issue?

xieqinan, 2023-05-24:
Hi Ravenen,
The problem described above can potentially be resolved by disabling the `bdbTrustCenterRequireKeyExchange` MACRO in the Zigbee Gateway. This solution is outlined in the Zigbee BDB specification provided below. An API for configuring the MACRO will be made available as soon as possible in ESP-ZigBee-SDK.
[image]

The End Device will be allowed to join the Zigbee network without the requirement of exchanging a link key, if the Zigbee Gateway disables `bdbTrustCenterRequireKeyExchange` option. However, it is important to note that this approach is considered unsafe for the Zigbee network.

closed 2023-05-26 by espressif-bot

chshu, 2023-05-26:
@Ravenen The issue should be resolved by disable the option: https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_secur.h#L133