espressif/esp-zigbee-sdk #23: [Feature request] Add the destination endpoint to the report callback

diazmanuel, 2023-05-18:
Hello, I would like you to add the endpoint to which the report is being sent to the callback function. 

In my case, I have a device with multiple endpoints (as switches) and an external device for each endpoint with which it is bind. when the external device send the report the callback only indicates the source endpoint and not the destination.

currently the callback prototype only contemplates the source endpoint and not the destination as show in the doc :
[https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/esp_zigbee_core.html?highlight=esp_zb_device_add_report_attr_cb#_CPPv429esp_zb_report_attr_callback_t](url)

Thanks

likunqiao097304, 2023-05-19:
@diazmanuel Currently we do not support callback function to have destination endpoint. 

If we keep the current code as it is, when you do the binding or configure report command, you may remember which remote device endpoint and local endpoint you are bind to. So we when you trigger a report callback with remote address and remote endpoint, you should know which destination endpoint on your local device is. 

Again, we can add destination endpoint info into the callback, but we need to consider how much benefit we could get if we change the code and callback structure.

diazmanuel, 2023-05-19:
There are several cases in which having the destination endpoint would make it easier and would allow the functions to be handled correctly.

1) This first case I am not 100% sure, but in the callback of the report the address that returns is in short format which I think can change, for example, in the event that the device reboots and re-enters the zigbee network unlike the long address (Zigbee MAC Address) which is recorded in the silicon of the chip. when the device reboots the bind remains but said short address I think may vary

2) when the system has 3 types of device, Gateway, switch, and end device. and the binding between the Switch and the end device (such as a system of switches and lights) is established from the gateway (since this is allowed by the Zigbee 3.0 standard), Since the switch has not established a callback for external bindings, it would not have the mac address of the end device to which it was binded. This is a typical case in which the configuration arrives through the gateway (via Wi-Fi-Ethernet) and it configures the system and is in charge of carrying out the bindings.

3) Among the most used cases of the Zigbee 3.0 protocol is the handling of multiple ON-OFF signals wirelessly, typically a light switch that has 2-3 outputs has an endpoint for each of these outputs that in turn contains the ON-OFF cluster. This device could be connected by each of its endpoints to different light sensors. Although it could be obtained to which endpoint it is directed by means of the address, as you explained before, it would be much easier if the destination endpoint were sent in the callback. In this way, you would simply act on an output based on the received endpoint.

In these cases, I think it would be advisable to add the destination endpoint to the callback, not only to facilitate the programming logic, but also to provide the possibility of binding from a third device (typically a gateway).

As always, thanks for the quick response.

likunqiao097304, 2023-05-22:
@diazmanuel 
for the question 1, If the device joined before, reboots and rejoin, the device should keep the same short address as before. Also, in report callback argument type `esp_zb_zcl_addr_t` contains either short or long address.

for the question 2,  If you configure bind/report for the switch to a light through gateway, you are right, when switch get the report callback it doesn't know the destination endpoint. But my question is, is it necessary for switch to get the report information? Should it better to report light info back to gateway?

Again, from the technical point of view, adding this destination endpoint is not so difficult, but we'd like to know more about the real usage circumstances. Thanks

diazmanuel, 2023-05-22:
Regarding the utility, use the best known case to try to demonstrate it, perhaps it was not the best example. The case in which I work is a room air conditioning system, where there is a gateway, one or more zone controllers and a thermostat for each zone to be air conditioned. each zone controller has 6 endpoints which activate 6 valve outlets to heat the different environments. each thermostat in turn pairs with one of the endpoints and controls the room temperature. In principle, the ecosystem does design with Nordic chips. As in the previous case, it was necessary to be able to identify in the zone controller which endpoint the thermostat report was directed to.

The architecture of the system was designed in this way to be modularizable. Since if the reports from the termosattos were direct to the gateway, this would have to have a very high number of endpoints (in practical cases, more than 10 endpoints only from zones) and this was also a limitation in the Nordic chips that only allowed to have 8 endpoints per device. We are currently looking at migrating this system to Espressif chips. I understand that adding the destination endpoint to the callback should not result in a very big structural change since it internally identifies which endpoint the incoming report comes from, so I would like to request that it be added, I'm sure it will be useful for other complex systems like this .

closed 2023-05-26 by espressif-bot

chshu, 2023-05-26:
@diazmanuel The new report callback API supports the destination endpoint now: https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_core.h#LL69C6-L69C6