espressif/esp-zigbee-sdk #24: [Feature request] increase the allowed size of the maximum and minimum increase interval report

diazmanuel, 2023-05-18:
Hello, could you change the data type of the max interval and min interval within the report config to uin16_t as the zigbee 3.0 standard says?
[image]

Currently the data type is set as uint8_t 
https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/zcl/esp_zigbee_zcl_command.html?highlight=esp_zb_zcl_config_report_cmd_req#_CPPv430esp_zb_zcl_config_report_cmd_s

likunqiao097304, 2023-05-19:
Thanks, we will do this update

diazmanuel, 2023-05-19:
Thanks, is there a roadmap or a log of the changes that are being implemented in each update? I would like to keep abreast of the changes that are being added.

likunqiao097304, 2023-05-22:
@diazmanuel Currently no road map, but we have noticed this need from many users, we will add this later. Thanks for your advice

closed 2023-05-26 by espressif-bot