espressif/esp-zigbee-sdk #27: strange behavior in reports

diazmanuel, 2023-05-19:
I am currently working on the following system. A gateway that acts as a zigbee network coordinator, a device (switch) that handles multiple outputs, and a sensor device for each of these outputs. The coordinator is in charge of performing the binding between the different endpoints of the switch and the sensors. and the switch establishes a binding between itself and the gateway to report a custom cluster each time an output is activated from the sensor

When analyzing the system, I found an error in the behavior of the switch reports to the gateway. After binding the switch to the gateway, the reports are configured as follows

static void bind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{

    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "bind_cb status:%d", zdo_status);
        /* configure report attribute command */
        esp_zb_zcl_config_report_cmd_t report_cmd;
        report_cmd.zcl_basic_cmd.dst_addr_u.addr_short=esp_zb_get_short_address();
        report_cmd.zcl_basic_cmd.dst_endpoint = 1;
        report_cmd.zcl_basic_cmd.src_endpoint = 1;
        report_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        report_cmd.clusterID = 0xFFFE;
        report_cmd.attributeID = 0;
        report_cmd.attrType = ESP_ZB_ZCL_ATTR_TYPE_U32;
        report_cmd.min_interval = 0;
        report_cmd.max_interval = 30;
        report_cmd.reportable_change =3;
        esp_zb_zcl_config_report_cmd_req(&report_cmd);
    }
}

and the callback of the sensor device report is handle as follow:

static void esp_zb_dev_reporting_cb(esp_zb_zcl_addr_t *addr, uint8_t endpoint, uint16_t cluster_id, uint16_t attr_id, esp_zb_zcl_attr_type_t attr_type, void *value)
{

    esp_zb_zcl_attr_t *value_r = esp_zb_zcl_get_attribute(1,0xFFFE,ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,0);
    *((uint32_t*) value_r->data_p)+=1;
    esp_zb_zcl_set_attribute_val(1,0xFFFE,ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,0,value_r->data_p);
}

then the expected behavior of the system would be that the switch reports to the gateway every 30 seconds and when the switch receives a report from the sensor it increases the value of attribute 0 of the custom cluster 0XFFFE by 1, if the value increases exceeds what is established in the reports configuration (in this case 3) a new report should be produced from the switch to the gateway.

The real behavior was the following, in principle the switch reported every 30 seconds to the gateway as expected, but upon receiving the command from a sensor the callback of said report was triggered and increased the custom attribute, in theory the switch should not report to the gateway since the increase of said attribute was less than 3 compared to its original value but contrary to what was expected, the switch generated a report to the gateway ignoring the configuration condition (also verify that this report did not correspond to the periodic report every 30 seconds). In addition to having generated the report, it would seem that he deleted the configuration because from that moment on, periodic reports were no longer generated, nor were reports generated when modifying that attribute of the customized cluster again.

after that i try to force the reports after writing the attribute ,by modifying the callback of the incoming reports from the sensors as follows.

static void esp_zb_dev_reporting_cb(esp_zb_zcl_addr_t *addr, uint8_t endpoint, uint16_t cluster_id, uint16_t attr_id, esp_zb_zcl_attr_type_t attr_type, void *value)

{

    esp_zb_zcl_report_attr_cmd_t des;
    des.zcl_basic_cmd.dst_addr_u.addr_short=0;   // gateway
    des.zcl_basic_cmd.dst_endpoint = 1;
    des.zcl_basic_cmd.src_endpoint = 1;
    des.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    des.clusterID = 0xFFFE;
    des.attributeID = 0;
    des.cluster_role=ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
    esp_zb_zcl_attr_t *value_r = esp_zb_zcl_get_attribute(1,0xFFFE,ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,0);
    *((uint32_t*) value_r->data_p)+=1;
    esp_zb_zcl_set_attribute_val(1,0xFFFE,ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,0,value_r->data_p);
    esp_zb_zcl_report_attr_cmd_req(&des);
}
The behavior was similar to the previous one, only this time every time a sensor report arrived, the switch modified the attribute and when forcing the report it was sent, but as in the previous case, the established configuration of the periodic report was ignored and the minimum value that said report will generate

I think writing the custom cluster attribute ignores and clears the report settings. I would like to know if this can be due to a bad behavior of the sdk or if I am doing the configuration wrong

thanks

likunqiao097304, 2023-05-22:
Since you didn't get the period 30 seconds report and also didn't apply the rule of reportable change. It might be you didn't configure the report or bind correctly. 
`    report_cmd.zcl_basic_cmd.dst_addr_u.addr_short=esp_zb_get_short_address();` should this be the switch? should it be sending the configure report command from gateway to switch?

diazmanuel, 2023-05-22:
Maybe I am not explaining myself correctly, the code I show is the switch code, therefore esp_zb_get_short_address() obtains the switch address. It is also configured correctly because in principle periodic reports are sent, when modifying the attribute with the esp_zb_zcl_set_attribute_val function the report stops working

likunqiao097304, 2023-05-29:
@diazmanuel Sorry for the late following up. If I understand correctly, you would like to set the configure report to switch and ask switch to send the report to gateway in period of 30 seconds or attribute change larger than 3. However, you got a report command which is less than 3 delta value nor in 30 seconds period first time, and later all future expected report command is stopped. Right?

Firstly, for the work around solution, you could use `esp_zb_zcl_report_attr_cmd_req` to directly call this function send report to gateway, with a timer alarm 30 seconds (`esp_zb_scheduler_alarm`) or based on the attribute change more than 3 to trigger.

Secondly, could you show me your code how do you register the custom attribute on switch side and how do you do the binding from switch to gateway, I'd like to see if there is any other issue. It might be some issue when we set `esp_zb_zcl_set_attribute_val` on customized attribute, I will try to duplicate your condition to fix it.

diazmanuel, 2023-05-29:
_If I understand correctly, you would like to set the configure report to switch and ask switch to send the report to gateway in period of 30 seconds or attribute change larger than 3._ 
yes thats right but its stop working when i set that attribut. i change the increment the attribute by 1 and then the reports stop working. i will add the code so its more simple to understand.

here is the code of the switch

/*
 * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 *
 * Zigbee customized client Example
 *
 * This example code is in the Public Domain (or CC0 licensed, at your option.)
 *
 * Unless required by applicable law or agreed to in writing, this
 * software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
 * CONDITIONS OF ANY KIND, either express or implied.
 */
#include <string.h>
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_HA_customized_switch.h"
#include "nvs_flash.h"
/**
 * @note Make sure set idf.py menuconfig in zigbee component as zigbee end device!
*/
#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile light switch (End Device) source code.
#endif

static const char *TAG = "ESP_HA_ON_OFF_SWITCH";

struct cluster_data_t cluster_data;

typedef struct zdo_info_ctx_s {
    uint8_t endpoint;
    esp_zb_ieee_addr_t long_addr;
} zdo_info_user_ctx_t;

static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_ERROR_CHECK(esp_zb_bdb_start_top_level_commissioning(mode_mask));
}

static void bind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "bind_cb status:%d", zdo_status);
        /* configure report attribute command */
        esp_zb_zcl_config_report_cmd_t report_cmd;
        report_cmd.zcl_basic_cmd.dst_addr_u.addr_short=esp_zb_get_short_address();
        report_cmd.zcl_basic_cmd.dst_endpoint = 1;
        report_cmd.zcl_basic_cmd.src_endpoint = 1;
        report_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        report_cmd.clusterID = 0xFFFE;
        report_cmd.attributeID = 0;
        report_cmd.attrType = ESP_ZB_ZCL_ATTR_TYPE_U32;
        report_cmd.min_interval = 1;
        report_cmd.max_interval = 30;
        report_cmd.reportable_change = 1;
        esp_zb_zcl_config_report_cmd_req(&report_cmd);
    }
}
static void esp_zb_dev_reporting_cb(esp_zb_zcl_addr_t *addr, uint8_t endpoint, uint16_t cluster_id,
                                    uint16_t attr_id, esp_zb_zcl_attr_type_t attr_type, void *value)
{
    esp_zb_zcl_report_attr_cmd_t des;
    des.zcl_basic_cmd.dst_addr_u.addr_short=0;
    des.zcl_basic_cmd.dst_endpoint = 1;
[... 157 lines cut ...]

    //esp_zb_add_custom_cluster_command_cb(1,0xFFFE,attr_cb);
    //esp_zb_device_add_set_attr_value_cb(attr_cbxxx);
    esp_zb_device_add_report_attr_cb(esp_zb_dev_reporting_cb);
    esp_zb_add_read_attr_resp_cb(1, esp_zb_read_resp_cb);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    /* hardware related and device init */
    //switch_driver_init(button_func_pair, PAIR_SIZE(button_func_pair), esp_zb_buttons_handler);
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}

In addition, the gateway makes a bind between the end device and the switch. when the end device reports to the switch, it increases the attribute and generates a report to the gateway

likunqiao097304, 2023-06-02:
@diazmanuel Hi, I have tried to duplicate most of your code on my side. And here are some results.

1.  I used level control cluster's attribute to do binding and config report. It worked fine. It can send the report to remote device in period of max interval seconds or do `esp_zb_zcl_set_attribute_val` attribute change larger than delta_change value. 

2. I also tried to use the same customized cluster (0xfffe) with attribute (0x0) you created to do binding and config report. It could periodically send the report message (let's say 30 seconds). I also **couldn't** receive a report if attribute value is larger than the delta_change by using `esp_zb_zcl_set_attribute_val`. **However**, even I couldn't send the report by using `esp_zb_zcl_set_attribute_val`, I still can get the period report, it didn't stop. Yes, there is a bug, but strange behavior is different than yours.

3. The bug is that our structure defines `esp_zb_zcl_config_report_cmd_t` `reportable_change` as uint16_t. It should varies depends on the `attrType`. The fixes will be release either today or next week.

Again, for different behavior that you saw device stop reporting after `esp_zb_zcl_set_attribute_val`. It might be binding issue not report configuration.

closed 2023-06-05 by espressif-bot

likunqiao097304, 2023-06-05:
@diazmanuel The latest release has been fixed the reportable_change type from uint16_t to void type. Please try it out.

diazmanuel, 2023-06-05:
@likunqiao097304 with this change the strange behavior that it had disappeared, now it works correctly thanks