espressif/esp-zigbee-sdk #30: How to handle leave request from coordinator?

Tropaion, 2023-05-30:
Hello, me again :D

I'm trying to handle the case when the coordinator sends a leave request to my end device.
I managed to find out that there isn't a explicit signal for it but the signal ```ESP_ZB_ZDO_SIGNAL_LEAVE``` is triggered.
But it looks like there isn't a fitting leave type or other parameter defined for this case.
When I try to read the leave type in this case, I get the value "306905088".
Is there another way to identify this case?

xieqinan, 2023-05-31:
Hello @Tropaion ,
Can you tell me the commit ID for your esp-zigbee-sdk? The `leave_type` value has been fixed in commit `503dce9f2`.

Tropaion, 2023-05-31:
How do I get the commit ID? I'm using the VS Code Plugin.

xieqinan, 2023-05-31:
You can modify the `idf_component.yml` file to obtain the latest versions of `esp-zigbee-lib` and `esp-zboss-lib`, which can be found in the main folder of every example. The most recent version, as of May 31, 2023, is as follows:

```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp-zboss-lib: "~0.5.0"
  espressif/esp-zigbee-lib: "~0.6.0"
  ## Required IDF version
  idf:
    version: ">=5.0.0"

```

Tropaion, 2023-05-31:
@xieqinan Thanks, I suspected something like that.
I tried to look into what the ```~``` is for, but didn't find anything in docu so I changed my file to:
```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp-zigbee-lib: ">=0.5.0"
  espressif/esp-zboss-lib: ">=0.3.0"
  espressif/button: ">=2.5.1"
  ## Required IDF version
  idf:
    version: ">=5.0.0"
```
I thought with this it automatically always uses the newest version, but appearently it does not.
Is there a way to do this?

Tropaion, 2023-05-31:
Ok, thank you.
I updated to the version you suggested but now I get an error I think I cannot solve:
```
[7/9] Linking CXX executable ZB_HTE501.elf
FAILED: ZB_HTE501.elf 
cmd.exe /C "cd . && C:\Users\Tropaion\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imac_zicsr_zifencei -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C6=0 -Wl,--Map=C:/Users/Tropaion/Desktop/ZB_ESP32_dev/build/ZB_HTE501.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32c6.peripherals.ld -T esp32c6.rom.ld -T esp32c6.rom.api.ld -T esp32c6.rom.rvfp.ld -T esp32c6.rom.newlib.ld -T esp32c6.rom.version.ld -T esp32c6.rom.phy.ld -T esp32c6.rom.coexist.ld -T esp32c6.rom.net80211.ld -T esp32c6.rom.pp.ld -T esp32c6.rom.wdt.ld -T esp32c6.rom.newlib-normal.ld -T esp32c6.rom.heap.ld -T memory.ld -T sections.ld @CMakeFiles\ZB_HTE501.elf.rsp -o ZB_HTE501.elf  && cd ."
c:/users/tropaion/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/Tropaion/Desktop/ZB_ESP32_dev/managed_components/espressif__esp-zboss-lib/lib/esp32c6/libzboss_port.a(zb_esp_mac.c.obj): in function `.L0 ':
(.text.zb_transceiver_set_channel+0x4): undefined reference to `esp_ieee802154_set_channel'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

likunqiao097304, 2023-06-01:
@Tropaion You might need to update your esp-idf, it seems like the ieee802154 components version mismatch

Tropaion, 2023-06-01:
@likunqiao097304 I will try, thanks you.

chshu, 2023-06-15:
@Tropaion  Closing, please reopen if any other follow question.

closed 2023-06-15 by chshu

Tropaion, 2023-07-24:
> Hello @Tropaion , Can you tell me the commit ID for your esp-zigbee-sdk? The `leave_type` value has been fixed in [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk/commit/503dce9f20df5a437843f1de489ff11eb4cde40a).

With the latest version, I again looked into the leave type but there are only two types:
```
typedef enum {
    ESP_ZB_NWK_LEAVE_TYPE_RESET                 = 0x00, /*!< Leave without rejoin */
    ESP_ZB_NWK_LEAVE_TYPE_REJOIN                = 0x01  /*!< Leave with rejoin */
} esp_zb_nwk_leave_type_t;
```
Not sure why I would need leave with rejoin, but still, how do I handle the leave request from the coordinator?