espressif/esp-zigbee-sdk #32: queries about the operation of the zigbee stack

diazmanuel, 2023-06-07:
I understand that the zigbee stack is usually a problem since once it fills up, problems begin to occur

Looking at the examples, notice that each time the device is restarted, the bindings are created again. One of the queries is if, when creating these bindings, an internal check is made to see if said binding already existed.

If this check is not done, is there any method to delete the different components of the stack such as bindings, report configurations and the network to which it was connected, or to know the status of the zigbee stack?

I have noticed some strange behaviors that were solved by erasing the flash memory of the chip, maybe it is due to the zigbee stack

I take the opportunity to make some other queries, from time to time I receive signal 50 (ESP_ZB_NLME_STATUS_INDICATION ) in the connection handler but I don't know what it means exactly, after this signal I receive a signal to reconnect to the network and everything starts to work correctly.

Finally, I would like to know if it is possible to declare a cluster as client and server simultaneously, I tried implementing it with an or between the role definitions but it does not work correctly, then I declared it twice on the same endpoint (one as server and the other as client) but the reports do not work correctly, specifically it has problems with the periodic reports, if more information is needed for this case I can upload the code, I am working with a system that has at least 8 endpoints which could also be the cause of this malfunction

Thank you in advance and sorry for mixing several topics in the same issue

likunqiao097304, 2023-06-12:
> If this check is not done, is there any method to delete the different components of the stack such as bindings, report configurations and the network to which it was connected, or to know the status of the zigbee stack

Are you looking for something like `esp_zb_factory_reset` to remove history data or `esp_zb_nvram_erase_at_start` to erase nvram data which contains those information every time you restart?

> I have noticed some strange behaviors that were solved by erasing the flash memory of the chip, maybe it is due to the zigbee stack

Could you specify what exactly strange behaviors you saw?

> I would like to know if it is possible to declare a cluster as client and server simultaneously

You'd like to have a one API to declare both server and client at one endpoint? For the issue about the report, could you give us more detail information about how is your network topology and binding, reporting set up? Also it would be nice to share us some code, We will try to solve it together

diazmanuel, 2023-06-13:
@likunqiao097304 
The first part comes as a question of the bindings table, in the zboss library at a certain point the table in charge of the bindings was filled, for example every time the device was restarted, the binding table was filled with the same bind. For this, the zboss library allowed you to read said table to check if the binding you were wanting to enter already existed, and in that case avoid adding it again.

Regarding the strange behavior, I could not replicate it because once the flash was deleted and the device was rewritten, it began to work correctly, but what I could notice was that the reports stopped working, and also it did not allow me to add more than one cluster to the same endpoint. the chip was directly restarted by the simple declaration of the cluster, below I leave the code in case you want to analyze it

what i mean by declaring a cluster as both server and client is as bellow
this way 
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, FFFF, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE );
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, FFFF, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
or this
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, FFFF, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE | ESP_ZB_ZCL_CLUSTER_SERVER_ROLE );

The purpose of this is that this device we call B receives a report from A to said cluster and it must replicate the report to C. This is so because C must be sure that the report from A reached B correctly (this is important since otherwise A could report to both B and C)

The topology is as follows: there is a gateway, a control device with multiple outputs (let's call it Switch), up to 6 thermostats and 1 boiler. The code corresponds to the switch.
So the thermostats report through the cluster on off to the different endpoints of the switch and this activates the outputs according to which thermostat makes the report. If there is an output activated, the Switch reports to the boiler through the on-off cluster so that it is activated. In turn, if the boiler detects an error, it reports said error to the switch through a custom cluster (FFFF) and the switch in turn reports to each of the thermostats through said cluster. In parallel each status of the outputs and the status of the boiler are reported from the switch to the gateway. here is the code of the switch:
```
#include "taskZigbee.h"

static const char *TAG = "taskZigbee";

struct cluster_data_t cluster_data;

typedef struct zdo_info_ctx_s {
    uint8_t src_endp;
    uint16_t cluster_id;
    //esp_zb_ieee_addr_t long_addr;
} zdo_info_user_ctx_t;

static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_ERROR_CHECK(esp_zb_bdb_start_top_level_commissioning(mode_mask));
}

static void bind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "bind_cb status:%d, clsuter %d, src_endp %d", zdo_status,((zdo_info_user_ctx_t *)user_ctx)->cluster_id,((zdo_info_user_ctx_t *)user_ctx)->src_endp);
        /* configure report attribute command */
        static esp_zb_zcl_config_report_cmd_t report_cmd;
        static esp_zb_zcl_config_report_cmd_t report_cmd_2;
        if(((zdo_info_user_ctx_t *)user_ctx)->cluster_id == 0xFFFF){
            int16_t report_change=1;
            report_cmd.zcl_basic_cmd.dst_addr_u.addr_short=esp_zb_get_short_address();
            report_cmd.zcl_basic_cmd.dst_endpoint = 1;
            report_cmd.zcl_basic_cmd.src_endpoint = ((zdo_info_user_ctx_t *)user_ctx)->src_endp;
            report_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
            report_cmd.clusterID = ((zdo_info_user_ctx_t *)user_ctx)->cluster_id;
            report_cmd.attributeID = 0x05;
            report_cmd.attrType = ESP_ZB_ZCL_ATTR_TYPE_S16;
            report_cmd.min_interval = 0;
            report_cmd.max_interval = 300;
            report_cmd.reportable_change = (void*)&report_change;
            esp_zb_zcl_config_report_cmd_req(&report_cmd);
        }
        if(((zdo_info_user_ctx_t *)user_ctx)->cluster_id == 0x0006){
            bool report_change_2=1;
            report_cmd_2.zcl_basic_cmd.dst_addr_u.addr_short=esp_zb_get_short_address();
            report_cmd_2.zcl_basic_cmd.dst_endpoint = 1;
            report_cmd_2.zcl_basic_cmd.src_endpoint = ((zdo_info_user_ctx_t *)user_ctx)->src_endp;
            ESP_LOGI(TAG,"%d    %d",((zdo_info_user_ctx_t *)user_ctx)->cluster_id,((zdo_info_user_ctx_t *)user_ctx)->src_endp);
            report_cmd_2.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
            report_cmd_2.clusterID = ((zdo_info_user_ctx_t *)user_ctx)->cluster_id;
            report_cmd_2.attributeID = ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID;
            report_cmd_2.attrType = ESP_ZB_ZCL_ATTR_TYPE_BOOL;
            report_cmd_2.min_interval = 0;
            report_cmd_2.max_interval = 0xFFFF;
            report_cmd_2.reportable_change = (void*)&report_change_2;
            esp_zb_zcl_config_report_cmd_req(&report_cmd_2);
        }
[... 166 lines cut ...]
    esp_zb_device_add_set_attr_value_cb(attr_cb);
    esp_zb_device_add_report_attr_cb(esp_zb_dev_reporting_cb);

    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
    
}
uint8_t  InitTaskZigbee(){

esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    if(xTaskCreate(ZigbeeHandlertTask, "zigbeeHandlertTask", ZIGBEE_HANDLER_TASK_SIZE, NULL, ZIGBEE_HANDLER_TASK_PRIORITY, NULL)!=pdPASS) return false;

    ESP_LOGI(TAG, "Zigbee Handler Task Created");

    return true;

}

```

xieqinan, 2023-07-21:
Hi,
Let's start by discussing the bind query issue first. If you need the binding table request feature, we can provide it. However, please note that the cost between the bind request and the binding table request may be similar.

> but what I could notice was that the reports stopped working, and also it did not allow me to add more than one cluster to the same endpoint.

Could you please provide the log of the chip related to the issue? I have executed your code on my end, and I haven't encountered any problems. Have you made any new discoveries or observations regarding this matter?

> esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, FFFF, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE );
> esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, FFFF, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

The way is right.

diazmanuel, 2023-07-31:
hi, @xieqinan,
I am interested in bring the bindings table to confirm if there are duplicates, that is, if a binding was created 2 times, since the binding is executed every time the device is restarted. In case duplicates are generated, I would use this table to decide whether or not to create the binding (if it already exists, I would not create it again)

I think I found the problem, although it is a bit difficult to explain, if in the first instance I created an endpoint with a single client and then re-write the chip but this time in that same endpoint number I created a client and a server, then the chip was rebooted, I think the reboot occurred because it was trying to create an endpoint that already existed, for example if you try to create an endpoint that already exists by calling the following line 2 times it produces the same reboot
```
esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, ZIGBEE_BOILER_ENDPOINT, 260, ESP_ZB_HA_CUSTOM_ATTR_DEVICE_ID);
esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, ZIGBEE_BOILER_ENDPOINT, 260, ESP_ZB_HA_CUSTOM_ATTR_DEVICE_ID);

```
In summary I think I was not cleaning the endpoint before recording the same endpoint with a different configuration, anyway this problem was fixed after cleaning the memory so i think its not a big issue

xieqinan, 2023-08-01:
> I am interested in bring the bindings table to confirm if there are duplicates,

The feature has already been added in [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zdo/esp_zigbee_zdo_command.h#L399C1-L399C5)

> In summary I think I was not cleaning the endpoint before recording the same endpoint with a different configuration

Good job!

closed 2023-08-01 by diazmanuel