espressif/esp-zigbee-sdk #33: [Feature request] add external binding callback

diazmanuel, 2023-06-09:
Hello!

I am working with a system in which the binding between 2 devices is established from a third device, to configure the reports of this binding I would need a callback that is called when a binding is done from a third party.

I take this opportunity to see if it is possible to configure the reports before performing the binding, that is, before knowing the destination of the report. One of the parameters required by the report configuration structure is the destination address that I did not have before the binding (since the binding is done from a third device). If the rest of the report parameters are configured, when binding is done the destination address configuration is automatically edited? Because if I don't configure the report and do the binding, the reports are sent with some default configuration

Thanks

likunqiao097304, 2023-06-12:
@diazmanuel Let me understand your question clearly. 
The first question is that you need a callback for the 2 devices when third device binding is done?
Second, you'd like to see configure report device before binding has done?

> Because if I don't configure the report and do the binding, the reports are sent with some default configuration

If you only do binding to a report attribute,yes, there is a default configure report setting.

diazmanuel, 2023-06-13:
@likunqiao097304 
For the first one yes, but it is for 1 device, for example in my case I have 1 gateway, 1 switch and 1 thermostat, so I configure the link for the thermostat to report to the switch, and this binding is only sent to the one that generates the report ( the thermostat), it is within the thermostat that I need to be able to register a callback that notifies me of this new binding in order to configure the reports and other things.

As for the second, it is actually related to the first, since I don't have a callback of a binding that is configured from a third party, I have no way to configure the reports of said binding since I don't have the address of the device to which I have to report. In the examples, the bindings are always done from the same device, so the destination address of the report is already available.

diazmanuel, 2023-07-04:
hello @likunqiao097304 will you add this feature? if yes may i ask when it will be ready?

kelin6, 2023-07-20:
@diazmanuel Based on our esp-zigbee-sdk, I tried to reproduce your setup. please take a look the  logs: [esp_zigbee_bind.zip](https://github.com/espressif/esp-zigbee-sdk/files/12101643/esp_zigbee_bind.zip)

Setup decription:
1) ZC
  - aa:aa:aa:aa:aa:aa:aa:aa
  - doesn't have clusters.
2) ZR (light)
  - 11:11:11:11:11:11:11:11
  - joins to ZC
  - clusters list: Identify, Basic, Groups, Scenes, On_off (server).
3) ZED (swtich)
  - 22:22:22:22:22:22:22:22
  - joins to ZR
  - clusters list : Identify, Basic, Groups, Scenes,  On_off(client)
  
ZC configures:
1) ZC send bind req to ZR (light), and the bind between ZR (source) and ZED (destination) for the on_off cluster;
2) ZC send config_report to ZR (light), and config the report on_off attribute to ZED(switch). THe config report direction field is 0x00, so the report attribute to each destination as resolved by the bindings, so you do not need a register a callback that notifies you of this new binding in order to configure the reports.
[image]

diazmanuel, 2023-07-31:
hi @kelin6 thanks for the answer
I understand that the solution you propose is that the same coordinator sends the configuration after sending the binding, using the same destination and source as the binding request. I will try to implement this solution, anyway it is very complex to add the callback that I proposed?

I take this opportunity to see if there is any way to modify the default report configuration (minimum delta to make the report and the maximum and minimum time parameters)

kelin6, 2023-08-07:
@diazmanuel The suggestion that you proposed to add a callback is not complex, but I think the callback is not  conventional, so if you want to do this callback, it need catch all the APS (Application Support Sublayer) data packets from zigbee stack and also this callback must not process packets in standard ZB profiles. Anyway, this callback is not routine.

chshu, 2023-09-05:
@diazmanuel Closing the issue, let us know if any follow up quesiton.

closed 2023-09-05 by chshu