espressif/esp-zigbee-sdk #34: esp_zb_zcl_set_attribute_val crash

xmow49, 2023-06-11:
Hi,
Im looking for sending an attibut state when I push a button. I tested `esp_zb_zcl_set_attribute_val` saw in #3. But I have a crash when I call it: 

`esp_zb_zcl_set_attribute_val(HA_ESP_SMOKE_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_IAS_ZONE_ZONESTATUS_ID, &lastState);
` 

```
I (538) main_task: Returned from app_main()
I (548) ESP_ZB_ON_OFF_LIGHT: ZDO signal: 23, status: -1
I (548) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (1168) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (1178) ESP_ZB_ON_OFF_LIGHT: Joined network successfully (Extended PAN ID: 29:dc:3c:60:9e:ab:a4:f5, PAN ID: 0x1e78, Channel:15)
Button pressed
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42027760  RA      : 0x42027760  SP      : 0x40819de0  GP      : 0x4080e0d0
0x42027760: zb_zcl_ias_zone_write_attr_hook_server at ??:?

0x42027760: zb_zcl_ias_zone_write_attr_hook_server at ??:?

TP      : 0x4080d4d0  T0      : 0x00000000  T1      : 0x4087ee0b  T2      : 0x00000000
0x4080d4d0: spi_flash_chip_generic_read at C:/Espressif/ESP-IDF/esp-idf/components/spi_flash/spi_flash_chip_generic.c:256

S0/FP   : 0x00000002  S1      : 0x40819e6c  A0      : 0x00000000  A1      : 0x0000effe
A2      : 0x0000ffff  A3      : 0x0000ffff  A4      : 0x00000007  A5      : 0x00000007
A6      : 0x00000000  A7      : 0x00000007  S2      : 0x0000000a  S3      : 0x0000ffff
S4      : 0x00000500  S5      : 0x00000002  S6      : 0x00000001  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x0000000a  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000006
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
40819de0: 0x40819e6c 0x00000002 0x0000000a 0x0000ffff 0x40819e6c 0x00000002 0x0000000a 0x42038006
0x42038006: zb_zcl_write_attr_hook at ??:?

40819e00: 0x40819e6c 0x00000000 0x4087ed26 0x0000000a 0x40819e6c 0x00000000 0x4087ed26 0x42024fbc
0x42024fbc: zb_zcl_set_attr_val at ??:?

40819e20: 0x00000000 0x4087ef04 0x40815eb0 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40819e40: 0x00000000 0x00000000 0x00000000 0x42007650 0x00000000 0x00000000 0x00000000 0x420070e8
0x42007650: esp_zb_zcl_set_attribute_val at ??:?

0x420070e8: buttonLoop at C:/Espressif/Workspace/TestSmoke/main/esp_zb_light.c:162

40819e60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4080a398
0x4080a398: vPortTaskWrapper at C:/Espressif/ESP-IDF/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205

40819e80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40819ea0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40819ec0: 0x00062749 0x40815484 0x40815484 0x596f2963 0xaef26566 0xaf58bad1 0xa140e0fd 0x023dfd67
40819ee0: 0xe608483b 0xf6ee7ac1 0x7780eb72 0x78d3fc96 0x29ff8574 0x967eff11 0xcfdb4f75 0xc817a76c
40819f00: 0xfb5d1c8b 0x98ed8439 0x8a95157b 0x687c0c2a 0x2f12d7b4 0xed6fd6b6 0xca545a7a 0xd9f32281
40819f20: 0x05de8c77 0x83327d34 0xde405c28 0x823c62f6 0x0b1a3efa 0x4a6e4b6d 0x08e6bad0 0x36498bb1
40819f40: 0xe7c84098 0xbb98d517 0xc8f5b453 0x1fa580e3 0x2ab21492 0x9e41aaab 0xc654b37d 0x53342cb7
40819f60: 0xedeb6c8c 0xb033ca47 0x435cbbb7 0x81b87591 0x1e903953 0xb19bd847 0xce28d3d9 0xab1f082f
40819f80: 0x2b820dd6 0x221b0f4e 0x2db1be03 0xf0130f21 0x0019f14b 0xa2891260 0x631ca339 0x204c0d95
40819fa0: 0xdfff79e7 0x280137e2 0x9c7e1e1b 0x4b9a1b81 0x6b6f1075 0xb33af362 0xce89c48c 0x529d4236
40819fc0: 0x249666a5 0x6b27ce9c 0x081db00a 0x0ab6ae42 0x659d9030 0x7e4ac771 0xbe273185 0x369f4ce3
40819fe0: 0xe02bb838 0x22cf7402 0x00c16998 0x35e9b8d2 0x3acf7870 0x7f973024 0x8f0dab4d 0xfc820f4a
4081a000: 0x59c2ec27 0xbb252acf 0xb9ea2867 0xc16801da 0x4663d0ef 0x645aae61 0x863fb78d 0x21705e89
4081a020: 0x87cf356c 0x92e65bed 0x5f351726 0x6a4c5e3d 0x0be96f56 0x6ce0c897 0xf00677a6 0x0dde4827
4081a040: 0x2ca6c3e3 0x51c96dfe 0x77bdfca0 0x9b443725 0x62df93d9 0x6eabb356 0x103ceb64 0x94f2c95d
4081a060: 0x08c0bfae 0x8cd01b93 0x4d71a954 0x8b6b9344 0xa7b820ea 0x7bd4adb9 0x4095b2cd 0x8a1532e9
4081a080: 0x7eaffb73 0x1c4121d6 0xcb017c66 0xa0506b37 0x2e88ae3c 0x4dece525 0x468f25bb 0xfa1835f1
4081a0a0: 0x3bc218f7 0x6e537b36 0x61870bc8 0xa7df41f3 0x2b8a87e7 0xdb43586a 0xeb709036 0xde1e1f89
4081a0c0: 0x445f3a7b 0x246582a1 0x83e0eeb7 0xe0afae46 0x81ecd584 0x37dd3d61 0xe4e2941e 0xdd2b95e7
[... 179 lines cut ...]
            }
        }
        vTaskDelay(100 / portTICK_PERIOD_MS);
    }
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    /* load Zigbee light_bulb platform config to initialization */
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    /* hardware related and device init */
    light_driver_init(LIGHT_DEFAULT_OFF);
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
    xTaskCreate(buttonLoop, "buttonLoop", 4096, NULL, 5, NULL);
}
```

Do you have any idea what im doing wrong ?
Can I have an example to use this function ?
Thanks

xieqinan, 2023-06-12:
Hi @xmow49,
The `esp_zb_zcl_set_attribute_val()` function allows you to set the attribute value of a ZCL cluster. However, it does not check the attribute's access properties, so it is important for users to ensure its safety. 

In the case of the attribute ID `ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE` of the IAS Zone cluster, it only supports read access according to the specification. This limitation can cause issues with `zb_zcl_write_attr_hook()`. A better implementation of `esp_zb_zcl_set_attribute_val()` will be released in the next version.

If you are certain that writing to the `ESP_ZB_ZCL_ATTR_IAS_ZONE_ZONESTATUS_ID` attribute will not have any adverse effects on your project, the following workaround may be helpful for you.
```
 *((uint16_t *)esp_zb_zcl_get_attribute(HA_ESP_SMOKE_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_IAS_ZONE_ZONESTATUS_ID)->data_p) = lastState;
```
Note: the `ESP_ZB_ZCL_ATTR_IAS_ZONE_ZONESTATUS_ID` is map16(uint16_t) type rather than unint32_t;

xmow49, 2023-06-12:
Hello, Thanks for the quick response.
Its work, the value change correctly. But on my Home Assistant, the value dont change. If I mannualy read attributes, the values changed correctly. I tested to add this to notify the server:
```c
esp_zb_zcl_ias_zone_resp_cmd_t resp = ESP_ZB_ZCL_CMD_IAS_ZONE_ZONE_STATUS_CHANGE_NOT_ID;
esp_zb_zcl_ias_zone_status_change_notif_cmd_req(&resp);
```
But the value isn't refresh.
How to report a new value to the coordinator ? Can you give me an example ?

 
> Note: the `ESP_ZB_ZCL_ATTR_IAS_ZONE_ZONESTATUS_ID` is map16(uint16_t) type rather than unint32_t;

Ok. but is it normal than in "esp_zigbee_type.h", zone_status is uint8_t ?? maybe an error
```c
typedef struct esp_zb_ias_zone_cluster_cfg_s {
    uint8_t  zone_state;                             /*!< zone state */
    uint16_t  zone_type;                             /*!< zone type */
    uint8_t   zone_status;                           /*!< zone type */
    esp_zb_ieee_addr_t ias_cie_addr;                 /*!< IAS CIE Address that commands generated by server SHALL be sent to */
    uint8_t   zone_id;                               /*!< zone id */
} esp_zb_ias_zone_cluster_cfg_t;
```

Last thing, when I send any command to the IAS Zone cluster, I have a crash.
Something missing to my init cluster code ? A callback I think ?
<details></summary>Crash repport</summary>

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42027742  RA      : 0x42027742  SP      : 0x4087e410  GP      : 0x4080e0d0  
0x42027742: check_cie_establishment at zcl_ias_zone_commands.c.obj:?

0x42027742: check_cie_establishment at zcl_ias_zone_commands.c.obj:?

TP      : 0x40871c40  T0      : 0x4200aaee  T1      : 0x4087ee0b  T2      : 0x00003a09  
0x4200aaee: zb_aps_hdr_parse at ??:?

S0/FP   : 0x00000006  S1      : 0x00000006  A0      : 0x00000000  A1      : 0x0000e000  
A2      : 0x0000ffff  A3      : 0x0000ffff  A4      : 0x00000007  A5      : 0x00000007  
A6      : 0x00000000  A7      : 0x00000007  S2      : 0x4081075c  S3      : 0x0000000a  
S4      : 0x00000009  S5      : 0x00000007  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000004  T4      : 0x0000001e  T5      : 0x385cfbff  T6      : 0xfee419df  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000006  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4087e410: 0x9a1458ea 0x000000bf 0x4087ea08 0x42027f06 0x00000000 0x00000000 0x0ad65b00 0x0500000a
0x42027f06: zb_zcl_ias_zone_init_test_mode_handler at zcl_ias_zone_commands.c.obj:?

4087e430: 0x00020104 0x000000f9 0x00000000 0x00000907 0x4081075c 0x00000000 0x0000003b 0x0000000a
4087e450: 0x4081075c 0x00000006 0x00000006 0x42028064 0x40880000 0x0f000304 0x00000005 0x4200e710
0x42028064: zb_zcl_process_ias_zone_specific_commands at ??:?

0x4200e710: ieee_search at zb_address.c.obj:?

4087e470: 0x00000000 0x00000000 0x00000000 0x0ad65b00 0x0500000a 0x00020104 0x000000f9 0x00000000
4087e490: 0x4081075c 0x00000006 0x00000006 0x42028124 0x4081075c 0x00000006 0x4081075c 0x42028bdc
0x42028124: zb_zcl_process_ias_zone_specific_commands_srv at ??:?

0x42028bdc: zb_zcl_handle_specific_commands at ??:?

4087e4b0: 0x4081075c 0x00000006 0x00000000 0x42028c2c 0x00000006 0x4087edd0 0x4081075c 0x42024cda
0x42028c2c: zb_zcl_handle at ??:?

0x42024cda: ep_process_zcl_cmd at zcl_common.c.obj:?

[... 28 lines cut ...]

4087e5b0: 0x00280000 0xffffff00 0xffffffff 0x000000ff 0x00000800 0x00000002 0x00060000 0x00000bb8
4087e5d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4080a398
0x4080a398: vPortTaskWrapper at C:/Espressif/ESP-IDF/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205

4087e5f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e610: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x4087e3e0 0x00017ceb
4087e630: 0x4080f3bc 0x4080f3bc 0x4087e628 0x4080f3b4 0x00000014 0x4087f0a0 0x4087f0a0 0x4087e628
4087e650: 0x00000000 0x00000005 0x4087d624 0x6267695a 0x6d5f6565 0x006e6961 0x00000000 0x00000000
4087e670: 0x4087e620 0x00000005 0x00000000 0x00000000 0x00000000 0x00000009 0x40815e48 0x40815eb0
4087e690: 0x40815f18 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4205187a
0x4205187a: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

4087e6b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e6d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e6f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e710: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e730: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e750: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e770: 0x00000000 0x00000000 0x40000000 0x00000054 0x00000000 0x4087e780 0x00000000 0x00000000
4087e790: 0x00000000 0x4087e798 0xffffffff 0x4087e798 0x4087e798 0x00000000 0x4087e7ac 0xffffffff
4087e7b0: 0x4087e7ac 0x4087e7ac 0x00000001 0x00000001 0x00000000 0x0000ffff 0x00000000 0xb33fffff
4087e7d0: 0x00000000 0x00000054 0x00000000 0x4087e7d8 0x00000000 0x00000000 0x00000000 0x4087e7f0
4087e7f0: 0xffffffff 0x4087e7f0 0x4087e7f0 0x00000000 0x4087e804 0xffffffff 0x4087e804 0x4087e804
</details>

likunqiao097304, 2023-06-12:
> Ok. but is it normal than in "esp_zigbee_type.h", zone_status is uint8_t ?? maybe an error

This is our bug, we will fix this. Thanks for the pointing out. For the command crash issue, we will try to duplicate from our side

likunqiao097304, 2023-06-12:
@xmow49 for the command you used. 
`void esp_zb_zcl_ias_zone_status_change_notif_cmd_req(esp_zb_zcl_ias_zone_status_change_notif_cmd_t *cmd_req);`
The command structure is this `esp_zb_zcl_ias_zone_status_change_notif_cmd_t` not  `esp_zb_zcl_ias_zone_resp_cmd_t`

xmow49, 2023-06-12:
Ok, my bad,
In `esp_zb_zcl_ias_zone_status_change_notif_cmd_t` there is `esp_zb_zcl_basic_cmd_t`: 
```c
typedef struct esp_zb_zcl_basic_cmd_s {
    esp_zb_addr_u dst_addr_u;                   /*!< Single short address or group address */
    uint8_t  dst_endpoint;                      /*!< Destination endpoint */
    uint8_t  src_endpoint;                      /*!< Source endpoint */
} esp_zb_zcl_basic_cmd_t;
```

I dont know what fill in this struct, the dst_addr is my coordinator, how to get them ?
Im litle bit lost, can you draft me a litle example please? 
Thanks

xmow49, 2023-06-12:
Ok, Thanks a lot, it works!!! 
There is a last thing, as mentiond earlier:
When I pair the ESP to my home assistant, The ZHA send command to the ESP. When the ESP receive commands on the IAS cluster, the ESP crash:
```
I (7640) ESP_ZB_ON_OFF_LIGHT: ZDO signal: 60, status: 0
I (7740) ESP_ZB_ON_OFF_LIGHT: ZDO signal: 60, status: 0
I (8090) ESP_ZB_ON_OFF_LIGHT: Joined network successfully (Extended PAN ID: 29:dc:3c:60:9e:ab:a4:f5, PAN ID: 0x1e78, Channel:15)
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420277c6  RA      : 0x420277c6  SP      : 0x4087e410  GP      : 0x4080e0d0  
0x420277c6: check_value_ias_zone_server at ??:?

0x420277c6: check_value_ias_zone_server at ??:?

TP      : 0x40871c40  T0      : 0x4200ab0e  T1      : 0x4087ee0b  T2      : 0x00001e78
0x4200ab0e: zb_aps_hdr_parse at ??:?

S0/FP   : 0x40811031  S1      : 0x0000000a  A0      : 0x00000000  A1      : 0x0000e000
A2      : 0x0000ffff  A3      : 0x0000ffff  A4      : 0x00000007  A5      : 0x00000007
A6      : 0x00000000  A7      : 0x00000007  S2      : 0x00000500  S3      : 0x00000001
S4      : 0x00000010  S5      : 0x40811031  S6      : 0x4081059b  S7      : 0x0000ffff
S8      : 0x4087ee0b  S9      : 0x00000010  S10     : 0x00000014  S11     : 0x00000001
T3      : 0x0000000a  T4      : 0x0000001d  T5      : 0x19df101e  T6      : 0x604b0000
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000006
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4087e410: 0x5344b267 0x0000000a 0xffffffbb 0x420380e2 0x00000000 0x00000001 0x40811039 0x40811054
0x420380e2: zb_zcl_check_attr_value_manuf at ??:?

4087e430: 0x00000000 0x4087ed30 0x4081102e 0x42024738 0x40811054 0x00000014 0x00000014 0x42026964
0x42024738: zb_zcl_check_attribute_writable at ??:?

0x42026964: zb_zcl_write_attr_handler at ??:?

4087e450: 0x00000001 0x00000003 0x00000002 0x42017d00 0x40880000 0x0f000304 0x00000019 0x40811031
0x42017d00: mac_traffic_dump_put at mac_dump.c.obj:?

4087e470: 0x00000000 0x0000ffff 0x00000012 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e490: 0x00000000 0x00000000 0x00000037 0x0000000a 0x40811054 0x00000014 0x00000014 0x420276ea
0x420276ea: zb_zcl_handle_general_commands at ??:?

4087e4b0: 0x40811054 0x00000014 0x00000000 0x42028c36 0x00000014 0x4087edd0 0x40811054 0x42024cfa
0x42028c36: zb_zcl_handle at ??:?

0x42024cfa: ep_process_zcl_cmd at zcl_common.c.obj:?

4087e4d0: 0x00000000 0x00000000 0x00000000 0x0a604b00 0x0500000a 0x00020104 0x000001ba 0x00000000
4087e4f0: 0x00000014 0x00000000 0x00000014 0x00000229 0x00000014 0x00000000 0x00000014 0x420253ce
0x420253ce: zb_zcl_process_parsed_zcl_cmd at ??:?

4087e510: 0x00000000 0x00000000 0x00000014 0x42025430 0x00000000 0x00000000 0x00000000 0x0a604b00
0x42025430: zb_zcl_process_device_command at ??:?

4087e530: 0x0500000a 0x00020104 0x000001ba 0x42010000 0x00000000 0x00000001 0x00000000 0x42014d88
0x42010000: zb_channel_page_channel_number_to_logical at ??:?

0x42014d88: zb_sched_loop_iteration at ??:?

4087e550: 0x00000000 0x420253da 0x00140000 0x423b3700 0x0000001e 0x00000002 0x00000000 0x40815000
0x420253da: zb_zcl_process_device_command at ??:?

4087e570: 0x4087ebe8 0x4087ec24 0x42007000 0x4202dbe4 0x4087ebe8 0x4087ec24 0x4087eda4 0x420095fe
0x42007000: esp_coex_common_semphr_give_wrapper at C:/Espressif/ESP-IDF/esp-idf/components/esp_coex/esp32c6/esp_coex_adapter.c:97

0x4202dbe4: zboss_main_loop_iteration at ??:?

[... 15 lines cut ...]
4087e630: 0x4080f3bc 0x4080f3bc 0x4087e628 0x4080f3b4 0x00000014 0x4087f0a0 0x4087f0a0 0x4087e628
4087e650: 0x00000000 0x00000005 0x4087d624 0x6267695a 0x6d5f6565 0x006e6961 0x00000000 0x00000000
4087e670: 0x4087e620 0x00000005 0x00000000 0x00000000 0x00000000 0x00000009 0x40815e48 0x40815eb0
4087e690: 0x40815f18 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4205189a
0x4205189a: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

4087e6b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e6d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e6f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e710: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e730: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e750: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e770: 0x00000000 0x00000000 0x40000000 0x00000054 0x00000000 0x4087e780 0x00000000 0x00000000
4087e790: 0x00000000 0x4087e798 0xffffffff 0x4087e798 0x4087e798 0x00000000 0x4087e7ac 0xffffffff
4087e7b0: 0x4087e7ac 0x4087e7ac 0x00000001 0x00000001 0x00000000 0x0000ffff 0x00000000 0xb33fffff
4087e7d0: 0x00000000 0x00000054 0x00000000 0x4087e7d8 0x00000000 0x00000000 0x00000000 0x4087e7f0
4087e7f0: 0xffffffff 0x4087e7f0 0x4087e7f0 0x00000000 0x4087e804 0xffffffff 0x4087e804 0x4087e804

ELF file SHA256: 3c4b6671be22234b

Rebooting...
```
Is it my code the problem ?
Is it necessary to configure a callback for the IAS commands?
Thanks a lot

likunqiao097304, 2023-06-13:
> When I pair the ESP to my home assistant, The ZHA send command to the ESP. When the ESP receive commands on the IAS cluster, the ESP crash:

could you tell us what command you are sending from ZHA? From the log, it seems like some attributes or handler has been registered successfully.

> Is it necessary to configure a callback for the IAS commands?

You don't necessary to need callback

xmow49, 2023-06-13:
The paring process work: 
[image]
[image]
[image]
But during the paring process, the esp crash and restart. After 1min, the device is successfully paired.
Logs are attached.
[logs.log](https://github.com/espressif/esp-zigbee-sdk/files/11734105/logs.log)

abdulrehmanee12, 2023-06-16:
@xmow49 could you also share what command you are sending from ZHA

xmow49, 2023-06-21:
During the parring process, I don't send anything. I don't know what command home-assistant send during this process.

abdulrehmanee12, 2023-06-25:
@xmow49 from crash logs ,it seems that attribute you send is not writable which cause this issue . But to give the final verdict, we need to know the command you are sending or is it possible for you to provide wireshark packet logs which help us to analyze it quickly ?

xmow49, 2023-06-25:
Hi,
Sorry, I don't have a Zigbee sniffer. I can't read Zigbee paquet.
With Home Assistant I can send command like this:
[image]
I tested some commands here: [logs.log](https://github.com/espressif/esp-zigbee-sdk/files/11859228/logs.log)

But the crash when I pair the device its in `check_value_ias_zone_server`. What does this function do ? Is there anything to configure during the init ? 
Is there a debug mode log for the zigbee in the ESP ?

abdulrehmanee12, 2023-06-26:
@xmow49 we have found the issue , it's because in zboss_stack during cie_address establishment they used custom attribute , that has not been registered in our SDK as it's not mandatory attribute, we will provide you the fix in latest release asap.

closed 2023-06-30 by espressif-bot

xieqinan, 2023-06-30:
Hi @xmow49 ,

The issue of the IAS Zone failing to handle the enroll command has been resolved in the [esp-zigbee-sdk](https://gitlab.espressif.cn:6688/espressif/esp-zigbee-sdk/-/commit/7a6375e134d92df06424c1639aadc815e7048f42). To fix the problem, you can update the esp-zigbee-sdk version and give it another try.

Additionally, if you're looking to implement your IAS Zone cluster, you can refer to section 8.2.2.1.3 of the Zigbee ZCL spec. It provides the necessary information and guidelines for making pairs in the IAS Zone cluster.
[image]

If you don't have any other problems, please go ahead and close the issue.