espressif/esp-zigbee-sdk #35: HA_color_dimmable_light crashing on ESP32-C6

jayavanth, 2023-06-11:
```
```

xieqinan, 2023-06-12:
Hi, @jayavanth 
After analyzing the log you provided, it appears that the problem is occurring due to the coexistence feature between WiFi and Zigbee. If we can confirm the following issues related to your project, the problem can be resolved quickly.

Firstly, could you please provide the version of the `esp-idf` that you are using as the base?

Secondly, does your project require the WiFi-Zigbee coexistence support?

Thirdly, if both `CONFIG_SW_COEXIST_ENABLE` and `CONFIG_EXTERNAL_COEX_ENABLE` are enabled, the WiFi-Zigbee coexistence feature will be enabled. However, based on the log, it seems that the value of `COEX_ADAPTER_MAGIC` is zero instead of the expected value `0xDEADBEAF`. Could you please explain the reason for modifying this value?

jayavanth, 2023-06-12:
1. I'm on commit dc016f59877d13e6e7d4fc193aa5aa764547f16d (Jun 4, 2023)
2. No, I only need Zigbee
3. I didn't modify this value actually. When I grep for the directive on `esp-idf`, I get this
```
✗ grep -RnIi COEX_ADAPTER_MAGIC
./components/esp_coex/esp32c2/esp_coex_adapter.c:164:    ._magic = COEX_ADAPTER_MAGIC,
./components/esp_coex/esp32c3/esp_coex_adapter.c:168:    ._magic = COEX_ADAPTER_MAGIC,
./components/esp_coex/include/esp_coexist_adapter.h:17:#define COEX_ADAPTER_MAGIC    0xDEADBEAF
./components/esp_coex/esp32/esp_coex_adapter.c:230:    ._magic = COEX_ADAPTER_MAGIC,
./components/esp_coex/esp32c6/esp_coex_adapter.c:163:    ._magic = COEX_ADAPTER_MAGIC,
./components/esp_coex/esp32h2/esp_coex_adapter.c:155:    ._magic = COEX_ADAPTER_MAGIC,
./components/esp_coex/esp32s3/esp_coex_adapter.c:242:    ._magic = COEX_ADAPTER_MAGIC,
./components/esp_coex/esp32s2/esp_coex_adapter.c:236:    ._magic = COEX_ADAPTER_MAGIC,
```

kelin6, 2023-06-16:
I have reproduced the issue you described, and it is caused by a mismatch in the esp-idf/components/esp_coex/lib directory. Based on your esp-idf (commit: dc016f5987), the submodule esp_coex/lib is at commit 67ba5893b08 (coex firmware version: 5315623), but in a normal esp-idf (commit: dc016f5987), the submodule esp_coex/lib commit is 28e28582 (coex firmware version: ebddf304). Therefore, it is due to the outdated esp-idf submodule.

To resolve this, please navigate to your esp-idf directory and execute the command `git submodule update --init`. This will update the esp-idf submodule, and After running this command, you will see that the esp-idf/component/esp_coex/lib commit will be 28e28582.

If you don't need to use the coexistence functionality, you can disable coexistence. Using `idf.py menuconfig`, then select 
`Component config` --> `Wire Coexistence` --> `Software control WIFI/Bluetooth coexistence`.

jayavanth, 2023-06-16:
@kelin6 ok that seems to work. It's not crashing anymore and I was able to get HA_on_off_light working. But HA_color_dimmable_light was not working. It's probably because I messed with the menuconfig a lot. Is there a way to clear all that?

closed 2023-06-16 by jayavanth