espressif/esp-zigbee-sdk #36: [Feature] Add more ZCL clusters support

xieqinan, 2023-06-12:
The following ZCL clusters still cannot work in the current version [933e0fd], but more efforts will be made to add support for these clusters.
 
Smart Energy Metering (0x0702)
Analog Input Clustering (0x000C)
Carbon Dioxide Measurement (0x040c)
PM2.5 Measurement (0x042A)
Multi State Value (0x0014)

g-chevalier, 2023-06-12:
Hi @xieqinan,

In the original issue ([[Feature request] Support more ZCL clusters (TZ-110) #20](https://github.com/espressif/esp-zigbee-sdk/issues/20)) requesting more clusters to be implemented, there were also mention of the Analog Value (0x000e) cluster (referenced as ESP_ZB_ZCL_CLUSTER_ID_ANALOG_VALUE in the post from @embts), it has been potentially forgotten in this new list.

Is it possible to add it also?
Thank you

pzeiler, 2023-06-20:
Hi @xieqinan,

I want to build a pH sensor with zigbee transport and I'd like to add a vote for the Analog Value cluster.

Thank you for your efforts!

pzeiler, 2023-07-23:
Any news on the progress? ESP_ZB_ZCL_CLUSTER_ID_ANALOG_VALUE
It's still deeply missed.

nfnvsc, 2023-07-25:
Hi, I would also like to add a vote to the `ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING` 

Thank you!

xieqinan, 2023-07-26:
@nfnvsc 
The `ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING` cluster has been added to [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_window_covering.h)

pzeiler, 2023-07-26:
@xieqinan any news on ESP_ZB_ZCL_CLUSTER_ID_ANALOG_VALUE ?

xieqinan, 2023-07-26:
@pzeiler 
I'm afraid it may take some more time. Thank you for your patience and patience.

Tropaion, 2023-08-17:
Hello @xieqinan,
I wanted to ask about the progress of implementing "Smart Energy Metering (0x0702)"?
I'm working on an open source project (https://github.com/Tropaion/ZigBee_SmartMeter_Reader) with the esp32c6 and one of the few things I'm still missing is this cluster.

lmahmutov, 2023-09-03:
HI. Also interesting on these two clusters 
Carbon Dioxide Measurement (0x040d)
PM2.5 Measurement (0x042A)

chshu, 2023-10-08:
The following clusters will be supported early next week:
- Analog Input Clustering (0x000C)
- Carbon Dioxide Measurement (0x040c)
- PM2.5 Measurement (0x042A)
- Multi State Value (0x0014)

@Tropaion Regarding to the `Smart Energy Metering (0x0702)` cluster, are you looking for the cluster itself, or the entire Smart Energy profile support including the security features defined in the Smart Energy standard?

Tropaion, 2023-10-08:
Hello @chshu, nice to hear back from you. I'm only interested in the cluster, since my device is reading data from it.

closed 2023-10-11 by espressif-bot

xieqinan, 2023-10-11:
Hello @g-chevalier @Tropaion @lmahmutov @pzeiler ,

The [Analog Input Clustering](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_analog_input.h),   [Analog value Clustering](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_analog_value.h) , [Carbon Dioxide Measurement](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_carbon_dioxide_measurement.h), [PM2.5 Measurement](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_pm2_5_measurement.h) and [Multi State Value](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_multistate_value.h) have been supported in [esp-zigbee-lib v1.0.0](https://github.com/espressif/esp-zigbee-sdk/tree/main/components/esp-zigbee-lib).

g-chevalier, 2023-10-29:
Hi @xieqinan,

Sorry for the delay answering this issue.
I started to test the Analog Value cluster just now, and I think there is a bug.
The "PresentValue" attribute of this cluster is of type uint8 (defined in esp_zb_analog_value_cluster_cfg_t) instead of Single as per the standard (0x39).

It seems that there is the same problem with the Analog Output cluster, however, the Analog Input cluster seems OK (defined as float_t).

I tried to compile flash and test. Running the code works, but reading this attribute with Zigbee2MQTT brings a weird value as the wrong type is used.
Trying to write the attribute brings the following error: "E (46338) ESP_ZIGBEE_CORE: Unsupported zcl attribute type(0x39)", which seems to be logic since Z2M sends a float and the ESP implementation looks for a short int.

xieqinan, 2023-10-30:
@g-chevalier ,

Thank you for your reminder. As you mentioned, there is a bug in the SDK where the present value of analog input/output should be of a single type. We will fix this issue as soon as possible

xieqinan, 2023-11-03:
@g-chevalier 

> I started to test the Analog Value cluster just now, and I think there is a bug. The "PresentValue" attribute of this cluster is of type uint8 (defined in esp_zb_analog_value_cluster_cfg_t) instead [...]

This issue has been fixed the esp-zigbee-sdk [v1.0.2](https://github.com/espressif/esp-zigbee-sdk), please update and test it.

xieqinan, 2023-11-03:
@Tropaion ,

> Hello @chshu, nice to hear back from you. I'm only interested in the cluster, since my device is reading data from it.

The ZCL metering cluster has been supported in esp-zigbee-sdk [v1.0.2](https://github.com/espressif/esp-zigbee-sdk), please refer the [esp_zigbee_zcl_metering.h](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_metering.h)

Tropaion, 2023-11-03:
@xieqinan Cool, thank you, I will try it in the next few days.

g-chevalier, 2023-11-11:
Hi @xieqinan,

I tested the Analog Value cluster with the new release, and it works.
At least with my use case where I'm able to read and write PresentValue attribute as a float number to/from my end-device using the Zigbee2MQTT Dev console (I did not test everything thoroughly…).

Thank you for this fix.