espressif/esp-zigbee-sdk #40: ESP_ERROR_CHECK failed on simple descriptor response after the request timeout

Ravenen, 2023-06-19:
Hello!

While experimenting with the SDK, I encountered an issue:

During my experimentation, I send a simple descriptor request from the gateway code to a Zigbee device, specifically a Zigbee button, triggered by the `DEVICE ANNCE` event. The Zigbee button operates as a sleepy node and enters sleep mode, causing the "simple descriptor request timeout" to occur. 

However, when I manually wake up the Zigbee button by pressing its button, it sends the expected simple descriptor response. Unfortunately, this action leads to a crash of the ESP32. 

It appears that the SDK attempts to delete a timeout timer that is no longer exists at that point (my apologies for possibly incorrect assumption).

I am using the latest version of the `esp-zigbee-sdk` on an ESP32C6, which is being utilized as a Zigbee gateway.

[image]

xieqinan, 2023-06-20:
Hi @Ravenen,
This issue is very valuable, as it may be triggered by a mismatch in the `simple_des_req_timeout` function. However, I have failed to reproduce the issue. Would you like to share the method for sending a simple request to a Zigbee button?
I call the following method to send a simple request from Zigbee gateway to Zigbee button, which get the correct response.
```
void simple_request_handler(esp_zb_zdp_status_t zdo_status, esp_zb_af_simple_desc_1_1_t *simple_desc, void *user_ctx)
{
    ESP_LOGI(TAG, "Simple Request status:%d, device:%d, endpoint:%d", zdo_status, simple_desc->app_device_id, simple_desc->endpoint);
}

void user_find_cb(esp_zb_zdp_status_t zdo_status, uint16_t addr, uint8_t endpoint, void *user_ctx)
{
    ESP_LOGI(TAG, "User find cb: response_status:%d, address:0x%x, endpoint:%d", zdo_status, addr, endpoint);
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        on_off_light.endpoint = endpoint;
        on_off_light.short_addr = addr;
        esp_zb_zdo_simple_desc_req_param_t cmd_req = {
            .endpoint = endpoint,
            .addr_of_interest = addr
        }; 
        ESP_LOGW(TAG, "send request to endpoint: %d, short address: %x", cmd_req.endpoint, cmd_req.addr_of_interest);
        esp_zb_zdo_simple_desc_req(&cmd_req, simple_request_handler, NULL);
    }
}

```

xieqinan, 2023-07-14:
Hi
Please feel free to discuss any problems you have with the esp-zigbee-sdk here. If the issue has been fixed, please close it.

closed 2023-08-11 by espressif-bot

xieqinan, 2023-08-11:
Hi,
The issue has been fixed in newest master, please update and try it again.