espressif/esp-zigbee-sdk #42: ESP32C6 coordinator: Unable to receive attribute reports from devices

pieterjanbuntinx, 2023-06-29:
Hi

I am developing a custom Zigbee gateway with an ESP32S3 as the main microcontroller and an ESP32C6 as the RCP. Currently, I am trying to receive attribute reports from some temperature/humidity sensors (Sonoff SNZB-02). I am using the `esp_zigbee_gateway` example (master & esp-idf v5.1rc2) with a few additions:
```
static void esp_zb_task(void *pvParameters) {
  /* initialize Zigbee stack with Zigbee coordinator config */
  esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZC_CONFIG();
  esp_zb_init(&zb_nwk_cfg);
  esp_zb_device_add_report_attr_cb(esp_zb_dev_reporting_cb);
  /* initiate Zigbee Stack start without zb_send_no_autostart_signal auto-start
  
  // .....
}
```

```
static void esp_zb_dev_reporting_cb(esp_zb_zcl_addr_t *addr, uint8_t endpoint,
                                    uint16_t cluster_id, uint16_t attr_id,
                                    esp_zb_zcl_attr_type_t attr_type,
                                    void *value) {
  ESP_LOGI(TAG,
           "Switch got report attribute from "
           "address:0x%x,cluster_id:0x%x,attr_id:0x%x,value:%d,attr_type:0x%x,",
           addr->u.short_addr, cluster_id, attr_id, *(uint8_t *)value,
           attr_type);
}
```

But the callback `esp_zb_dev_reporting_cb` never seems to be called. When using Zigbee2MQTT I can see the received "attributeReport" message about every 10 minutes or when the temperature/humidity changes. 

Am I doing something wrong or is this something that does not work yet in the SDK? 

For some door/window sensors, I have a similar issue. How can I have a callback when there is a "statusChangeNotification"?

abdulrehmanee12, 2023-06-30:
hi @pieterjanbuntinx Your implemented callback for receiving attribute reports is missing the 'dst_endpoint' information. We recently modified this callback based on a customer's request to include 'dst_endpoint' in the 'esp_zb_dev_reporting_cb' function. Additionally, we have implemented the report callback in esp_zigbee_customized_devices example [esp_HA_customized_switch](https://gitlab.espressif.cn:6688/espressif/esp-zigbee-sdk/-/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c) . You can refer to it for guidance and please make sure to add 'dst_endpoint' and try again.

`static void esp_zb_dev_reporting_cb(esp_zb_zcl_addr_t *addr, uint8_t src_endpoint, uint8_t dst_endpoint, uint16_t cluster_id,
                                    uint16_t attr_id, esp_zb_zcl_attr_type_t attr_type, void *value)
{
    ESP_LOGI(TAG, "Switch got report attribute from address:0x%x,src_ep:%d,dst_ep:%d,cluster_id:0x%x,attr_id:0x%x,value:%d,attr_type:0x%x,", addr->u.short_addr, src_endpoint, dst_endpoint, cluster_id,
    attr_id, *(uint8_t *)value, attr_type);
}`

pieterjanbuntinx, 2023-07-03:
Hi @abdulrehmanee12,
Thank you for your response!

Sadly changing the callback function did not solve my issue. Otherwise, with the wrong callback, I would have expected the ESP to crash every time that callback was called. 

The esp_HA_customized_switch example is where I found the original callback. I must have copied it from an earlier version of the SDK.

In that example, they bind the two devices. Is that necessary for reporting to work? I wanted to try and bind the devices but I was not able to create a find temp/humid sensor or find door sensor function as there exists in the example for the on_off_light (esp_zb_zdo_find_on_off_light). I was thinking of using the ZBOSS function `zb_uint8_t zb_zdo_match_desc_req(zb_uint8_t param, zb_callback_t cb);` directly. But I am not sure where I can get the `zb_uint8_t param` argument from (ZBOSS stack pointer  I think?). Is there perhaps an example of how to do this?

abdulrehmanee12, 2023-07-03:
@pieterjanbuntinx  Yes, binding two devices is necessary for reporting to function correctly. At present, our SDK does not support the "find temp/humid sensor" or "find door sensor" functions. However, in our next release, we will introduce a common function that caters to those requirements. As an example for your work, here is a guide on how to achieve it:

`void esp_zb_zdo_find_door_lock(esp_zb_zdo_match_desc_req_param_t *cmd_req, esp_zb_zdo_match_desc_callback_t user_cb, void *user_ctx)
{
    zb_uint8_t bufid = zb_buf_get_out();
    zb_uint8_t zdo_tsn = 0;
    zb_zdo_match_desc_param_t *req;
    // 1 is total number of the cluster list
    req = zb_buf_initial_alloc(bufid, sizeof(zb_zdo_match_desc_param_t) + (1) * sizeof(zb_uint16_t));
    req->nwk_addr = cmd_req->dst_nwk_addr;
    req->addr_of_interest = cmd_req->addr_of_interest;
    req->profile_id = ZB_AF_HA_PROFILE_ID;
    /*  searching for temp_sensor */
    req->num_in_clusters = 1;
    req->num_out_clusters = 0;
    req->cluster_list[0] = ZB_ZCL_CLUSTER_ID_DOOR_LOCK;
    zdo_tsn = zb_zdo_match_desc_req(bufid, find_device_cb);
    ZB_SCHEDULE_APP_ALARM(match_desc_req_device_timeout, zdo_tsn, ESP_ZB_MATCH_DESC_REQ_TIMEOUT);
    esp_zb_zdo_callback_register(zdo_tsn, ESP_ZDO_MATCH_DESC_REQ_CLID, (void *)user_cb, user_ctx);
}`

pieterjanbuntinx, 2023-07-03:
Great thanks! I will try this out now. 
Do you have any idea how I can receive `statusChangeNotifications` from an IAS zone like for example a door lock? I don't think this is implemented yet in the SDK. Do you know of a function in the ZBOSS lib?

abdulrehmanee12, 2023-07-03:
The IAS zone cluster has already been implemented in our SDK. To receive `statusChangeNotifications`  you can utilize the IAS zone status change notification command `esp_zb_zcl_ias_zone_status_change_notif_cmd_req`

pieterjanbuntinx, 2023-07-03:
> The IAS zone cluster has already been implemented in our SDK. To receive `statusChangeNotifications` you can utilize the IAS zone status change notification command `esp_zb_zcl_ias_zone_status_change_notif_cmd_req`

Correct me if I am wrong, but is this function not only for sending the status notifications? I need a function similar to `esp_zb_device_add_report_attr_cb` that calls the provided callback when such a command is received.

> @pieterjanbuntinx Yes, binding two devices is necessary for reporting to function correctly. At present, our SDK does not support the "find temp/humid sensor" or "find door sensor" functions. [...]

I was able to get most of it working. I now have the following code:
```
void find_temp_humid_sensor(zb_bufid_t bufid) {
    /*Get the beginning of the response.*/
    zb_zdo_match_desc_resp_t* resp = (zb_zdo_match_desc_resp_t*)zb_buf_begin(bufid);
    /* Get the pointer to the parameters buffer, which stores APS layer
     * response.
     */
    zb_apsde_data_indication_t* ind = ZB_BUF_GET_PARAM(bufid, zb_apsde_data_indication_t);
    zb_uint8_t*                 match_ep;

    if (resp->status == ZB_ZDP_STATUS_SUCCESS) {
        /* Match EP list follows right after response header. */
        match_ep = (zb_uint8_t*)(resp + 1);
        printf("Found cluster: addr %d, ep %d\n", ind->src_addr, *match_ep);
    }
}

void esp_zb_zdo_find_temp_humid_sensor(esp_zb_zdo_match_desc_req_param_t* cmd_req, esp_zb_zdo_match_desc_callback_t user_cb, void* user_ctx) {
    zb_uint8_t                 bufid   = zb_buf_get_out();
    zb_uint8_t                 zdo_tsn = 0;
    zb_zdo_match_desc_param_t* req;
    // 1 is total number of the cluster list
    req                   = zb_buf_initial_alloc(bufid, sizeof(zb_zdo_match_desc_param_t) + (1) * sizeof(zb_uint16_t));
    req->nwk_addr         = cmd_req->dst_nwk_addr;
    req->addr_of_interest = cmd_req->addr_of_interest;
    req->profile_id       = ZB_AF_HA_PROFILE_ID;
    /* searching for temp/humid sensor */
    req->num_in_clusters  = 1;
    req->num_out_clusters = 0;
    req->cluster_list[0]  = ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
    // req->cluster_list[1]  = ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT;
    zdo_tsn = zb_zdo_match_desc_req(bufid, find_temp_humid_sensor);
    // ZB_SCHEDULE_APP_ALARM(match_desc_req_device_timeout, zdo_tsn, ESP_ZB_MATCH_DESC_REQ_TIMEOUT);
    // esp_zb_zdo_callback_register(zdo_tsn, ESP_ZDO_MATCH_DESC_REQ_CLID, (void*)user_cb, user_ctx);
}
```
But I have the last two lines commented out. Most importantly, `esp_zb_zdo_callback_register` seems to be missing in the SDK and/or ZBoss lib. I guess this is something that will be released in the next release of the SDK? Do you perhaps have any estimate of when this will release?

abdulrehmanee12, 2023-07-04:
> Correct me if I am wrong, but is this function not only for sending the status notifications? I need a function similar to esp_zb_device_add_report_attr_cb that calls the provided callback when such a command is received.
Yes, you're correct. We haven't tested receiving `statusChangeNotifications` from an IAS zone yet, so we need to verify if the callback is supported. We will provide you with an update soon.

> But I have the last two lines commented out. Most importantly, esp_zb_zdo_callback_register seems to be missing in the SDK and/or ZBoss lib. I guess this is something that will be released in the next release of the SDK? Do you perhaps have any estimate of when this will release?
It has already been implemented in our SDK and I tested it on my side.

pieterjanbuntinx, 2023-07-04:
> It has already been implemented in our SDK and I tested it on my side.

If I search on GitHub I cannot find any reference to `esp_zb_zdo_callback_register` except for in this post. Are you sure it is in the main branch on GitHub?

abdulrehmanee12, 2023-07-04:
> If I search on GitHub I cannot find any reference to esp_zb_zdo_callback_register except for in this post. Are you sure it is in the main branch on GitHub?

Yes, my mistake. It's our internal implementation and is not currently exposed to users. However, in the next release, we plan to provide a common function that will support door sensors and temp/humid sensors.

pieterjanbuntinx, 2023-07-04:
Okay, thanks!
We will then be patiently waiting on the next release :)

closed 2023-07-14 by espressif-bot

xieqinan, 2023-07-14:
The ias zone cluster can not receive the status notification issue has been fixed in [esp-zigbee-lib](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_core.h#L740C1-L740C1).

Please update the latest [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk/tree/main) to fix your problem.

abdulrehmanee12, 2023-07-14:
@pieterjanbuntinx  We have released a new version of esp_zigbee_sdk that includes the common function to find the matched cluster `esp_zb_zdo_match_cluster `  please use the updated `esp_zb_zdo_match_desc_req_param_t` to create your required cluster list .  Additionally, we have included a callback function `esp_zb_add_ias_zone_status_notification_cb` for handling `statusChangeNotifications`. Please use the latest version and go ahead with your project.

xieqinan, 2023-07-18:
Hi,
I suggest registering an IAS zone request callback using `esp_zb_add_ias_zone_enroll_request_cb` to respond to the door sensor's request. Additionally, utilize `esp_zb_zcl_ias_zone_enroll_cmd_resp` to assign the ZoneID attribute for the door sensor.

pieterjanbuntinx, 2023-07-18:
Hi @xieqinan,

I am trying to do step 5 from your screenshot but I am not sure how to send the enroll response command inside of the callback shown below. In this callback, I have no access to the ieee/short address of the device that has sent the enroll request. 
For testing, I captured the address in a global variable when the device has joined and I used address that in the enroll response command. The device just leaves the network after a few seconds and does not reconnect. If that would have worked, it is not the correct way of doing this because I cannot make sure that the device that had joined is the same as the device that send the enroll request. Do you perhaps have an example of how to do this?

Can you maybe also link to the document where you got this screenshot from? I have found a similar document but from NXP, is there an ESP-specific document available?

```
void ias_zone_enroll_request_cb(uint16_t zone_type, uint16_t manuf) {
    printf("zone_enroll_req cb: zone_type: %i, manuf: %i\n", zone_type, manuf);

    esp_zb_zcl_ias_zone_enroll_response_cmd_t cmd_resp;
    // cmd_resp.zcl_basic_cmd.dst_addr_u // not available in callback
    cmd_resp.zcl_basic_cmd.dst_endpoint = 1;
    cmd_resp.zcl_basic_cmd.src_endpoint = 1;
    cmd_resp.address_mode               = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT; // ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT
    cmd_resp.zone_id                    = 1;
    cmd_resp.enroll_rsp_code            = ESP_ZB_ZCL_IAS_ZONE_ENROLL_RESPONSE_CODE_SUCCESS;
    esp_zb_zcl_ias_zone_enroll_cmd_resp(&cmd_resp);
}
```

xieqinan, 2023-07-18:
Hi @pieterjanbuntinx ,
If the IAS Zone cluster has not been created, will the door sensor leave the Zigbee network? It is difficult to explain the reason behind the IAS Zone command triggering the leave indication signal.

> I captured the address in a global variable when the device has joined and I used address that in the enroll response command.

The current method should work, but it is not optimal; a better solution that the network address of request side will as the parameter of callback will be provided in esp-zigbee-sdk v1.0.0. 

> Can you maybe also link to the document where you got this screenshot from? I have found a similar document but from NXP, is there an ESP-specific document available?

The screenshot is from the [Zigbee Zcl library specification 8.2.2.1.3](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf).

pieterjanbuntinx, 2023-07-19:
I am having trouble writing the CIE address to the sensor.  I am using the code below to write and read the CIE address to the door sensor. But it always reads out as 00:00:00:00:00:00:00:00 on one sensor and FF:FF:FF:FF:FF:FF:FF:FF on another.
I do get a `zone_enroll_cmd_request` from the door sensor. If I respond with a `zone_enroll_cmd_response`, I can read out the `ZoneState` attribute as 1 (enrolled). After a while, I get a `ZDO Leave Indication`. The sensor won't rejoin the network after this anymore and it looks like it went back into pairing mode (the led blinks). 

> If the IAS Zone cluster has not been created, will the door sensor leave the Zigbee network? It is difficult to explain the reason behind the IAS Zone command triggering the leave indication signal.

If I don't create the IAS Zone cluster locally and don't bind it, I get the exact same behavior.

> The current method should work, but it is not optimal; a better solution that the network address of request side will as the parameter of callback will be provided in esp-zigbee-sdk v1.0.0.

Will this be the next release?

``` c
void esp_zb_app_signal_handler(esp_zb_app_signal_t* signal_struct) {
// (...)

    switch (sig_type) {
    // (...)
        case ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE: {
            dev_annce_params = (esp_zb_zdo_signal_device_annce_params_t*)esp_zb_app_signal_get_params(p_sg_p);
            ESP_LOGI(TAG, "New device commissioned or rejoined (short: 0x%04hx)", dev_annce_params->device_short_addr);
            
            // Write CIE address 
            esp_zb_zcl_write_attr_cmd_t req_param = {0};
            req_param.zcl_basic_cmd.src_endpoint  = 1;
            req_param.zcl_basic_cmd.dst_endpoint  = 1;
            memcpy(req_param.zcl_basic_cmd.dst_addr_u.addr_long, dev_annce_params->ieee_addr, sizeof(esp_zb_ieee_addr_t));
            req_param.clusterID     = ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE;
            req_param.attributeID   = ESP_ZB_ZCL_ATTR_IAS_ZONE_IAS_CIE_ADDRESS_ID;
            req_param.attrType      = ESP_ZB_ZCL_ATTR_TYPE_IEEE_ADDR;
            esp_zb_ieee_addr_t addr = {0};
            esp_zb_get_long_address(addr);
            req_param.attrVal = addr;
            esp_zb_zcl_write_attr_cmd_req(&req_param);

            break;
        }
    // (...)
    }
}

// (...)
// read CIE address
esp_zb_zcl_read_attr_cmd_t read_req;
read_req.address_mode               = ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT;
read_req.attributeID                = ESP_ZB_ZCL_ATTR_IAS_ZONE_IAS_CIE_ADDRESS_ID;
read_req.clusterID                  = ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE;
read_req.zcl_basic_cmd.dst_endpoint = EXTERNAL_DEVICES_ENDPOINT;
read_req.zcl_basic_cmd.src_endpoint = EXTERNAL_DEVICES_ENDPOINT;
memcpy(read_req.zcl_basic_cmd.dst_addr_u.addr_long, test_addr, sizeof(esp_zb_ieee_addr_t));
esp_zb_zcl_read_attr_cmd_req(&read_req);
// (...)
```

```
I (11590) COREDUMP_UPLOAD: reset reason is 3
I (12760) ESP_ZB_GATEWAY: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (13140) ESP_ZB_GATEWAY: Formed network successfully (ieee extended address: 18:97:40:fe:ff:ca:4c:40, PAN ID: 0xc23b)
I (14000) ESP_ZB_GATEWAY: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (14400) ESP_ZB_GATEWAY: Network steering started
I (14640) ESP_ZB_GATEWAY: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (15090) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (15610) ESP_ZB_GATEWAY: New device commissioned or rejoined (short: 0xd475)
zone_enroll_req cb: zone_type: 21, manuf: 0
Bind successful! (d475)
received stat_ch_notif: zone_status: 1, ext status: 0, zone_id: 255, delay: 0
received stat_ch_notif: zone_status: 1, ext status: 0, zone_id: 0, delay: 0
received stat_ch_notif: zone_status: 1, ext status: 0, zone_id: 0, delay: 0
received stat_ch_notif: zone_status: 1, ext status: 0, zone_id: 0, delay: 0
requested attributes
requested attributes
received stat_ch_notif: zone_status: 0, ext status: 0, zone_id: 0, delay: 0
read_attr_resp_cb: status: 0, cluster_id: 1280, attr_id: 16, attr_type: 240, address: 00:00:00:00:00:00:00:00
received stat_ch_notif: zone_status: 1, ext status: 0, zone_id: 0, delay: 0
received stat_ch_notif: zone_status: 0, ext status: 0, zone_id: 0, delay: 0
received stat_ch_notif: zone_status: 1, ext status: 0, zone_id: 0, delay: 0
I (30160) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
read_attr_resp_cb: status: 0, cluster_id: 1280, attr_id: 16, attr_type: 240, address: 00:00:00:00:00:00:00:00
received stat_ch_notif: zone_status: 1, ext status: 0, zone_id: 0, delay: 0
received stat_ch_notif: zone_status: 1, ext status: 0, zone_id: 0, delay: 0
received stat_ch_notif: zone_status: 1, ext status: 0, zone_id: 0, delay: 0
requested attributes
```

pieterjanbuntinx, 2023-07-19:
With temperature/humidity sensors I get a similar behavior. The device joins the network, I do binding and configure reporting in the bind callback and the device starts sending reports for the configured attributes. But after like 10 seconds the device leaves the network and never rejoins. The device also seems to be in pairing mode with the led blinking.

``` c
I (11590) COREDUMP_UPLOAD: reset reason is 3
I (12820) ESP_ZB_GATEWAY: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (13200) ESP_ZB_GATEWAY: Formed network successfully (ieee extended address: 18:97:40:fe:ff:ca:4c:40, PAN ID: 0x7166)
I (14030) ESP_ZB_GATEWAY: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (14430) ESP_ZB_GATEWAY: Network steering started
I (24800) ESP_ZB_GATEWAY: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (25030) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (25890) ESP_ZB_GATEWAY: Switch got report attribute from address:0xfb3d,src_ep:1,dst_ep:1,cluster_id:0x405,attr_id:0x0,value:248,attr_type:0x21,
I (25910) ESP_ZB_GATEWAY: New device commissioned or rejoined (short: 0xfb3d)
Bind successful! (fb3d)
I (28870) ESP_ZB_GATEWAY: Switch got report attribute from address:0xfb3d,src_ep:1,dst_ep:1,cluster_id:0x402,attr_id:0x0,value:108,attr_type:0x29,
I (29070) ESP_ZB_GATEWAY: Switch got report attribute from address:0xfb3d,src_ep:1,dst_ep:1,cluster_id:0x405,attr_id:0x0,value:109,attr_type:0x21,
I (35050) ESP_ZB_GATEWAY: Switch got report attribute from address:0xfb3d,src_ep:1,dst_ep:1,cluster_id:0x402,attr_id:0x0,value:117,attr_type:0x29,
I (35090) ESP_ZB_GATEWAY: Switch got report attribute from address:0xfb3d,src_ep:1,dst_ep:1,cluster_id:0x405,attr_id:0x0,value:232,attr_type:0x21,
I (38050) ESP_ZB_GATEWAY: Switch got report attribute from address:0xfb3d,src_ep:1,dst_ep:1,cluster_id:0x402,attr_id:0x0,value:108,attr_type:0x29,
I (40080) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (40880) ESP_ZB_GATEWAY: ZDO signal: ZDO Leave Indication (0x13), status: ESP_OK
I (40920) ESP_ZB_GATEWAY: ZDO signal: ZDO Leave Indication (0x13), status: ESP_OK
```

``` c
// (...)

// is called when device has joined in esp_zb_app_signal_handler
esp_zb_zdo_bind_req_param_t bind_req;
memcpy(bind_req.src_address, device->ieee_addr, sizeof(esp_zb_ieee_addr_t));
bind_req.src_endp      = 1;
bind_req.cluster_id    = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
esp_zb_get_long_address(bind_req.dst_address_u.addr_long);
bind_req.dst_endp     = EXTERNAL_DEVICES_ENDPOINT;
bind_req.req_dst_addr = device->short_addr;
esp_zb_zdo_device_bind_req(&bind_req, bind_cb, (void*)device);

// (...)

// bind callback, attribute reporting is here configured
static void bind_cb(esp_zb_zdp_status_t zdo_status, void* user_ctx) {
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        device_t* device = (device_t*)user_ctx;

        printf("Bind successful! (%x)\n", device->short_addr);

        /* configure report attribute command */
        esp_zb_zcl_config_report_cmd_t report_cmd    = {0};
        int16_t                        report_change = 2;
        memcpy(&report_cmd.zcl_basic_cmd.dst_addr_u.addr_long, device->ieee_addr, sizeof(esp_zb_ieee_addr_t));
        report_cmd.zcl_basic_cmd.dst_endpoint = 1;
        report_cmd.zcl_basic_cmd.src_endpoint = EXTERNAL_DEVICES_ENDPOINT;
        report_cmd.address_mode               = ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT;
        report_cmd.clusterID                  = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
        report_cmd.attributeID                = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID;
        report_cmd.attrType                   = ESP_ZB_ZCL_ATTR_TYPE_S16;
        report_cmd.min_interval               = 1;
        report_cmd.max_interval               = 10;
        report_cmd.reportable_change          = (void*)&report_change;
        esp_zb_zcl_config_report_cmd_req(&report_cmd);
    }
}

xieqinan, 2023-07-20:
Let's address the leave indication problem first. Can you please refer to the related [issue](https://github.com/espressif/esp-zigbee-sdk/issues/21)?  Ensure the same preconfigured key between Zigbee gateway and the sensor.

pieterjanbuntinx, 2023-07-20:
I have set `esp_zb_secur_link_key_exchange_required_set` to false and that seems to work for both kinds of sensors, thanks! I do find it strange that none of the sensors I have provide the install code.

xieqinan, 2023-07-20:
Yes, it is a better way to join Zigbee network by the install code.

Stelios1908, 2024-03-17:
ηι!!!
Is there finally an example to show how I can get from an ED a zone state?

straga, 2024-05-27:
before was esp_zb_add_ias_zone_status_notification_cb, what now need use for cb IAS zone change ?