espressif/esp-zigbee-sdk #44: [Feature request] add Router device example

diazmanuel, 2023-07-04:
Hello again, 
so actually i am trying to make my end device into a router so that the device will also repeat the zigbee network. after several tries i couldn't get it to work.
I make the configuration according to the documentation and get the device to continue working normally. but when disconnecting the coordinator the device leaves the network. Being a router I would expect the device to maintain the network . In addition, if I restart the device with the coordinator disconnected, it does not generate the network either.

What I expect when configuring the device as a router is that if the coordinator is disconnected, it emits the zigbee network. If the coordinator is connected, it should repeat the signal to increase the range of the zigbee network.

Is there something I'm misunderstanding? Reviewing the examples, I realized that none of them configure the device as a router. Is it possible to add an example of a repeater configured as a router?

thanks

xieqinan, 2023-07-06:
Hi @diazmanuel ,
The ESP-ZIGBEE-SDK can simply use the pre-configured settings in the header file to set the router role.
```
#define ESP_ZB_ZR_CONFIG()                                                                          \
    {                                                                                               \
        .esp_zb_role = ESP_ZB_DEVICE_TYPE_ROUTER,                                                   \
        .install_code_policy = INSTALLCODE_POLICY_ENABLE,                                           \
        .nwk_cfg.zczr_cfg = {                                                                       \
            .MAX_CHILDREN = MAX_CHILDREN,                                                           \
        },                                                                                          \
    }
```

> but when disconnecting the coordinator the device leaves the network. Being a router I would expect the device to maintain the network .

In this case, the router can maintain communication with joined end devices but not maintain the zigbee network, because it has the capability to upgrade to a coordinator and forms network. Therefore, if the router device is restarted with the coordinator disconnected, it only will stay on  the original network.

> What I expect when configuring the device as a router is that if the coordinator is disconnected, it emits the zigbee network. If the coordinator is connected, it should repeat the signal to increase the range of the zigbee network.

Yes, you are correct. When the coordinator is disconnected, the router will continue to emit the original Zigbee network. However, if the coordinator is connected,  it will repeat the signal to increase the range of the zigbee network.

xieqinan, 2023-07-14:
Hi,
The zigbee router device example has been add in [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_color_dimmable_light). 

If there is anything we can help you with, you can reopen it or create a new one.

closed 2023-07-14 by diazmanuel

xieqinan, 2024-05-07:
@Layochi ，

Could you please provide more details about your requirements? I don't have a complete understanding of "reverse engineering" and would appreciate more context.

Layochi, 2024-05-07:
I tried to understand the code you provided to figure out which parts of the code are used to set the ESP as a router without disturbing the endpoint part.