espressif/esp-zigbee-sdk #45: Code working on Dev.-Board but not on ESP32-C6-WROOM

Tropaion, 2023-07-04:
Hello,
I developed an application on the Dev.-Board, which is working without any problems.
Now, I developed a small custom board with the recommended layout for the ESP32-C6-WROOM-1. After flashing the application via UART0 I get following error, which doesn't make any sense to me, because I'm not even using the eFuse.
```
I (27) boot: ESP-IDF v5.2-dev-1349-gcf7e743a9b 2nd stage bootloader
I (27) boot: compile time Jul  4 2023 13:58:41
I (29) boot: chip revision: v0.0
I (31) boot.esp32c6: SPI Speed      : 40MHz
I (36) boot.esp32c6: SPI Mode       : DIO
I (41) boot.esp32c6: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00089800
I (84) boot:  3 zb_storage       Unknown data     01 81 0009a000 00004000
I (92) boot:  4 zb_fct           Unknown data     01 81 0009e000 00000400
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=42058020 size=0df80h ( 57216) map
I (135) esp_image: segment 1: paddr=0001dfa8 vaddr=40800000 size=02070h (  8304) load
I (140) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=522fch (336636) map
I (274) esp_image: segment 3: paddr=00072324 vaddr=40802070 size=0b74ch ( 46924) load
I (296) esp_image: segment 4: paddr=0007da78 vaddr=4080d7c0 size=01964h (  6500) load
I (307) boot: Loaded app from partition at offset 0x10000
I (307) boot: Disabling RNG early entropy source...
I (318) cpu_start: Unicore app
I (319) cpu_start: Pro cpu up.
W (329) clk: esp_perip_clk_init() has not been implemented yet
I (335) cpu_start: Pro cpu start user code
I (335) cpu_start: cpu freq: 160000000 Hz
I (336) cpu_start: Application information:
I (338) cpu_start: Project name:     ZB_ESP32
I (343) cpu_start: App version:      1
I (348) cpu_start: Compile time:     Jul  4 2023 13:57:10
I (354) cpu_start: ELF file SHA256:  8c5abced4...
I (359) cpu_start: ESP-IDF:          v5.2-dev-1349-gcf7e743a9b
I (366) cpu_start: Min chip rev:     v0.0
I (370) cpu_start: Max chip rev:     v0.99
I (375) cpu_start: Chip rev:         v0.0
I (380) heap_init: Initializing. RAM available for dynamic allocation:
I (387) heap_init: At 408154F0 len 00067120 (412 KiB): D/IRAM
I (394) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (400) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (407) spi_flash: detected chip: generic
I (411) spi_flash: flash io: dio
W (415) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (428) eFuse: calibration efuse version does not match, set default version to 0
W (458) eFuse: calibration efuse version does not match, set default version to 0

assert failed: esp_efuse_rtc_calib_get_chan_compens esp_efuse_rtc_calib.c:63 ((version >= ESP_EFUSE_ADC_CALIB_VER_MIN) && (version <= ESP_EFUSE_ADC_CALIB_VER_MAX))
Core  0 register dump:
MEPC    : 0x40800452  RA      : 0x40807524  SP      : 0x4087e020  GP      : 0x4080dfc0
0x40800452: panic_abort at C:/Users/Tropaion/esp/esp-idf/components/esp_system/panic.c:452

0x40807524: __ubsan_include at C:/Users/Tropaion/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000024  S1      : 0x00000001  A0      : 0x4087e05c  A1      : 0x4080e289
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x40815000
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x4087e1d3
S4      : 0x4080e288  S5      : 0x42000000  S6      : 0x40800258  S7      : 0x000522fc
0x40800258: call_start_cpu0 at C:/Users/Tropaion/esp/esp-idf/components/esp_system/port/cpu_start.c:294

S8      : 0x0000df80  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000
[... 61 lines cut ...]
4087e380: 0x00000000 0x00000045 0x000053c0 0x4086c49c 0x77932578 0x14ed3896 0x00000000 0xffffffff
4087e3a0: 0x00000000 0x00000000 0x00010000 0x00089800 0x00000000 0x00000000 0x00000000 0x00000000
4087e3c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e400: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 8c5abced4

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x1c (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
0x4001975a: software_reset_cpu in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x40875730,len:0x181c
load:0x4086c410,len:0xd44
load:0x4086e610,len:0x2e2c
entry 0x4086c41a
```
@likunqiao097304 Any idea why this happens?
Thanks,
Tropaion

xieqinan, 2023-07-05:
@Tropaion，
I checked the output log, and it appears that your C6 module sample is at an early stage and has not yet had the eFuse burnt.

A potential workaround for this issue is to switch your ESP-IDF commit to `38b4952` at a later stage. However, please note that this workaround is intended for testing purposes only and is not recommended for use in production environments where eFuse versions are crucial.

chshu, 2023-07-05:
@Tropaion If you bought the modules from official store, the eFuse should be flashed already. Could you run this command and share with us the output:

```
espefuse.py summary
```

To checkout specific IDF commit, you can use this command:
```
git checkout 38b4952
git submodule update --init --recursive
```

Tropaion, 2023-07-05:
@chshu If bought the module from Adafruit, they were the only who had them in stock: https://www.adafruit.com/product/5671

> espefuse.py summary
I will send you the result tomorrow, I'm at home office today.

> To checkout specific IDF commit, you can use this command:
> 
> ```
> git checkout 38b4952
> git submodule update --init --recursive
> ```

I don't know if this is possible, I didn't download it via git. I installed my environment with the VS Code and espressif plugin.
And if I use this workaround, is it still possible to use the latest zigbee-sdk?

Tropaion, 2023-07-06:
I tried running ```espefuse.py summary``` but nothing happens.
[image]

xieqinan, 2023-07-10:
Hi @Tropaion ,
Thank you for providing the log. The problem has been resolved and will be publis hed this week.

If you need to stay on a specific commit, please add the patch to your branch.
[efuse_patch.zip](https://github.com/espressif/esp-zigbee-sdk/files/11998466/efuse_patch.zip)

Tropaion, 2023-07-10:
@xieqinan Thanks, what exactly was the problem?
Will my modules work again when I update my installation next week?
I don't need a specific commit, I normally use the latest version.

xieqinan, 2023-07-12:
The reason for the chip may not being burnt with the ADC calibration block. We have already fixed the issue and will sync it within this week. You can use the provided patch to solve your problem temporarily.

xieqinan, 2023-07-14:
The issue has been fixed in the latest [esp-idf](https://github.com/espressif/esp-idf/commit/a6e2ea141ab5aa62bfa7b796643dfa1576542d58).

Please test whether the fix is helpful to you and close the issue.

closed 2023-07-18 by chshu

Tropaion, 2023-07-18:
@xieqinan Sorry, was on holidays.
I tested it now with the master branch and it's working, thank you.