espressif/esp-zigbee-sdk #46: Does the current esp-zigbee-sdk support low power applications for ESP32-H2?

diazmanuel, 2023-07-04:
So The other day I bought some esp32H2 development boards and I would like to know if the sdk is already in a version that allows working with these devices in low consumption to create end devices. that the consumption is truly what the datasheet specifies and that it allows light and deep sleep without losing the zigbee network.

Thanks

xieqinan, 2023-07-04:
@diazmanuel ,
The low-power feature for Zigbee is currently being tested internally and is expected to be available in the coming months.

diazmanuel, 2023-07-04:
@xieqinan okey thanks you for the quick response

closed 2023-07-14 by diazmanuel

joshelboy, 2023-08-08:
Hi, I am also interested in this. Are there any updates/docs on this topic?

chshu, 2023-08-31:
@joshelboy @diazmanuel The Zigbee sleepy device example is available here: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/sleepy_end_device

joshelboy, 2023-08-31:
@chshu Great! Needed this for my bachelor thesis, sorry for commenting this on two issues with the same question 😅