espressif/esp-zigbee-sdk #49: Handling custom commands

simonborje, 2023-07-16:
I'm using an ESP32-C6 device in a coordinator role to try to catch custom commands being sent to it from a 3rd party remote control device.

I could not find an example for how to handle such custom commands, but I did find the `esp_zb_add_custom_cluster_command_cb` function in the documentation which sounded promising. I used that to register a callback function, which does get invoked when I press buttons on the remote, but I have not been able to figure out how to interpret the data that is available in the callback.

Is it possible to read out the source address, endpoint, command id and the command payload for a custom command this way or by using some other method?

xieqinan, 2023-07-17:
Hi, 
The custom cluster of ESP Zigbee SDK currently only provides the payload of commands with a custom type, which originates from the remote custom cluster client. Support for additional information such as the source address, endpoint, and command ID will be added in the next release version for users.

Bosemani, 2023-07-17:
Hai @xieqinan 
  I'm developing a Zigbee device using esp32-h2. I compiled and tested the HA Sample Light with zigbee2mqtt as a coordinator.
In ON/OFF Cluster I need ESP_ZB_ZCL_ATTR_ON_OFF_START_UP_ON_OFF attribute. so I created a on/off cluster in custom way.
Here is the code 
` uint8_t on_off = 0,start_onoff =0;
    esp_zb_attribute_list_t *esp_zb_onoff_cluster_cus = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_ON_OFF);
    esp_zb_on_off_cluster_add_attr(esp_zb_onoff_cluster_cus, ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID, &on_off);
    esp_zb_on_off_cluster_add_attr(esp_zb_onoff_cluster_cus, ESP_ZB_ZCL_ATTR_ON_OFF_START_UP_ON_OFF, &start_onoff);`
After creating the endpoint & Register I register the call back `esp_zb_device_add_set_attr_value_cb(attr_cb);` 

`void attr_cb(uint8_t status, uint8_t endpoint, uint16_t cluster_id, uint16_t attr_id, void *new_value)
{
    ESP_LOGI(TAG, "Data received on on/off Cluster attr_id %d EP %d\n",attr_id,endpoint);
    if (cluster_id == ESP_ZB_ZCL_CLUSTER_ID_ON_OFF) {
        uint8_t value = *(uint8_t *)new_value;
        if (attr_id == ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID) {
            /* implemented light on/off control */
            ESP_LOGI(TAG, "on/off light set to %hd Endpoint %d", value,endpoint);
            SET_RELAY1_LED_STATUS(value);
        }
        else{
             ESP_LOGI(TAG, "Data received on on/off Cluster attr_id %d value %hd EP %d\n",attr_id,value,endpoint);
        }
    } else {
        /* Implement some actions if needed when other cluster changed */
        ESP_LOGI(TAG, "cluster:0x%x, attribute:0x%x changed ", cluster_id, attr_id);
    }
}`

ON/OFF attributer Commands like ON, OFF, TOGGLE from Coordinator, I'm able to receive in "attr_cb" callback. But ESP_ZB_ZCL_ATTR_ON_OFF_START_UP_ON_OFF commands I'm not able to receive.

Is anything need to do for that?

Thanks in advance

xieqinan, 2023-07-18:
Hi,
As far as I am concerned, the `START_UP_ON_OFF` attribute is used to indicate the power on state of the on/off cluster. It is not a received command defined by the zigbee spec. To set the value of this attribute, you can use the `esp_zb_zcl_set_attribute_val` function.

Bosemani, 2023-07-18:
Hai @xieqinan 
Yes, you are correct, Using [zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt) I tested ESP32-H2 ON/OFF Cluster. In zigbee2mqtt there is one  functionality called [power_on_behaviour ](https://www.zigbee2mqtt.io/devices/PEHWE20.html#power-on-behavior-enum-l1-endpoint) it uses the `START_UP_ON_OFF `  attribute. If I set power_on_behaviour  to on/off/previous values 0/1/254 will be written in `START_UP_ON_OFF `  attribute in ON/OFF cluster. 
Writing is happening perfectly (I checked it by read the attr after write). But i'm not getting any Event in callback `esp_zb_device_add_set_attr_value_cb(attr_cb);`  in ESP32-H2. I just need the callback event for this, how can do it?

xieqinan, 2023-07-21:
> `esp_zb_device_add_set_attr_value_cb(attr_cb);` in ESP32-H2. I just need the callback event for this, how can do it?

All right, this is a internal bug, we will fix it as soon as possiable.

Bosemani, 2023-07-21:
Hai @xieqinan 
Thank you for conformation.
Let's wait for update!

Bosemani, 2023-08-14:
> > `esp_zb_device_add_set_attr_value_cb(attr_cb);` in ESP32-H2. I just need the callback event for this, how can do it?
> 
> All right, this is a internal bug, we will fix it as soon as possiable.

Hai @xieqinan 
After new release, Its working.

Thank you

closed 2023-08-16 by chshu

xieqinan, 2023-08-22:
Hi @simonborje ,
> I could not find an example for how to handle such custom commands, but I did find the esp_zb_add_custom_cluster_command_cb function in the documentation which sounded promising. I used that to [...]

This issue has been resolved in the internal version, and we are currently testing the solution. We plan to release it within this week. We appreciate your patience and understanding.

xieqinan, 2023-08-30:
Hi @simonborje ,

The latest version of esp-zigbee-sdk introduces additional parameters for custom cluster callbacks. For further guidance on their usage, you can consult [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L252C7-L252C7).

However, please note that while the custom cluster example itself has not been provided, if you find yourself with queries regarding custom clusters, please don't hesitate to open an issue on the repository. This will allow for a discussion to address any questions you may have.