espressif/esp-zigbee-sdk #50: Level commands throwing exceptions

nfnvsc, 2023-07-18:
Hello,

I'm facing some issues running when running ZB level cluster related commands, particularly, the function `esp_zb_zcl_level_move_to_level_with_onoff_cmd_req`. I also tested a few non level cluster functions and they seem to be working as expected.

I'm unsure if I'm missing anything related to the values of the `esp_zb_zcl_move_to_level_cmd_t` struct or if the error is coming from within the library.

For reference and reproducibility, I leave here a sample a sample code with the issue and the output logs observed:

<details><summary>Output logs</summary>
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
0x4001975a: software_reset_cpu in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x181c
load:0x4086c410,len:0xd44
load:0x4086e610,len:0x2df8
entry 0x4086c41a
I (26) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (27) boot: compile time Jul 16 2023 01:09:13
I (28) boot: chip revision: v0.0
I (31) boot.esp32c6: SPI Speed      : 40MHz
I (35) boot.esp32c6: SPI Mode       : DIO
I (40) boot.esp32c6: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 000c8000
I (84) boot:  3 zb_storage       Unknown data     01 81 000df000 00004000
I (91) boot:  4 zb_fct           Unknown data     01 81 000e3000 00000400
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=42068020 size=0c8d8h ( 51416) map
I (132) esp_image: segment 1: paddr=0001c900 vaddr=40800000 size=03718h ( 14104) load
I (140) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=67d80h (425344) map
I (312) esp_image: segment 3: paddr=00087da8 vaddr=40803718 size=0996ch ( 39276) load
I (331) esp_image: segment 4: paddr=0009171c vaddr=4080d090 size=01950h (  6480) load
I (341) boot: Loaded app from partition at offset 0x10000
I (342) boot: Disabling RNG early entropy source...
I (353) cpu_start: Unicore app
I (354) cpu_start: Pro cpu up.
W (363) clk: esp_perip_clk_init() has not been implemented yet
I (370) cpu_start: Pro cpu start user code
I (370) cpu_start: cpu freq: 160000000 Hz
I (370) cpu_start: Application information:
I (373) cpu_start: Project name:     fibonacci-app
I (379) cpu_start: App version:      6c29f81-dirty
I (384) cpu_start: Compile time:     Jul 16 2023 13:52:10
I (390) cpu_start: ELF file SHA256:  5fc227265...
I (395) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND
I (401) cpu_start: Min chip rev:     v0.0
I (406) cpu_start: Max chip rev:     v0.99 
I (411) cpu_start: Chip rev:         v0.0
I (416) heap_init: Initializing. RAM available for dynamic allocation:
I (423) heap_init: At 40814FD0 len 00067640 (413 KiB): D/IRAM
I (429) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (436) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (443) spi_flash: detected chip: generic
I (447) spi_flash: flash io: dio
W (451) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (464) sleep: Configure to isolate all GPIO pins in sleep state
I (471) sleep: Enable automatic switching of GPIO sleep configuration
I (478) coexist: coex firmware version: ebddf30
I (483) coexist: coexist rom version 5b8dcfa
I (488) app_start: Starting scheduler on CPU0
I (493) main_task: Started on CPU0
[... 269 lines cut ...]
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);

    esp_zb_zcl_move_to_level_cmd_t cmd_req;
    cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = 0x0;
    cmd_req.zcl_basic_cmd.dst_endpoint = 1;
    cmd_req.zcl_basic_cmd.src_endpoint = HA_COLOR_DIMMABLE_LIGHT_ENDPOINT;
    cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    cmd_req.level = 0;
    cmd_req.transition_time = 0;
    esp_zb_zcl_level_move_to_level_with_onoff_cmd_req(&cmd_req);

    // esp_zb_zcl_on_off_cmd_t cmd_req;
    // cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = 0x0;
    // cmd_req.zcl_basic_cmd.dst_endpoint = 1;
    // cmd_req.zcl_basic_cmd.src_endpoint = HA_COLOR_DIMMABLE_LIGHT_ENDPOINT;
    // cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    // cmd_req.on_off_cmd_id = 0;
    // esp_zb_zcl_on_off_cmd_req(&cmd_req);
}
```
---

xieqinan, 2023-07-19:
Hi @nfnvsc,

There are two errors in your code.

Error 1, the function `esp_zb_zcl_level_move_to_level_with_onoff_cmd_req(&cmd_req);` in your code is ineffective because the device has not yet joined the Zigbee network.

Error 2, the command `esp_zb_zcl_level_move_to_level_with_onoff_cmd_req()` is intended to send the move to level with on/off effect from the client to the server side. However, in your code, you are sending the command from the server to the client, which causes the program to crash. The Zigbee stack checks the identity of the sender to ensure the command's validity and safety. If you change `esp_zb_cluster_list_add_level_cluster(esp_zb_cluster_list, esp_zb_level_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);` to `esp_zb_cluster_list_add_level_cluster(esp_zb_cluster_list, esp_zb_level_cluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);`, the crash problem will be resolved, but the command will still be invalid(refer to Error 1).

nfnvsc, 2023-07-19:
Hello @xieqinan, 

Thank you for the quick response.

About error 1, sorry about that. I wanted to create a reproducible code and completly forgot about waiting for zigbee to join.
Error 2, indeed, changing `ESP_ZB_ZCL_CLUSTER_SERVER_ROLE` to `ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE` does correct the issue.

Although, I noticed that after swaping the role I was no longer able to receive callbacks from the coordinator (using zigbee2mqtt for testing). 

I'm quite confused on how to able to both set the level on the the device and from the coordinator. 

Thank you.

xieqinan, 2023-07-20:
Hello,

> Although, I noticed that after swaping the role I was no longer able to receive callbacks from the coordinator (using zigbee2mqtt for testing).

The `esp_zb_zcl_level_move_to_level_with_onoff_cmd_req()` is used to send a command that instructs the level cluster server to move to a specified level with on/off functionality. When this command is sent from the level cluster client, the coordinator will receive it if it has the level cluster server.

> I'm quite confused on how to able to both set the level on the the device and from the coordinator.

So, what is your application's sense? In order to satisfy your requirements, it may be necessary for both the end device and the coordinator to create the level control cluster server.

xieqinan, 2023-07-24:
Hi @nfnvsc,

I am pleased to discuss the problem of esp-zigbee-sdk with you here. If the issue has already been resolved, please close it.

closed 2023-07-24 by nfnvsc