espressif/esp-zigbee-sdk #51: Unable to use more than 5 endpoints as a device

nomis, 2023-07-23:
I need at least 34 endpoints but I'm unable to use more than 5 because the rest of them don't appear.

Example project: https://github.com/espressif/esp-zigbee-sdk/compare/main...nomis:esp-zigbee-sdk:multiple-endpoints
I'm running this on an ESP32-C6 connecting to a SONOFF "Zigbee 3.0 USB Dongle plus".

Whatever order I create them in, only the first 5 endpoints show up (and they show up in the order that `esp_zb_ep_list_add_ep()` was called):
[image]

nomis, 2023-07-23:
This value needs to be increased in the zboss library and then recompiled by espressif:
```
#ifndef ZB_MAX_EP_NUMBER
/* max supported EP number, increase if needed */
/** Maximum number of supported endpoints per device */
#define ZB_MAX_EP_NUMBER 5U
#endif
```

xieqinan, 2023-07-24:
@nomis , Thank you for your information; the MACRO will be added as soon as possible.

closed 2023-07-31 by espressif-bot

xieqinan, 2023-07-31:
Hi,
The `ZB_MAX_EP_NUMBER` has been enlarged to 35 in the recent version. Distributing your endpoints to more devices may be more helpful in implementing your application.

For other topics related to esp-zigbee-sdk, they can be discussed in an new issue.

nomis, 2023-07-31:
> Distributing your endpoints to more devices may be more helpful in implementing your application.

No, the zboss SDK needs to be modified so that memory is allocated dynamically and this arbitrary limit is removed. The setting only exists because it assumes you have the source code and can rebuild it to change the fixed size arrays.