espressif/esp-zigbee-sdk #52: The `status` parameter in `esp_zb_set_attr_callback_t` should be an output, not an input

nomis, 2023-07-23:
In the `esp-zboss-lib`, the device callback struct has `status` as a return value:
```
typedef struct zb_zcl_device_callback_param_s
{
  /** Type of device callback (see @ref zb_zcl_device_callback_id_e) */
  zb_zcl_device_callback_id_t device_cb_id;
  zb_uint8_t endpoint;
  zb_zcl_attr_access_t attr_type;
  /** Return status (see zb_ret_t) */
  zb_ret_t status;
...
```

In `esp-zigbee-sdk`, the set attribute callback has it as an input:
```
/** ZCL set attribute callback
 *
 * @param[in] status The status for ZCL set attribute 0 - status success
 * @param[in] endpoint The endpoint of this cluster lists to change
 * @param[in] cluster_id Cluster ID
 * @param[in] attr_id Attribute ID
 * @param[in] value The pointer of the attribute data
 *
 * @note The Zigbee device callback is the default ZCL cluster handler to notify the application about a certain event.
 * It covers attribute changes like on-off, level and color cluster.
 */
typedef void (*esp_zb_set_attr_callback_t)(uint8_t status, uint8_t endpoint, uint16_t cluster_id, uint16_t attr_id, void *value);
```

Shouldn't `status` be an output to indicate if an attribute has been set to a valid value or not?

xieqinan, 2023-07-24:
Yes, you are right.  The `status`, `endpoint`, `cluster_id`, `attr_id`, and `value` should are the output properties.

nomis, 2023-08-06:
I think `endpoint`, `cluster_id`, `attr_id`, and `value` should remain inputs?

You've got the same problem with `esp_zb_custom_cluster_cmd_callback_t`, the `status` should be an output.

xieqinan, 2023-08-07:
I think that  the `[in]` is used when you want to pass parameters as input for the rule. `[out]` is used when you want to send the parameter as output. So `endpoint`, `cluster_id`, `attr_id`, and `value` also should are the output properties.

nomis, 2023-08-07:
`[in]` and `[out]` refer to the function itself, so only `status` is an output of the function and it must be a pointer. This applies to both callback functions.

xieqinan, 2023-08-08:
Thank you for the reminder. I will address and resolve the issue.

xieqinan, 2023-08-14:
Hi, 
The parameters of `esp_zb_set_attr_callback_t`  has been changed. Do you have  other questions for this issue, if not, please close it.