espressif/esp-zigbee-sdk #55: Is there any API for discover attributes?

Zmmfly, 2023-07-25:
Hi, i used this library to develop a ZigBee gateway.
For the reason of want connect to most zigbee device, I need to discover the attributes for cluster.
However, I read all the examples and didn't find some api that can do this, there is something like `esp_zb_zdo_find_on_off_light`, but it's not what I want.
I searched for information on Google again and found the ZCL command 0x0c, but in the online documentation of esp_zigbee_lib, I did not find the implementation of this command; there is a related command `esp_zb_zcl_get_attribute`, but this command is only for local.

xieqinan, 2023-07-26:
Hi,
Could you please provide a detailed description of the method you would like to use for discovering attributes for a cluster in a Zigbee network? Specifically, I'd like to know what parameters you can provide to find the attribute.

The ESP-Zigbee-SDK offers two essential functions to aid in this process: `esp_zb_zdo_active_ep_req()` is used to search for endpoints in the Zigbee network, and `esp_zb_zdo_simple_desc_req()` is employed to find the clusters associated with specific endpoints. Are you interested in a command that allows you to retrieve an attribute list based on a specific endpoint ID and cluster ID?

Zmmfly, 2023-07-29:
Hi @xieqinan, thanks for your reply.

> Specifically, I'd like to know what parameters you can provide to find the attribute.

Yes, I can provide node address (16-bit short address), endpoint number, cluster ID, callback function, callback function context.

> `esp_zb_zdo_active_ep_req()` is used to search for endpoints in the Zigbee network, and `esp_zb_zdo_simple_desc_req()` is employed to find the clusters associated with specific endpoints.

I have used these two functions, and the clusterID list provided by `esp_zb_af_simple_desc_1_1_t` as the input parameter of the callback function only contains two clusterIDs, which is too few, and some devices have many clusterIDs under one endpoint. Also, I also did not find an API that can discover all clusterIDs.

> Are you interested in a command that allows you to retrieve an attribute list based on a specific endpoint ID and cluster ID?

Yes, I am interested in finding attribute by specific endpoint and specific clusterID.

In the documentation, there are many APIs related to HA creation, cluster creation, and attribute creation, but the APIs related to HA discovery, cluster discovery, and attribute discovery are not perfect. This is friendly for creating End devices, but unfriendly for creating coordinators.

I'm not very familiar with Zigbee, so sorry for any mistakes or omissions.

xieqinan, 2023-07-31:
Hi,
> I have used these two functions, and the clusterID list provided by `esp_zb_af_simple_desc_1_1_t` as the input parameter of the callback function only contains two clusterIDs, which is too few, and [...]

The `esp_zb_af_simple_desc_1_1_t` provides a clusterID list whose size is `app_input_cluster_count` + `app_output_cluster_count`, rather than containing only two clusterIDs.

> In the documentation, there are many APIs related to HA creation, cluster creation, and attribute creation, but the APIs related to HA discovery, cluster discovery, and attribute discovery are not perfect. This is friendly for creating End devices, but unfriendly for creating coordinators.

Thank you for your feedback and suggestion, the discovery feature will be added in August.

closed 2023-09-11 by espressif-bot

xieqinan, 2023-09-11:
Hello,
The API for discovering attributes has been supported in [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L994C6-L994C34).

petergjoel, 2023-12-30:
@xieqinan, can you give a hint on how to correctly use this API?
I have the following code:
```C
                                esp_zb_zcl_disc_attr_cmd_t cmd;
                                cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
                                cmd.cluster_id = simple_desc->app_cluster_list[i]; // <-- id from esp_zb_zdo_simple_desc_req
                                cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
                                cmd.max_attr_number = 10;
                                cmd.start_attr_id = 0;
                                cmd.zcl_basic_cmd.dst_addr_u.addr_short = 0x5118; // <-- matches device
                                cmd.zcl_basic_cmd.dst_endpoint = 1;
                                cmd.zcl_basic_cmd.src_endpoint = 1;
                                auto seq = esp_zb_zcl_disc_attr_cmd_req(&cmd);
```

but in my callback on `esp_zb_core_action_handler_register` I get `status_code=0x82` (ESP_ZB_ZCL_STATUS_UNSUP_GEN_CMD).

I'm trying using a Sonoff SNZB-02 sensor as my test-device currently; my final application requires me to discover the attributes.

Cycov, 2024-01-16:
It's odd that such a basic example does not exist. A simple coordinator that forms a network and then allows joining of new devices forever and display all the nodes, clusters and attributes of each such connected device in a nice, easy-to-read way.

And this example not be be locked behind closed API, like the `esp_zb_zdo_find_on_off_light` function is. Too much stuff is behind the closed source and without proper documentation and examples for the exposed cryptic functions. Also, why is this issue closed?

In my case, I'm trying to have my coordinator allow any type of off-the-shelf home automation (HA) zigbee device to connect and then read their data.

Daniel-Craciun, 2025-01-05:
I used this code and it worked for me. And then I used the callback_id `ESP_ZB_CORE_CMD_DISC_ATTR_RESP_CB_ID` in my callback for the `esp_zb_core_action_handler_register` function.

`
  esp_zb_zcl_disc_attr_cmd_t attr_cmd;
  attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
  attr_cmd.zcl_basic_cmd.src_endpoint = ESP_ZB_GATEWAY_ENDPOINT;
  attr_cmd.zcl_basic_cmd.dst_endpoint = simple_desc->endpoint;
  attr_cmd.zcl_basic_cmd.dst_addr_u.addr_short = zb_sensor_to_connect.short_addr;
  attr_cmd.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_BASIC;
  attr_cmd.start_attr_id = 0x00;
  attr_cmd.max_attr_number = 200;

  esp_zb_lock_acquire(portMAX_DELAY);
  esp_zb_zcl_disc_attr_cmd_req(&attr_cmd);
  esp_zb_lock_release();
`