espressif/esp-zigbee-sdk #59: Adding HVAC related Cluster Support

forthchina, 2023-08-01:
The Cluster Library Specification Document 07-5123 Revision 8 lists 5 HVAC clusters with ID 0x0200 - 0x0204. I think at least below needs to be implement in ESP32-H2:
- 0x0201 Thermostat
- 0x0202 Fan Control and
- 0x0204 Thermostat User Interface Configuration

OR 
A method for user-self implement any ZCL may provided.

xieqinan, 2023-08-02:
Hi @forthchina 
These clusters provided by you will be added as soon as possiable.

closed 2023-10-11 by espressif-bot

xieqinan, 2023-10-11:
Hello @forthchina 

The [Thermostat Device](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_thermostat.h) has been supported in esp-zigbee-lib v1.0.0, Could you please try using and testing it?