espressif/esp-zigbee-sdk #61: factory reset not working

diazmanuel, 2023-08-01:
hello 
I am trying to implement a factory reset, for this I use the function  
`esp_zb_factory_reset() `
but after the restart the device continues to pair with the network coordinator, I also tried to use the function
`esp_zb_nvram_erase_at_start(true)`
and then call 
`esp_zb_factory_reset() `
, but in both cases it keeps connecting to the network, I assume that the nvram is not being cleared, any suggestions?

xieqinan, 2023-08-02:
Hi @diazmanuel,
Could you please provide information about the network topology of your application? Additionally, if the end device is the only one being erased, it will try to find and join Zigbee network each time it starts up.

To verify your assumption, could you try erasing the flash for the coordinator while keeping the end device unchanged?

diazmanuel, 2023-08-02:
If that was happening, the coordinator was letting it in even though the end device wiped the memory, I'm guessing it's because it remembers the mac address of the device. Thanks for the help

closed 2023-08-02 by diazmanuel