espressif/esp-zigbee-sdk #62: abort after using a zigbee alarm

diazmanuel, 2023-08-01:
Hello,

I am testing the operation of my device configured as a repeater, for this I disconnect the coordinator from the network and restart my repeater,

after the restart in the ESP_ZB_BDB_SIGNAL_STEERING signal  I  try to communicate with the coordinator through 

```
            esp_zb_zdo_ieee_addr_req_param_t ieee_req;
            ieee_req.addr_of_interest = 0;
            ieee_req.dst_nwk_addr = 0;
            ieee_req.request_type = 0;
            ieee_req.start_index = 0;
            esp_zb_zdo_ieee_addr_req(&ieee_req, ieee_cb, NULL);
```
then in the ieee_cb() handler I make a loop in case the request to the coordinator fails

```

static void bdb_start_ieee_addr_req_cb(uint8_t param)
{
    esp_zb_zdo_ieee_addr_req_param_t ieee_req;
    ieee_req.addr_of_interest = 0;
    ieee_req.dst_nwk_addr = 0;
    ieee_req.request_type = 0;
    ieee_req.start_index = 0;
    esp_zb_zdo_ieee_addr_req(&ieee_req, ieee_cb, NULL);
}

static void ieee_cb(esp_zb_zdp_status_t zdo_status, esp_zb_ieee_addr_t ieee_addr, void *user_ctx)
{
    
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {

        esp_zb_zdo_bind_req_param_t bind_req;
        esp_zb_get_long_address(bind_req.src_address);

        bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        memcpy(bind_req.dst_address_u.addr_long, ieee_addr, sizeof(esp_zb_ieee_addr_t));

        ESP_LOGI(TAG, "local address is: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x",bind_req.src_address[7], bind_req.src_address[6], bind_req.src_address[5],bind_req.src_address[4], bind_req.src_address[3], bind_req.src_address[2],bind_req.src_address[1], bind_req.src_address[0]);

    }else{
        ESP_LOGW(TAG,"unable to communicate with coordinator");
        esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_ieee_addr_req_cb, 0, ZIGBEE_CONNECTION_TO_CORDINATOR_DELAY_MS);
        
    }
}
```

but when I reconnect the coordinator and it responds to the ieee_request the device reboots with the following log

```
I (126827) ESP_ZIGBEE_API_ZDO: ZDO IEEE address request timeout!
E (126827) taskZigbee: unable to communicate with coordinator
W (129541) taskZigbee: ZDO signal: 48, status: 0
W (129555) taskZigbee: ZDO signal: 2, status: 0
W (132375) taskZigbee: ZDO signal: 50, status: 0
W (133786) ESP_ZIGBEE_API_ZDO: The tsn of callback is not found
W (133786) ESP_ZIGBEE_API_ZDO: The tsn of callback is not found and can't remove
W (133790) taskZigbee: ZDO signal: 60, status: 0
W (135247) taskZigbee: ZDO signal: 50, status: 0
W (135843) taskZigbee: ZDO signal: 48, status: 0
W (135849) taskZigbee: ZDO signal: 2, status: 0
W (135982) taskZigbee: ZDO signal: 24, status: 0
I (136897) taskZigbee: ieee_cb address is: f4:ce:36:12:c8:70:0d:af
I (136898) taskZigbee: local address is: 00:00:48:99:f6:f9:55:60
I (136900) taskZigbee: local short address is: 0x28fe
W (137619) taskZigbee: ZDO signal: 2, status: 0
I (143157) taskZigbee: Incomming Report | address: 0x68c8, ID: 26824,src_endpoint:1, des_endpoint:14, cluster_id:0xfdee, attr_type:0x29, attr_id:0x0, value:1
W (144362) taskZigbee: ZDO signal: 48, status: 0
[... 114 lines cut ...]
I (581) cpu_start: Application information:
I (584) cpu_start: Project name:     Multizona_G2
I (589) cpu_start: App version:      3641f02-dirty
I (595) cpu_start: Compile time:     Jul 31 2023 16:39:52
I (601) cpu_start: ELF file SHA256:  ddfe5906a...
I (606) cpu_start: ESP-IDF:          v5.2-dev-823-g903af13e84
I (613) cpu_start: Min chip rev:     v0.0
I (617) cpu_start: Max chip rev:     v0.99 
I (622) cpu_start: Chip rev:         v0.0
I (627) heap_init: Initializing. RAM available for dynamic allocation:
I (634) heap_init: At 4081C1D0 len 00060440 (385 KiB): D/IRAM
I (640) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (647) heap_init: At 50000010 len 00003FF0 (15 KiB): RTCRAM
I (654) spi_flash: detected chip: generic
I (658) spi_flash: flash io: dio
I (665) sleep: Configure to isolate all GPIO pins in sleep state
I (669) sleep: Enable automatic switching of GPIO sleep configuration
I (676) coexist: coex firmware version: ebddf30
I (681) coexist: coexist rom version 5b8dcfa
I (687) app_start: Starting scheduler on CPU0
I (691) main_task: Started on CPU0
I (695) main_task: Calling app_main()
```

Any idea what might be causing this behavior?

xieqinan, 2023-08-02:
Hi @diazmanuel ,

The issue may be caused by using an older version of esp-zigbee-lib. The latest version available is v0.7.4. If you are not already using this version, please update to it.

diazmanuel, 2023-08-02:
hi @xieqinan thanks thats solve the problem