espressif/esp-zigbee-sdk #65: Updating attribute values with `esp_zb_zcl_set_attribute_val()` doesn't work reliably or for more than ~15 endpoints

nomis, 2023-08-02:
I'm updating my own on/off endpoints with `esp_zb_zcl_set_attribute_val()` in response to GPIOs but this doesn't work reliable or after the first ~15 endpoints.

Example project: https://github.com/espressif/esp-zigbee-sdk/compare/main...nomis:esp-zigbee-sdk:update-attribute-values

For testing, this project will create 35 lights and when one of them is toggled it will toggle the next one.

Toggle each of the light endpoints and you'll see that the 15th light (which should toggle the 16th) doesn't have that effect. The attribute has been changed locally because toggling the 16th light has no effect (nothing is logged) but that attribute change hasn't been transmitted. (Coordinator logs show that the message to change the attribute isn't received.)

If you turn some of the lights on and then turn them all off it can also get out of sync so that actions that should toggle the next light do nothing because the next light was already the opposite state from what is displayed. This needs to work reliably. The local state of the device must be updated on the coordinator even if many changes are happening at once.

I'd also like to be confident that `esp_zb_zcl_set_attribute_val()` will retry even if the network communication fails. There already appears to be a lack of synchronisation for the existing attributes (to turn them all off) if I just restart the device.

nomis, 2023-08-03:
> You attempted to simulate the control of a light using `esp_zb_zcl_set_attribute_val()`. However, you observed that when switching the 15th light, the coordinator did not issue the command.

Yes, switching the 15th light using the coordinator did not result in the 16th light changing at the coordinator.

> Subsequently, you proceeded to send the command to switch the 16th light, but on the end device side, it still displayed control over the 15th light.

No, the device had the 16th light as "on" but the coordinator state for it was "off". When the coordinator set the light "on" there was no event on the device because the light was already on. This is expected behaviour but the light state shouldn't have been stale.

> You are suggesting that `esp_zb_zcl_set_attribute_val()` should transmit back the state of the set light to the coordinator in order to establish a reliable control relationship. Is this your understanding?

Yes.

> If that's the case, there might be a misunderstanding. esp_zb_zcl_set_attribute_val() does not have this capability; it only sets attribute values locally. To establish a stable control relationship, you should use esp_zb_zcl_read_attr_cmd_req() on the coordinator side to retrieve the values from the end device.

The coordinator is a SONOFF "Zigbee 3.0 USB Dongle Plus V2", not an ESP32.

This works for the first 15 endpoints and it works for other devices (although those only have 2 endpoints).

Here's a log where I turn on light 1 (endpoint 10) and receive the message that light 2 has also been turned on on the device: [esp-zigbee-sdk-65_TZ-220_log1_light1.txt](https://github.com/espressif/esp-zigbee-sdk/files/12253836/esp-zigbee-sdk-65_TZ-220_log1_light1.txt)

Here's a log where I turn on light 19 (endpoint 29) and don't receive the message that light 20 has been turned on: [esp-zigbee-sdk-65_TZ-220_log2_light19.txt](https://github.com/espressif/esp-zigbee-sdk/files/12253837/esp-zigbee-sdk-65_TZ-220_log2_light19.txt)

nomis, 2023-08-04:
I'm also getting a 5 second per-endpoint delay updating the endpoints that do work.

For comparison, the SONOFF S26R2ZB can update with a 1 second interval.

This is probably caused by `ZB_ZCL_MIN_REPORTING_INTERVAL_DEFAULT` with no documented way to change it. The only example for `esp_zb_zcl_config_report_cmd_req()` is with a bind callback, but I'm not binding two devices together.

nomis, 2023-08-05:
Now that I've connected up GPIOs so that they affect the endpoints, the 4th and 9th endpoints can be updated with no delay. I can repeatedly toggle them very quickly and it shows up at the coordinator immediately.

The 14th and 19th endpoints will only update at a 5 second interval.

I've also got the 23rd endpoint updating but the 18th endpoint won't update at all. Sometimes it works for certain endpoints and sometimes it doesn't. I've previously had endpoints that I could update stop being updateable after I toggle them manually from the coordinator. This is what I mean by it being unreliable.

All I'm doing is calling `esp_zb_zcl_set_attribute_val()` from the zigbee thread (using `esp_zb_scheduler_alarm()`).

nomis, 2023-08-07:
It looks like the coordinator needs to request reporting of the attributes with a min interval of 0 (which it does) but I have no way to debug this on the device. There's no callback to tell me when there's a change to the reporting configuration for any of the endpoints/attributes.

COM8, 2023-08-12:
I'm seeing the same thing. I have temperature and humidity endpoints and I'm unable to publish my values only every 30 seconds, which in my case is not enough.
My code: https://github.com/electronics4fun/HASS_EnviromentSensor/blob/1a2d6a4ad47ff8c3ed1e40e9d89262c3af563147/esp32_h2/main/zigbee/ZDevice.cpp#L68-L76

nomis, 2023-08-12:
Looking at the ZHA code, it requests a different minimum interval based on the attribute type. This is 30 seconds for sensor-type attributes. As far as I can tell there is no option in the integration between ZHA and HA to configure it differently. I have modified the ZHA code to request a 0 second interval for the Binary Input attribute that I'm using, but that will be overwritten on upgrade.

I think my problems are caused by a request for reporting from the coordinator failing and not being retried but I will need to check the debug logs and try a custom ESP32 coordinator to verify that it is possible with many endpoints. Some of my endpoints can update at 0 second intervals, some of them at 5 second intervals and others never update but ZHA should be requesting a 0 second minimum interval for all of them. It's not consistent.

xieqinan, 2023-08-14:
Hi @COM8 ,
I can not open your code link. For your issue, we need more discussion.

xieqinan, 2023-08-14:
Hi @nomis ,

> There's no callback to tell me when there's a change to the reporting configuration for any of the endpoints/attributes.

The API for reading report configuration will be made available this week. This will allow you to get any changes in the reporting configuration for attributes.

>  Some of my endpoints can update at 0 second intervals, some of them at 5 second intervals and others never update

For this issue, I also will confirm it. Are you doubting that the interval time of the report is related to the endpoint ID? I will test it on your explanation.

nomis, 2023-08-15:
> > Some of my endpoints can update at 0 second intervals, some of them at 5 second intervals and others never update
> 
> For this issue, I also will confirm it. Are you doubting that the interval time of the report is related to the endpoint ID? I will test it on your explanation.

I have 23 endpoints in https://github.com/nomis/candle-dribbler when `NUTT_MAX_LIGHTS=4` ("Component config" -> "Candle Dribbler" -> "Number of lights supported").

If I rapidly connect/disconnect one of the switch GPIOs (3, 2, 11, 10) to GND, then currently I observe in ZHA:
* GPIO 3 (Light 1)
  * Light Endpoint (4th on device, 8th on ZHA, number 11) - updates every time with no minimum interval
  * Binary Input Endpoint (7th on device, 4th on ZHA, number 61) - updates every time with no minimum interval
* GPIO 2 (Light 2)
  * Light Endpoint (9th on device, 12th on ZHA, number 12) - updates with a minimum interval of 5 seconds
  * Binary Input Endpoint (12th on device, 5th on ZHA, number 62) - updates every time with no minimum interval
* GPIO 11 (Light 3)
  * Light Endpoint (14th on device, 16th on ZHA, number 13) - updates with a minimum interval of 5 seconds
  * Binary Input Endpoint (17th on device, 6th on ZHA, number 63) - updates every time with no minimum interval
* GPIO 10 (Light 4)
  * Light Endpoint (19th on device, 20th on ZHA, number 14) - updates with a minimum interval of 5 seconds
  * Binary Input Endpoint (22nd on device, 7th on ZHA, number 64) - updates with a minimum interval of 5 seconds

The lower numbered/positioned endpoints more reliably have a working lower minimum interval. I suspect that ZHA is failing to configure the reporting interval on one or more of the endpoints and they're still using the default of 5 seconds. If I could just default them to 0 seconds on the device it would avoid this type of problem.

(Are the reconfigured reporting intervals stored in flash or does it have to be reconfigured every time the device starts? That could be a problem for anyone developing a battery powered device.)

I have modified ZHA to use a minimum interval of 0 seconds for the binary input endpoint (otherwise it uses 30 seconds):

`homeassistant/components/zha/core/channels/general.py`:
```
@registries.ZIGBEE_CHANNEL_REGISTRY.register(general.BinaryInput.cluster_id)
class BinaryInput(ZigbeeChannel):
    """Binary Input channel."""

    REPORT_CONFIG = (
        AttrReportConfig(attr="present_value", config=REPORT_CONFIG_IMMEDIATE),
    )
```

ZHA is aware of the order in which I've set up the endpoints because they're in that order when I look at the list of clusters, but it also sorts them by the name of the cluster type:

[image]

nomis, 2023-08-16:
A related problem is that if I turn all the light endpoints on and then restart the end device (which means all the endpoints should now be off) it's taking forever to synchronise the change. It took 3 minutes for a single endpoint to show as off and some of them took 15 minutes while some are still on 30 minutes later. If I explicitly read the `on_off` attribute it will have the correct (`false`) value.

If I power cycle a SONOFF R26R2ZB socket while it's single endpoint is "on" it will show as "off" within a second of turning it back on.

nomis, 2023-08-31:
While debugging #86, this behaviour exactly 5 seconds after starting to connect explains why I get spurious "device unavailable" errors before the connection to the network is established and probably explains why the status after restarting the device isn't updated... it's trying to report the status of all endpoints using the default minimum reporting interval before the connection is established.
```
2023-08-31 22:40:48.304154 D (234) nutt.ZigbeeDevice: Zigbee stack initialized
2023-08-31 22:40:48.307376 I (234) nutt.ZigbeeDevice: Device address: 40:4c:ca:ff:fe:43:0b:d0/0x6de5 (network configured)
2023-08-31 22:40:48.312591 D (234) nutt.ZigbeeDevice: Connecting (ZDO Skip Start Up)
2023-08-31 22:40:48.317739 D (241) esp_core_dump_flash: Blank core dump partition!
2023-08-31 22:40:48.320905 D (244) nutt: Free stack: 1316
2023-08-31 22:40:48.343502 D (274) nutt.Device: Refresh
2023-08-31 22:40:48.343859 D (274) nutt.Device: Refresh
2023-08-31 22:40:48.344043 D (274) nutt.Device: Refresh
2023-08-31 22:40:48.344183 D (274) nutt.Device: Refresh
2023-08-31 22:40:53.294570 D (5224) nutt: zb_zcl_send_report_attr_command(0x4081e310 { direction=0, ep=61, cluster_id=15 cluster_role=1 attr_id=111 flags=7 }, 23)
2023-08-31 22:40:53.302741 W (5224) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.314780 D (5224) nutt: zb_zcl_send_report_attr_command(0x4081e340 { direction=0, ep=61, cluster_id=15 cluster_role=1 attr_id=85 flags=71 }, 23)
2023-08-31 22:40:53.323158 W (5244) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.334762 D (5244) nutt: zb_zcl_send_report_attr_command(0x4081ef7c { direction=0, ep=12, cluster_id=6 cluster_role=1 attr_id=0 flags=71 }, 23)
2023-08-31 22:40:53.343094 W (5264) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.354966 D (5264) nutt: zb_zcl_send_report_attr_command(0x4081f2f8 { direction=0, ep=22, cluster_id=6 cluster_role=1 attr_id=0 flags=71 }, 23)
2023-08-31 22:40:53.363373 W (5284) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.374895 D (5284) nutt: zb_zcl_send_report_attr_command(0x4081f674 { direction=0, ep=32, cluster_id=6 cluster_role=1 attr_id=0 flags=71 }, 23)
2023-08-31 22:40:53.383253 W (5304) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.395209 D (5314) nutt: zb_zcl_send_report_attr_command(0x4081f898 { direction=0, ep=62, cluster_id=15 cluster_role=1 attr_id=111 flags=7 }, 23)
2023-08-31 22:40:53.403449 W (5324) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.415142 D (5334) nutt: zb_zcl_send_report_attr_command(0x408208d0 { direction=0, ep=23, cluster_id=6 cluster_role=1 attr_id=0 flags=71 }, 23)
2023-08-31 22:40:53.423488 W (5344) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.435190 D (5354) nutt: zb_zcl_send_report_attr_command(0x40820c4c { direction=0, ep=33, cluster_id=6 cluster_role=1 attr_id=0 flags=71 }, 23)
2023-08-31 22:40:53.443623 W (5364) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.455500 D (5374) nutt: zb_zcl_send_report_attr_command(0x40820e70 { direction=0, ep=63, cluster_id=15 cluster_role=1 attr_id=111 flags=7 }, 23)
2023-08-31 22:40:53.463807 W (5384) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.475732 D (5394) nutt: zb_zcl_send_report_attr_command(0x40820ea0 { direction=0, ep=63, cluster_id=15 cluster_role=1 attr_id=85 flags=71 }, 23)
2023-08-31 22:40:53.484029 W (5404) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.495775 D (5414) nutt: zb_zcl_send_report_attr_command(0x40821adc { direction=0, ep=14, cluster_id=6 cluster_role=1 attr_id=0 flags=71 }, 23)
2023-08-31 22:40:53.504044 W (5424) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.515793 D (5434) nutt: zb_zcl_send_report_attr_command(0x40821e58 { direction=0, ep=24, cluster_id=6 cluster_role=1 attr_id=0 flags=71 }, 23)
2023-08-31 22:40:53.524225 W (5444) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.535869 D (5454) nutt: zb_zcl_send_report_attr_command(0x408221d4 { direction=0, ep=34, cluster_id=6 cluster_role=1 attr_id=0 flags=71 }, 23)
2023-08-31 22:40:53.544053 W (5464) nutt.ZigbeeDevice: Device unavailable (e0:79:8d:ff:fe:b3:dc:82/0x0000)
2023-08-31 22:40:53.556081 D (5474) nutt: zb_zcl_send_report_attr_command(0x408223f8 { direction=0, ep=64, cluster_id=15 cluster_role=1 attr_id=111 flags=7 }, 23)
2023-08-31 22:40:56.554041 D (8474) nutt.ZigbeeDevice: Connecting (BDB Device Reboot)
2023-08-31 22:40:56.554444 I (8484) nutt.ZigbeeDevice: Joined network successfully (6f:06:fc:4a:5b:92:dd:9c/0x8813@26 as 0x6de5)
```

nomis, 2023-09-02:
By wrapping `zb_zcl_send_report_attr_command()` I can see that it thinks it's sending attribute reports for all of the endpoints that are changed on the device itself (71 to 74) but I only receive one of these at the co-ordinator. It's consistently endpoint 74 that always works and the others that never work:

```
2023-09-02 11:43:47.020411 D (1913444) nutt.Light: Light 1 set persistent enable 1 -> 0
2023-09-02 11:43:47.029468 D (1913454) nutt.Light: Light 1 set temporary enable 1 -> 0 (auto)
2023-09-02 11:43:47.033451 I (1913454) nutt.Light: Light 1 update state 0 -> 0
2023-09-02 11:43:47.033529 D (1913454) nutt.Device: Refresh
2023-09-02 11:43:47.038907 D (1913454) nutt.Light: Light 1 report temporary enable 0
2023-09-02 11:43:47.051530 D (1913464) nutt.ZigbeeDevice: zb_zcl_send_report_attr_command: ep=71 cluster_id=6 cluster_role=1 attr_id=0 manuf_code=65535 attr=0x408210fe
2023-09-02 11:43:47.064143 D (1913474) nutt.ZigbeeDevice: zb_zcl_send_report_attr_command: ep=81 cluster_id=6 cluster_role=1 attr_id=0 manuf_code=65535 attr=0x4082147a
2023-09-02 11:43:53.480378 D (1919904) nutt.Light: Light 2 set persistent enable 1 -> 0
2023-09-02 11:43:53.489535 D (1919914) nutt.Light: Light 2 set temporary enable 1 -> 0 (auto)
2023-09-02 11:43:53.493322 I (1919914) nutt.Light: Light 2 update state 0 -> 0
2023-09-02 11:43:53.493365 D (1919914) nutt.Device: Refresh
2023-09-02 11:43:53.499049 D (1919914) nutt.Light: Light 2 report temporary enable 0
2023-09-02 11:43:53.511393 D (1919924) nutt.ZigbeeDevice: zb_zcl_send_report_attr_command: ep=72 cluster_id=6 cluster_role=1 attr_id=0 manuf_code=65535 attr=0x408226d2
2023-09-02 11:43:53.524178 D (1919934) nutt.ZigbeeDevice: zb_zcl_send_report_attr_command: ep=82 cluster_id=6 cluster_role=1 attr_id=0 manuf_code=65535 attr=0x40822a4e
2023-09-02 11:43:55.710578 D (1922134) nutt.Light: Light 3 set persistent enable 1 -> 0
2023-09-02 11:43:55.719702 D (1922144) nutt.Light: Light 3 set temporary enable 1 -> 0 (auto)
2023-09-02 11:43:55.723456 I (1922144) nutt.Light: Light 3 update state 0 -> 0
2023-09-02 11:43:55.723531 D (1922144) nutt.Device: Refresh
2023-09-02 11:43:55.729016 D (1922144) nutt.Light: Light 3 report temporary enable 0
2023-09-02 11:43:55.741492 D (1922154) nutt.ZigbeeDevice: zb_zcl_send_report_attr_command: ep=73 cluster_id=6 cluster_role=1 attr_id=0 manuf_code=65535 attr=0x40823c5e
2023-09-02 11:43:55.754395 D (1922164) nutt.ZigbeeDevice: zb_zcl_send_report_attr_command: ep=83 cluster_id=6 cluster_role=1 attr_id=0 manuf_code=65535 attr=0x40823fda
2023-09-02 11:43:58.070443 D (1924494) nutt.Light: Light 4 set persistent enable 1 -> 0
2023-09-02 11:43:58.079784 D (1924504) nutt.Light: Light 4 set temporary enable 1 -> 0 (auto)
2023-09-02 11:43:58.083451 I (1924504) nutt.Light: Light 4 update state 0 -> 0
2023-09-02 11:43:58.083527 D (1924504) nutt.Device: Refresh
2023-09-02 11:43:58.089012 D (1924504) nutt.Light: Light 4 report temporary enable 0
2023-09-02 11:43:58.101510 D (1924514) nutt.ZigbeeDevice: zb_zcl_send_report_attr_command: ep=74 cluster_id=6 cluster_role=1 attr_id=0 manuf_code=65535 attr=0x408251e6
2023-09-02 11:43:58.151775 D (1924574) nutt.ZigbeeDevice: zb_zcl_send_report_attr_command: ep=84 cluster_id=6 cluster_role=1 attr_id=0 manuf_code=65535 attr=0x40825562
```

```
2023-09-02 11:43:46.989 DEBUG (MainThread) [zigpy.zcl] [0x6DE5:81:0x0006] Sending request: off()
2023-09-02 11:43:46.989 DEBUG (MainThread) [zigpy.device] [0x6de5] Extending timeout for 0x7f request
2023-09-02 11:43:46.989 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=81, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6DE5), dst_ep=81, source_route=[0xfa82], extended_timeout=True, tsn=127, profile_id=260, cluster_id=6, data=Serialized[b'\x01\x7f\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-09-02 11:43:47.070 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=81, destinationEndpoint=81, options=<EmberApsOption.APS_OPTION_DESTINATION_EUI64|APS_OPTION_SOURCE_EUI64|APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 3392>, groupId=0, sequence=236), 156, -61, 0x6de5, 255, 255, b'\x18\x7f\x0b\x00\x00']
2023-09-02 11:43:47.070 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6DE5), src_ep=81, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=81, source_route=None, extended_timeout=False, tsn=236, profile_id=260, cluster_id=6, data=Serialized[b'\x18\x7f\x0b\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=156, rssi=-61)
2023-09-02 11:43:47.070 DEBUG (MainThread) [zigpy.zcl] [0x6DE5:81:0x0006] Received ZCL frame: b'\x18\x7f\x0b\x00\x00'
2023-09-02 11:43:47.070 DEBUG (MainThread) [zigpy.zcl] [0x6DE5:81:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=127, command_id=11, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[... cut ...]
2023-09-02 11:43:58.138 DEBUG (MainThread) [zigpy.zcl] [0x6DE5:74:0x0006] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2023-09-02 11:43:58.139 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=74, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6DE5), dst_ep=74, source_route=[0xfa82], extended_timeout=False, tsn=71, profile_id=260, cluster_id=6, data=Serialized[b'\x18G\x0b\n\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
```
```

nomis, 2023-09-10:
I think I'm reaching one of the `ZB_*_BINDING_TABLE_SIZE` limits but I'm not sure why because I think there should be 24 entries but the limit is 32?

```
2023-09-10 15:41:53.654 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6C1B:73:0x0006]: bound 'on_off' cluster: Status.TABLE_FULL
2023-09-10 15:41:53.670 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6C1B:81:0x0006]: bound 'on_off' cluster: Status.TABLE_FULL
2023-09-10 15:41:53.731 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6C1B:82:0x0006]: bound 'on_off' cluster: Status.TABLE_FULL
2023-09-10 15:41:53.810 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6C1B:84:0x0006]: bound 'on_off' cluster: Status.TABLE_FULL
```

nomis, 2023-09-12:
Could you provide an example of how to use `esp_zb_zdo_binding_table_req()` properly?

I get 3 callbacks in response to most of my requests, so I have to use the scheduler to de-duplicate requests for the next entry to stop it getting completely out of control:
```
void ZigbeeDevice::print_binding_table() {
	ESP_LOGI(TAG, "Binding table:");
	print_binding_table_next(0);
}

void ZigbeeDevice::print_binding_table_next(uint8_t index) {
	esp_zb_zdo_mgmt_bind_param_t req{};
	req.start_index = index;
	req.dst_addr = esp_zb_get_short_address();

	esp_zb_zdo_binding_table_req(&req, print_binding_table_cb, nullptr);
}

void ZigbeeDevice::print_binding_table_cb(const esp_zb_zdo_binding_table_info_t *record, void *user_ctx) {
	if (!record) {
		ESP_LOGI(TAG, "Binding table nullptr");
		return;
	}

	ESP_LOGI(TAG, "%03u/%03u [%02x,%02x]: %s/%u/%04x -> %s/%u",
		record->index, record->total, record->count, record->status,
		zigbee_address_string(record->record.src_address).c_str(),
		record->record.src_endp, record->record.cluster_id,
		(record->record.dst_addr_mode == ESP_ZB_ZDO_BIND_DST_ADDR_MODE_16_BIT_GROUP
			? zigbee_address_string(record->record.dst_address.addr_short).c_str()
			: zigbee_address_string(record->record.dst_address.addr_long).c_str()),
		record->record.dst_endp);

	if (record->index + 1 < record->total) {
		esp_zb_scheduler_alarm_cancel(print_binding_table_next, record->index + 1);
		esp_zb_scheduler_alarm(print_binding_table_next, record->index + 1, 100);
	}
}
```

This produces the following output, which contains lots of duplicates:
```
2023-09-12 20:09:48.303391 I (68195) nutt.ZigbeeDevice: 000/016 [01,00]: 40:4c:ca:ff:fe:42:30:78/11/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.314886 I (68195) nutt.ZigbeeDevice: 000/016 [02,00]: 40:4c:ca:ff:fe:42:30:78/12/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.324836 I (68205) nutt.ZigbeeDevice: 000/016 [03,00]: 40:4c:ca:ff:fe:42:30:78/13/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.428386 I (68325) nutt.ZigbeeDevice: 001/016 [04,00]: 40:4c:ca:ff:fe:42:30:78/12/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.439581 I (68325) nutt.ZigbeeDevice: 001/016 [05,00]: 40:4c:ca:ff:fe:42:30:78/13/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.450515 I (68335) nutt.ZigbeeDevice: 001/016 [06,00]: 40:4c:ca:ff:fe:42:30:78/14/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.548357 I (68445) nutt.ZigbeeDevice: 002/016 [07,00]: 40:4c:ca:ff:fe:42:30:78/13/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.559503 I (68445) nutt.ZigbeeDevice: 002/016 [08,00]: 40:4c:ca:ff:fe:42:30:78/14/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.570121 I (68455) nutt.ZigbeeDevice: 002/016 [09,00]: 40:4c:ca:ff:fe:42:30:78/21/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.678329 I (68575) nutt.ZigbeeDevice: 003/016 [0a,00]: 40:4c:ca:ff:fe:42:30:78/14/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.689523 I (68575) nutt.ZigbeeDevice: 003/016 [0b,00]: 40:4c:ca:ff:fe:42:30:78/21/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.700130 I (68585) nutt.ZigbeeDevice: 003/016 [0c,00]: 40:4c:ca:ff:fe:42:30:78/22/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.798370 I (68695) nutt.ZigbeeDevice: 004/016 [0d,00]: 40:4c:ca:ff:fe:42:30:78/21/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.809482 I (68695) nutt.ZigbeeDevice: 004/016 [0e,00]: 40:4c:ca:ff:fe:42:30:78/22/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.820169 I (68705) nutt.ZigbeeDevice: 004/016 [0f,00]: 40:4c:ca:ff:fe:42:30:78/23/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.918340 I (68815) nutt.ZigbeeDevice: 005/016 [10,00]: 40:4c:ca:ff:fe:42:30:78/22/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.929556 I (68815) nutt.ZigbeeDevice: 005/016 [11,00]: 40:4c:ca:ff:fe:42:30:78/23/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:48.940146 I (68825) nutt.ZigbeeDevice: 005/016 [12,00]: 40:4c:ca:ff:fe:42:30:78/24/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:49.048531 I (68945) nutt.ZigbeeDevice: 006/016 [13,00]: 40:4c:ca:ff:fe:42:30:78/23/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:49.059546 I (68945) nutt.ZigbeeDevice: 006/016 [14,00]: 40:4c:ca:ff:fe:42:30:78/24/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:49.070164 I (68955) nutt.ZigbeeDevice: 006/016 [15,00]: 40:4c:ca:ff:fe:42:30:78/31/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:49.168349 I (69065) nutt.ZigbeeDevice: 007/016 [16,00]: 40:4c:ca:ff:fe:42:30:78/24/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:49.179495 I (69065) nutt.ZigbeeDevice: 007/016 [17,00]: 40:4c:ca:ff:fe:42:30:78/31/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:49.190168 I (69075) nutt.ZigbeeDevice: 007/016 [18,00]: 40:4c:ca:ff:fe:42:30:78/32/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:49.288534 I (69185) nutt.ZigbeeDevice: 008/016 [19,00]: 40:4c:ca:ff:fe:42:30:78/31/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:49.299848 I (69185) nutt.ZigbeeDevice: 008/016 [1a,00]: 40:4c:ca:ff:fe:42:30:78/32/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:49.310457 I (69195) nutt.ZigbeeDevice: 008/016 [1b,00]: 40:4c:ca:ff:fe:42:30:78/33/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:49.408437 I (69305) nutt.ZigbeeDevice: 009/016 [1c,00]: 40:4c:ca:ff:fe:42:30:78/32/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
2023-09-12 20:09:49.419694 I (69305) nutt.ZigbeeDevice: 009/016 [1d,00]: 40:4c:ca:ff:fe:42:30:78/33/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
[... 20 lines cut ...]
[... cut ...]
40:4c:ca:ff:fe:42:30:78/33/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
40:4c:ca:ff:fe:42:30:78/34/0006 -> e0:79:8d:ff:fe:b3:dc:82/1

40:4c:ca:ff:fe:42:30:78/61/000f -> e0:79:8d:ff:fe:b3:dc:82/1
40:4c:ca:ff:fe:42:30:78/62/000f -> e0:79:8d:ff:fe:b3:dc:82/1
40:4c:ca:ff:fe:42:30:78/64/000f -> e0:79:8d:ff:fe:b3:dc:82/1

40:4c:ca:ff:fe:42:30:78/71/0006 -> e0:79:8d:ff:fe:b3:dc:82/1
```

This *exactly* matches the behaviour in Home Assistant.

I don't get reports for switch endpoints 72, 73 or 74 and binary input endpoint 63 is always delayed.

nomis, 2023-09-16:
> Could you provide an example of how to use `esp_zb_zdo_binding_table_req()` properly?

This API is unusable. Each `Mgmt_Bind_rsp` command response contains multiple entries in the `BindingTableList` but the callback is called for each of those without knowing how many of them there are. If you don't know how many there are you don't know when to make the next request and can't calculate the next `StartIndex`.

nomis, 2023-09-16:
@xieqinan I've updated the title of this issue because I think the issue is a 16 entry limit on the binding table.

From `include/zb_vendor_default.h`:
```
/* APS BIND TABLE SIZE */
#ifndef ZB_ED_ROLE
#define ZB_APS_SRC_BINDING_TABLE_SIZE 16
#define ZB_APS_DST_BINDING_TABLE_SIZE 32
#else
#define ZB_APS_SRC_BINDING_TABLE_SIZE 16
#define ZB_APS_DST_BINDING_TABLE_SIZE 32
#endif /* ZB_ED_ROLE */
```

These need to be increased to at least 64 and preferably something higher like 240. It is possible to have more than one cluster per endpoint and each cluster needs its own binding.

While you're at it the maximum endpoints could do with being at least 64 too. The underlying code really needs to dynamically allocate memory so that there are none of these arbitrary restrictions. You're selling a device with 512KB of RAM and then restricting what can be done with it by these arbitrary low limits that were designed for devices with 64KB of RAM.

xieqinan, 2023-09-18:
Hello @nomis 

> These need to be increased to at least 64 and preferably something higher like 240. It is possible to have more than one cluster per endpoint and each cluster needs its own binding.

We will carefully consider your proposal, and before adoption, I will assess the allocation of RAM space. Thank you for your test results.

xieqinan, 2023-09-18:
> Could you provide an example of how to use `esp_zb_zdo_binding_table_req()` properly?
> 
> I get 3 callbacks in response to most of my requests, so I have to use the scheduler to de-duplicate requests for the next entry to stop it getting completely out of control:

For this issue, it occurs because the maximum payload of IEEE802154 is 127, limiting a frame to include only 3 bind table records. We will consider filtering out duplicated bind table information internally. In the meantime, please use your own method to perform the filtering.

nomis, 2023-09-18:
> We will carefully consider your proposal, and before adoption, I will assess the allocation of RAM space. Thank you for your test results.

You already have separate builds of ZBOSS for "co-ordinator/router" and "end-device" so in the absence of dynamic allocation you could include additional variants for "higher limits" (64 endpoints, 64 binding table entries) and "maximum limits" (240 endpoints, 255 binding table entries).

> In the meantime, please use your own method to perform the filtering.

The problem occurs because your callback API doesn't indicate which call represents the last binding table entry in the current response. There's no filtering required, you just need to provide all of the response in one call as an array or provide an index/count for it.

The requests are identical to those that can be done over the network, so it's possible to just query it remotely and get the same information for debugging.

nomis, 2023-10-15:
This bug is still present in 1.0.0

nomis, 2023-10-19:
I've not checked but there are probably other limits that are too low for the number of entities I have, like the maximum number of group memberships. These should be scaled appropriately too.

xieqinan, 2023-10-24:
Hello @nomis,

The entities in the bind table are the main reason for the issue you're encountering. Dynamic allocation might be necessary for the bind table. However, implementing this isn't straightforward. We will have a discussion and work on providing a more flexible way for users to utilize the `table` feature in the esp-zigbee-sdk.

> The problem occurs because your callback API doesn't indicate which call represents the last binding table entry in the current response. There's no filtering required, you just need to provide all of the response in one call as an array or provide an index/count for it.

Thank you for your suggestion, the binding table response issue will be resolved as soon as possiable.

closed 2023-11-24 by espressif-bot

nomis, 2023-11-25:
This works ok now and the existing device appears to have survived the increase to 64 entries without introducing extra bindings.