espressif/esp-zigbee-sdk #67: Implement the identify cluster fully with examples

nomis, 2023-08-06:
There are functions to create identify clusters but not to use them. This leads to devices that claim to implement that feature but do nothing when it's used.

If I do nothing then the identify command sets the `IdentifyTime` and I'll get the following error once a second while the `IdentifyTime` attribute counts down:
```
W (58523) ESP_ZIGBEE_CORE: cluster ID:3, attribute ID:0 is currently not supported for updating!
W (58523) ESP_ZIGBEE_CORE: cluster ID:3, attribute ID:0 is currently not supported for updating!
W (59543) ESP_ZIGBEE_CORE: cluster ID:3, attribute ID:0 is currently not supported for updating!
W (60553) ESP_ZIGBEE_CORE: cluster ID:3, attribute ID:0 is currently not supported for updating!
W (61563) ESP_ZIGBEE_CORE: cluster ID:3, attribute ID:0 is currently not supported for updating!
W (62583) ESP_ZIGBEE_CORE: cluster ID:3, attribute ID:0 is currently not supported for updating!
W (63593) ESP_ZIGBEE_CORE: cluster ID:3, attribute ID:0 is currently not supported for updating!
W (64613) ESP_ZIGBEE_CORE: cluster ID:3, attribute ID:0 is currently not supported for updating!
```
(These log messages are missing the endpoint ID.)

If I set the callback using the function referenced in https://github.com/espressif/esp-zigbee-sdk/issues/22#issuecomment-1573063055 then it gets called when the identify command is used but I receive garbage values and it no longer sets the `IdentifyTime`:
```
I (118573) identify_cb: status=0 data_type=789184517 value=219
I (121083) identify_cb: status=0 data_type=2785214469 value=16
I (122943) identify_cb: status=0 data_type=3589799941 value=102
```

Ideally I should be able to get a callback to `esp_zb_device_add_set_attr_value_cb()` every time the `IdentifyTime` attribute changes, and the examples should make use of that by logging that the "identify" action should occur for the next `value` seconds.

xieqinan, 2023-08-07:
Hi @nomis ,
The issue will be fixed to support almost all attributes.

closed 2023-08-11 by espressif-bot

xieqinan, 2023-08-11:
Hello,
The issue has been resolved in the  [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk)  master branch, which includes some breaking changes. Would you be able to give it another try?

nomis, 2023-08-12:
I'm getting `ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID` attribute changes now.