espressif/esp-zigbee-sdk #68: Not able to write into the occupancy sensing cluster  "W (2982064) ESP_ZIGBEE_CORE: cluster ID:1030, attribute ID:16 is currently not supported for updating!"

Bosemani, 2023-08-07:
Hai,
 I crated the Occupancy sensing server cluster and tested reporting occupancy values. Its Working fine.
But In occupancy cluster (0x0406), attribute ID **0x0010(ESP_ZB_ZCL_ATTR_OCCUPANCY_SENSING_PIR_OCC_TO_UNOCC_DELAY_ID), 0x0011(ESP_ZB_ZCL_ATTR_OCCUPANCY_SENSING_PIR_UNOCC_TO_OCC_DELAY_ID), 0x0012 (ESP_ZB_ZCL_ATTR_OCCUPANCY_SENSING_PIR_UNOCC_TO_OCC_THRESHOLD_ID)** I'm not able to write.
I'm getting this Warning message 
`
W (2982064) ESP_ZIGBEE_CORE: cluster ID:1030, attribute ID:16 is currently not supported for updating!

W (3161994) ESP_ZIGBEE_CORE: cluster ID:1030, attribute ID:17 is currently not supported for updating!

W (3224044) ESP_ZIGBEE_CORE: cluster ID:1030, attribute ID:18 is currently not supported for updating!`
I'm not getting any other messages. How to solve this?

Thanks in Advance!
        Bose

xieqinan, 2023-08-10:
Hi,
This issue has been internally resolved and will be released either this week or the next.

xieqinan, 2023-08-11:
Hello,
 the issue has been resolved in the  [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk)  master branch, which includes some breaking changes. Would you be able to give it another try?

Bosemani, 2023-08-11:
Hai @xieqinan 
 I tried the same cluster to write.
Now, I'm getting the callback for the written values. But I'm not getting correct cluster id in the callback.
Callback function output :

`I (48242) ZIGBEE_HAL: Data received on Cluster id 3 attr_id 16 EP 1

I (71652) ZIGBEE_HAL: Data received on Cluster id 4 attr_id 17 EP 1

I (78482) ZIGBEE_HAL: Data received on Cluster id 5 attr_id 18 EP 1` 

all three are written to Cluster ID 0x0406 but I got in call back 3,4,5.

xieqinan, 2023-08-14:
Could you attach the code of attr_cb()? The cluster id 3, 4 and 5 are identity, groups and scenes.

Bosemani, 2023-08-14:
Hai @xieqinan 
Here I'm registering callback `esp_zb_device_add_set_attr_value_cb(attr_cb);`

and Function

void attr_cb(uint8_t status, uint8_t endpoint, uint16_t cluster_id, uint16_t attr_id, void *new_value)
{
  ESP_LOGI(TAG, "Data received on Cluster id %d attr_id %d EP %d",cluster_id, attr_id, endpoint);
}

I'm not doing anything in callback. Just I'm printing.

xieqinan, 2023-08-14:
Hi,
In the current version, the parameters of `esp_zb_device_add_set_attr_value_cb` have been modified to provide additional options for users. The API can be employed as shown below:
```
static esp_err_t attr_cb(const esp_zb_zcl_set_attr_value_message_t message)
{
    esp_err_t ret = ESP_OK;
    bool light_state = 0;
    if (message.info.status == ESP_ZB_ZCL_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "Received message: endpoint(0x%x), cluster(0x%x), attribute(0x%x), data size(%d)", message.info.dst_endpoint,
                 message.info.cluster, message.attribute, message.data.size);
        if (message.info.dst_endpoint == HA_ESP_LIGHT_ENDPOINT) {
            if (message.info.cluster == ESP_ZB_ZCL_CLUSTER_ID_ON_OFF) {
                if (message.attribute == ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID && message.data.type == ESP_ZB_ZCL_ATTR_TYPE_BOOL) {
                    light_state = message.data.value ? *(bool *)message.data.value : light_state;
                    ESP_LOGI(TAG, "Light sets to %s", light_state ? "On" : "Off");
                    light_driver_set_power(light_state);
                }
            }
        }
    } else {
        ESP_LOGE(TAG, "Received message: status(%d) error", message.info.status);
        ret = ESP_ERR_INVALID_ARG;
    }
    return ret;
}
```

Bosemani, 2023-08-14:
Hai @xieqinan 
Thank you for Information.

It was working perfectly.