espressif/esp-zigbee-sdk #69: custom cluster with string attribute

luigidoria, 2023-08-08:
I'm attempting to send a string using `esp_zb_zcl_write_attr_cmd_req`, but when I declare the cluster type as `ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING`, the following error occurs: `ESP_ZIGBEE_CORE: The data type of the value is currently not supported.`

Is it currently possible to send a string? Alternatively, is there a way to identify which device is sending the message?

xieqinan, 2023-08-10:
The issue will be fixed in this week.

closed 2023-08-11 by espressif-bot

xieqinan, 2023-08-11:
Hello,
 The issue has been resolved in the  [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk)  master branch, which includes some breaking changes. Would you be able to give it another try?

luigidoria, 2023-08-22:
Now it worked, but the first character looks like it can't be a lowercase letter.  "ABC" is send correctly but "abc" didn't work. 

Coordinator code:

```
static esp_err_t attr_cb(const esp_zb_zcl_set_attr_value_message_t message)
{
    esp_err_t ret = ESP_OK;
    if (message.info.status == ESP_ZB_ZCL_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "Received message: endpoint(0x%x), cluster(0x%x), attribute(0x%x), data size(%d)", message.info.dst_endpoint,
                 message.info.cluster, message.attribute, message.data.size);
        if (message.info.dst_endpoint == CD_ENDPOINT) {
            if (message.info.cluster == 0xFFFE) {
                if (message.attribute == 0x00 && message.data.type == ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING) {
                    const char *ptr = message.data.value;
                    printf("Recived uppercase message: ");
                    while (*ptr != '\0') {
                        printf("%c", *ptr);
                        ptr++;   
                    }
                    printf("\n");
                }
                if (message.attribute == 0x01 && message.data.type == ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING) {
                    const char *ptr = message.data.value;
                    printf("Recived lowercase message: ");
                    while (*ptr != '\0') {
                        printf("%c", *ptr);
                        ptr++;   
                    }
                    printf("\n");
                }
            }
        }
    } else {
        ESP_LOGE(TAG, "Received message: status(%d) error", message.info.status);
        ret = ESP_ERR_INVALID_ARG;
    }
    return ret;
}
```

End-Device code:

```
void* string_generator(void* arg)
{
    while (1) {
        ESP_LOGI(TAG, "==========================");
        
        char turn_string_uppercase[50] = "ABC";
        char turn_string_lowercase[50] = "abc";

        esp_zb_zcl_write_attr_cmd_t cmd_req;
        cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = network_params.short_addr;
        cmd_req.zcl_basic_cmd.dst_endpoint = network_params.endpoint;
        cmd_req.zcl_basic_cmd.src_endpoint = ED_ENDPOINT;
        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        cmd_req.clusterID = 0xFFFE;
        cmd_req.attributeID = 0x00;
        cmd_req.attrType = ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING;
        cmd_req.attrVal = (uint8_t *)turn_string_uppercase;

        ESP_EARLY_LOGI(TAG, "Send the write command (uppercase).");
        esp_zb_zcl_write_attr_cmd_req(&cmd_req);

        esp_zb_zcl_write_attr_cmd_t cmd_req2;
        cmd_req2.zcl_basic_cmd.dst_addr_u.addr_short = network_params.short_addr;
        cmd_req2.zcl_basic_cmd.dst_endpoint = network_params.endpoint;
        cmd_req2.zcl_basic_cmd.src_endpoint = HYDROMETER_ED_ENDPOINT;
        cmd_req2.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
[... 57 lines cut ...]
I (1113) END DEVICE: endpoint ID list: 10
I (1173) END DEVICE: Simple_desc_cb device_id:2,app_version:0,profile_id:0x104,endpoint ID:10
I (1173) END DEVICE: cluster ID list:0xfffe
I (1173) END DEVICE: cluster ID list:0xfffe
I (1183) END DEVICE: ==========================
I (1183) END DEVICE: Send the write command (uppercase).
I (1193) END DEVICE: Send the write command (lowercase).
I (1203) END DEVICE: ==========================
I (6203) END DEVICE: ==========================
I (6203) END DEVICE: Send the write command (uppercase).
I (6203) END DEVICE: Send the write command (lowercase).
I (6203) END DEVICE: ==========================
I (11213) END DEVICE: ==========================
I (11213) END DEVICE: Send the write command (uppercase).
I (11213) END DEVICE: Send the write command (lowercase).
I (11213) END DEVICE: ==========================
I (16223) END DEVICE: ==========================
I (16223) END DEVICE: Send the write command (uppercase).
I (16223) END DEVICE: Send the write command (lowercase).
I (16223) END DEVICE: ==========================
I (21233) END DEVICE: ==========================
I (21233) END DEVICE: Send the write command (uppercase).
I (21233) END DEVICE: Send the write command (lowercase).
I (21233) END DEVICE: ==========================
```

xieqinan, 2023-08-22:
Hi, 
If the attribute type is string, the first byte of value indicates the string length.
Suggestion:
```
char turn_string_uppercase[50] = " ABC";
char turn_string_lowercase[50] = " abc";
turn_string_uppercase[0] = strlen(turn_string_uppercase);
turn_string_lowercase[0] = strlen(turn_string_lowercase);
```
Note the distinction between " ABC" and "ABC."

luigidoria, 2023-08-22:
thank you for the help