espressif/esp-zigbee-sdk #70: Usability Improvement Suggestions

Zmmfly, 2023-08-10:
1. `esp_zb_add_read_attr_resp_cb` Its description is: "Set the ZCL read attribute response callback for specific endpoint." The endpoint here needs to be clear whether it is for the local (client) side or the remote (server) side. It is also recommended to add context pointers as parameter.
2. `esp_zb_read_attr_resp_callback_t` It is recommended to add nwkAddr or IEEE address, and endpoint number as parameters, so as to distinguish the reply from which device when multiple devices response.
3. `esp_zb_report_attr_callback_t ` It is recommended to add the context pointer as parameter.
4. I think that source nwkAddr, source endpoint, and context pointer are necessary in all callback functions except special cases.

xieqinan, 2023-08-14:
Thank you, your suggestion is very useful and practical. I will update it as soon as possiable.

closed 2023-08-30 by espressif-bot

xieqinan, 2023-08-30:
Hello,
The feature request has been implemented, please refer to the newest [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L227C18-L227C43).