espressif/esp-zigbee-sdk #72: Support transparent decompression when using OTA

nomis, 2023-08-14:
Using `gzip -9` on both my app and the example `ota_file.bin` makes it 53% of the original size.

Automatically decompressing the OTA content on the client while it's streamed from the OTA server would speed up the OTA process. The OTA download is controlled by the client so it will always receive all of the data in order.

As the OTA process is hidden within the library, a callback that allows other compression algorithms to be used would allow for more flexibility. An uncompressed and zlib implementation of that would look something like this:
```
enum ota_state { OTA_DATA_START, OTA_DATA_BLOCK, OTA_DATA_FINISH, OTA_DATA_ABORT };

/**
 * @param arg User callback argument
 * @param state Data stream state
 * @param buf Input OTA data
 * @param count Input OTA data size
 * @param output Output OTA data function
 */

esp_err_t ota_convert_uncompressed_cb(void *arg, enum ota_state state, const char *buf, size_t count,
       esp_err_t (*output)(const char *buf, size_t count)) {
    return output(buf, count);
}

esp_err_t ota_convert_zlib_cb(void *arg, enum ota_state state, const char *buf, size_t count,
        esp_err_t (*output)(const char *buf, size_t count)) {
    /* TODO: error handling of output() omitted */
    if (state == OTA_DATA_START) {
        /* zlib init and decompress first block */
        output(decompress_buf, decompress_count);
    } else if (state == OTA_DATA_BLOCK) {
        /* zlib decompress */
        output(decompress_buf, decompress_count);
    } else if (state == OTA_DATA_FINISH) {
        /* zlib decompress last block */
        output(decompress_buf, decompress_count);
        /* zlib flush */
        output(decompress_buf, decompress_count);
        /* zlib free */
    } else if (state == OTA_DATA_ABORT) {
        /* zlib free */
    }
    return ESP_OK;
}
```

The output is a function call because it could vary in size and all of the input needs to be consumed. Note that the output function could be called multiple times for the same input if the data is highly compressed. There must always be a finish/abort call to free any allocated buffers.

chshu, 2023-10-11:
@nomis We will expose the OTA callbacks to the application layer, which should be able to meet your requirement.

closed 2023-10-23 by espressif-bot

xieqinan, 2023-10-23:
@nomis 

> @nomis We will expose the OTA callbacks to the application layer, which should be able to meet your requirement.

This feature has already been supported in the latest esp-zigbee-sdk v1.0.1

nomis, 2023-11-25:
(It looks like it's now mandatory to do the OTA in the application because I see nothing in the example to enable this?)

There's nothing implemented in the example for `ESP_ZB_ZCL_OTA_UPGRADE_STATUS_ABORT` or `ESP_ZB_ZCL_OTA_UPGRADE_STATUS_ERROR` so `esp_ota_begin()` could be called multiple times if an OTA is restarted.

There's nothing implemented for `ESP_ZB_ZCL_OTA_UPGRADE_STATUS_OK` or `ESP_ZB_ZCL_OTA_UPGRADE_STATUS_BUSY` either.