espressif/esp-zigbee-sdk #73: Network slowness when disconnecting the coordinator

diazmanuel, 2023-08-15:
hello,
I am currently migrating a device from end device to router.
The system would have 1 coordinator, this router and 7 end devices.

I am testing the stability of the system, and for this I disconnect the coordinator looking to verify that the end devices maintain the connection, since when there is a router they should connect to it without losing the established network.
up to this point everything works as expected, the devices do not lose connection but the problem is that the response time increases considerably, what used to take a second or less now takes up to a couple of minutes.

any idea why this happens?
I am not sure if this is due to disconnecting the coordinator or the devices connecting to the router (I mention this because when the system has the coordinator connected, I am not sure if the end devices connect directly to it or pass through the router)

xieqinan, 2023-08-16:
Hello,

> what used to take a second or less now takes up to a couple of minutes.

What have you done that has led to the different phenomenon between the current and the past?

> I am not sure if this is due to disconnecting the coordinator or the devices connecting to the router

In the scenario where the network formation is completed and the coordinator powers off, if an end device establishes a connection with the coordinator, it will persistently transmit data requests to the coordinator that is no longer connected. This continues until the end device acknowledges the loss of the coordinator and subsequently attempts to rejoin the network. This entire process takes approximately 1 minute. 

To confirm your assumption, you can adjust the `MAX_CHILDREN` value to establish an upper limit on the number of connected devices for the coordinator or router.

diazmanuel, 2023-08-23:
hello @xieqinan ,

> What have you done that has led to the different phenomenon between the current and the past?

this phenomenon happens when i disconnect the cordinator and wait for the network to reorganize

> In the scenario where the network formation is completed and the coordinator powers off, if an end device establishes a connection with the coordinator, it will persistently transmit data requests [...]

yes thats right but after reorganize of the network (that it takes a couple of minutes), the zigbee network works with a very long delay, as I mentioned before all the actions take significantly longer than when the coordinator was connected. After analyzing it, I deduce the cause may be one of the following

**1.** The zigbee protocol works in this way and whenever the coordinator is disconnected, these delays cannot be avoided.

**2.** The router is the one that causes this delay, when disconnecting the coordinator the endpoint devices are forced to connect through the router. This would mean that there is some problem with the SDK that would be generating a delay with the router. Have you tried a device configured as a router by connecting several endDevices to it? Did you notice any delay like the one I mentioned?

**3.** The endDevice is responsible for the delay. I have configured the end device to report to both the coordinator and another end device. When a report is made internally, is an ACK expected to continue? If so, could the delay occur since the report fails to reach the coordinator and until a timeout is activated, the device does not report to the next endDevice?

What do you think of these 3 hypotheses, do you have any other suggestions as to why these delays may be occurring?

chshu, 2023-10-12:
@diazmanuel I tested the same flow, but didn't see the significant delay.

It's as expected when coordinator is just powered off, the EDs take some time to disconnect and rejoin to the router, but once the network becomes stable again, there is no significant delay.

Could you try with the latest SDK, if the issue still present, please share us the detailed reproduce steps and sniffer capture file.

chshu, 2023-11-19:
@diazmanuel Please reopen if you have any follow up on this topic.

closed 2023-11-19 by chshu