espressif/esp-zigbee-sdk #75: Example for esp_zb_add_cluster_privilege_command_cb

danshuk, 2023-08-17:
Are there any examples for handling callbacks via esp_zb_add_cluster_privilege_command_cb, specifically how to respond to a command request?

I'm trying to implement code to handle e.g. ESP_ZB_ZCL_CMD_LEVEL_CONTROL_STOP and can do that via the callback, however as I can't find an obvious way to respond to the coordinator, I get timeout errors.

For esp_zb_add_custom_cluster_command_cb callback we have esp_zb_zcl_custom_cluster_cmd_resp, but what is the equivalent for privilege command, or is there another way to handle these messages without skipping the stack?

xieqinan, 2023-08-22:
Hi @danshuk ,

The privilege command provides an easy way to define a customized command for the standard cluster.

Firstly, the privilege command serves as a customized attribute command, not regulated by the Zigbee specification. For instance, in the On/off cluster, command IDs such as `Off(0x00)`, `On(0x01)`, `Toggle(0x02)`, `Off with effect(0x40)`, `On with recall global scene(0x41)`, and `On with timed off(0x42)` exist. By using the `esp_zb_add_cluster_privilege_command_cb()` function, you can register the `Off(0x00)` command. As a result, the command becomes accessible through the customized callback, and the stack will skip its handling.

Secondly, you can also register the `xxx(0x80)` command for the On/off cluster, even though it isn't defined in the Zigbee specification. Similar to before, this command can be retrieved through a customized callback that's registered using `esp_zb_add_cluster_privilege_command_cb()`.

Thirdly, if a standard command like `ESP_ZB_ZCL_CMD_LEVEL_CONTROL_STOP` from the Level cluster hasn't been registered as a privilege command, the stack will manage its processing in the usual manner.

xieqinan, 2023-09-21:
Hi @danshuk ,

The privilege command API is designed for handling non-standard commands within standard ZCL clusters. It only provides a basic interface for ZCL cluster. Users are responsible for designing both the command request and its corresponding response. The SDK will use the return value provided by users to send a default response. However, if users wish to send a specific response, they must design it themselves. This distinguishes it from custom cluster commands.

Furthermore, it's important to note that this is a non-standard Zigbee API. Could you please clarify the specific situations in which you need to use it?

closed 2023-11-22 by chshu