espressif/esp-zigbee-sdk #76: Custom Cluster not working as expected

TJvV, 2023-08-18:
Hi,

I'm trying out the ESP32 ZigBee stack after some experience with the NXP stack (which will be my reference).
Using the latest version (0.9.0).

I'm having some issues getting my custom cluster to work.
I started out from the `HA_on_off_light` and `HA_on_off_switch` examples.
The OnOff endpoint/cluster communication works pretty well.

When I try to send a custom command, or write the value from my button event, the light toggles as expected.
However I don't see anything happening regarding my custom cluster.

I've added my custom cluster + endpoint to the `HA_on_off_switch` as a `ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE`, and to the `HA_on_off_light` as `ESP_ZB_ZCL_CLUSTER_SERVER_ROLE`.

I have omitted a bunch of error checking and logging code for brevity:

```C
#define LSL_ENDPOINT 1
#define LSL_PROFILE_ID (0xFFFFU) // ZB_AF_WILDCARD_PROFILE_ID
#define LSL_DEVICE_ID (0xFFF2)   // ZB_HA_CUSTOM_ATTR_DEVICE_ID
#define LSL_CLUSTER_ID 0xFC00

static uint32_t lsl_cluster_attr_0 = 0;

void lsl_cluster_register_callbacks()
{
    esp_zb_add_custom_cluster_command_cb(LSL_ENDPOINT, LSL_CLUSTER_ID, lsl_cluster_cmd_callback);
    esp_zb_device_add_set_attr_value_cb(lsl_cluster_set_attribute_callback);
    esp_zb_device_add_report_attr_cb(lsl_cluster_report_attribute_callback);
}

static void esp_zb_task(void *pvParameters)
{
    [...]
    esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);

    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_attribute_list_t *attributes_list = esp_zb_zcl_attr_list_create(LSL_CLUSTER_ID);

    const uint16_t attr_id = 0;
    const uint8_t attr_type = ESP_ZB_ZCL_ATTR_TYPE_U32;
    const uint8_t attr_access = ESP_ZB_ZCL_ATTR_MANUF_SPEC | ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE;
    void *attr_value = &lsl_attr_0;

    esp_zb_custom_cluster_add_custom_attr(attributes_list, attr_id, attr_type, attr_access, attr_value);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, attributes_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_ep_list_add_ep(esp_zb_on_off_light_ep, esp_zb_cluster_list, LSL_ENDPOINT, LSL_PROFILE_ID, LSL_DEVICE_ID);

    esp_zb_device_register(esp_zb_on_off_light_ep);
    lsl_cluster_register_callbacks();
    [...]
}
```

The callbacks are empty, except for a call to the log function.

```C
static void esp_zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL)
    {
        /* implemented light switch toggle functionality */
        esp_zb_zcl_on_off_cmd_t cmd_req;
        cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = on_off_light.short_addr;
        cmd_req.zcl_basic_cmd.dst_endpoint = on_off_light.endpoint;
        cmd_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_TOGGLE_ID;
        ESP_EARLY_LOGI(TAG, "send 'on_off toggle' command");
        esp_zb_zcl_on_off_cmd_req(&cmd_req);

        static uint32_t val = 0xA5;
        esp_zb_zcl_custom_cluster_cmd_req_t lsl_cmd_req;
        lsl_cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = on_off_light.short_addr;
        lsl_cmd_req.zcl_basic_cmd.dst_endpoint = LSL_ENDPOINT;
        lsl_cmd_req.zcl_basic_cmd.src_endpoint = LSL_ENDPOINT;
        lsl_cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        lsl_cmd_req.value = &val;
        lsl_cmd_req.data_type = ESP_ZB_ZCL_ATTR_TYPE_U32;
        lsl_cmd_req.custom_cmd_id = E_CLD_LSL_CMD_UPDATE_FROM_LSL;
        esp_zb_zcl_custom_cluster_cmd_req(&lsl_cmd_req);

        esp_zb_zcl_write_attr_cmd_t write_req;
        write_req.zcl_basic_cmd.dst_addr_u.addr_short = on_off_light.short_addr;
        write_req.zcl_basic_cmd.dst_endpoint = LSL_ENDPOINT;
        write_req.zcl_basic_cmd.src_endpoint = LSL_ENDPOINT;
        write_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        write_req.clusterID = LSL_CLUSTER_ID;
        write_req.attributeID = 0;
        write_req.attrType = ESP_ZB_ZCL_ATTR_TYPE_U32;
        write_req.attrVal = (uint8_t *)&val;
        esp_zb_zcl_write_attr_cmd_req(&write_req);
    }
}
```

I can find my custom attribute on the `HA_on_off_light` using
```C
esp_zb_zcl_attr_t *attr = esp_zb_zcl_get_attribute(LSL_ENDPOINT, LSL_CLUSTER_ID, LSL_CLUSTER_ROLE, 0);

uint32_t new_val = 0x5A;
esp_zb_zcl_set_attribute_val(LSL_ENDPOINT, LSL_CLUSTER_ID, LSL_CLUSTER_ROLE, 0, &new_val, true);
```

When I run some general discovery commands (Active Endpoints, Simple Descriptor, Discover Attributes, Discover Commands), I can find the endpoint and cluster (0x01, 0xFC00); but no commands or attributes; I don't even receive a  Discover Attributes Response.

Targeting the OnOff endpoint/cluster (0x0A, 0x0006), does give me valid responses.

What am I missing here?

It would be really nice if there would be some logging or an example of custom clusters as I'm pretty much flying blind right now.
I have added `esp_log_level_set("*", ESP_LOG_VERBOSE);` to the programs, but not seeing anything about the traffic or discovery.

xieqinan, 2023-08-21:
Hello,
Please change the `LSL_PROFILE_ID` from `0xffff` to `0x0104`, because the current version only support `Home Automation Profile ID`.

TJvV, 2023-08-25:
Hi,
Thank you for pointing this out.

With the Home Automation profile ID, I am able to read and write my attributes.

I am however still struggling getting the Reporting to work.
I couldn't find a good way to configure it locally using the `esp_zb_zcl_config_report_cmd_t`, so I tried it with the ZBOSS API instead.

After adding `ESP_ZB_ZCL_ATTR_ACCESS_REPORTING` to the attribute access specifier, I tried to set it up using 

```C
for( uint16_t id = 0; id < NUM_ATTRS; id++ )
{
  zb_zcl_reporting_info_t info;
  memset(&info, 0, sizeof(info));
  
  // Attribute
  info.direction = ZB_ZCL_CONFIGURE_REPORTING_SEND_REPORT;
  info.ep = LSL_ENDPOINT;
  info.cluster_id = LSL_CLUSTER_ID;
  info.cluster_role = LSL_CLUSTER_ROLE;
  info.manuf_code = LSL_MANUF_CODE;
  info.attr_id = id;
  
  // Configuration: send every change
  info.u.send_info.min_interval = 0;       // no limit
  info.u.send_info.max_interval = 0;       // no limit
  info.u.send_info.def_min_interval = 0;   // no limit
  info.u.send_info.def_max_interval = 0;   // no limit
  info.u.send_info.delta.u32 = 0;          // Report every change
  info.u.send_info.reported_value.u32 = 0; // Initial value is zero
  
  // Target: Co-ordinator
  info.dst.endpoint = LSL_ENDPOINT;
  info.dst.profile_id = LSL_PROFILE_ID;
  info.dst.short_addr = 0;
  
  zb_bool_t override = true;
  zb_ret_t ret = zb_zcl_put_reporting_info(&info, override);
  [...]
  ret = zb_zcl_start_attr_reporting_manuf(LSL_ENDPOINT,
                                                    LSL_CLUSTER_ID,
                                                    LSL_CLUSTER_ROLE,
                                                    id,
                                                    LSL_MANUF_CODE);
  [...]
}
```
This is returning `RET_OK` for both calls; yet when I change the value of my attributes with `esp_zb_zcl_set_attribute_val`, I don't see any reports on my Coordinator.
Reading out the attributes manually does show the values change.

xieqinan, 2023-08-30:
Hello,

> I couldn't find a good way to configure it locally using the `esp_zb_zcl_config_report_cmd_t`, so I tried it with the ZBOSS API instead.

Could you try to use the following code to configure the report locally?, it will send report configuration to itself.
```
      esp_zb_zcl_config_report_cmd_t report_cmd;
      report_cmd.zcl_basic_cmd.dst_addr_u.addr_short = 0;
      report_cmd.zcl_basic_cmd.dst_endpoint = LSL_ENDPOINT;
      report_cmd.zcl_basic_cmd.src_endpoint = LSL_ENDPOINT;
      report_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
      report_cmd.clusterID = LSL_CLUSTER_ID;
      report_cmd.attributeID = id;
      report_cmd.attrType = ESP_ZB_ZCL_ATTR_TYPE_BOOL;
      report_cmd.min_interval = 0;
      report_cmd.max_interval = 0;
      report_cmd.reportable_change = 0;
      esp_zb_zcl_config_report_cmd_req(&report_cmd);
      
```

> This is returning `RET_OK` for both calls; yet when I change the value of my attributes with `esp_zb_zcl_set_attribute_val`, I don't see any reports on my Coordinator.
> Reading out the attributes manually does show the values change.

For this issue, I will try to re-product it and give you feedback as soon as possiable.

xieqinan, 2023-08-30:
Here are some suggestions that might prove helpful for your situation:

First, it's essential to ensure that the attribute you're working with is writable, as set by the function `esp_zb_zcl_set_attribute_val`.

Second, to make use of the report attribute feature, you need to associate it with the bind operation. You can refer to [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L72) for guidance, as the report attribute command is dispatched through a search in the binding table.

Third, the report attribute command frame needs to be sent to the cluster client. Please make sure that a custom cluster client exists on your coordinator side.

Fourth, for handling the report attribute callback, you can find a reference implementation at [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L258).

TJvV, 2023-09-05:
> Could you try to use the following code to configure the report locally?, it will send report configuration to itself.

Hi,
I've tried this, and it seems the device gets stuck.
It only logs that it's gonna try to configure the first attribute, and then nothing.

The binding is done with
```C
static void bind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS)
    {
        lsl_cluster_config_reporting();
    }
}

static void bind_to_coordinator()
{
    esp_zb_zdo_bind_req_param_t bind_req;

    esp_zb_get_long_address(bind_req.src_address);
    bind_req.src_endp = LSL_ENDPOINT;
    bind_req.cluster_id = LSL_CLUSTER_ID;
    bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;

    esp_zb_ieee_address_by_short(0, bind_req.dst_address_u.addr_long);
    bind_req.dst_endp = LSL_ENDPOINT;
    bind_req.req_dst_addr = esp_zb_get_short_address();

    esp_zb_zdo_device_bind_req(&bind_req, bind_cb, NULL);
}
```
Which was inspired by the `HA_on_off_switch` example.
`bind_to_coordinator` is called after a successful `ESP_ZB_BDB_SIGNAL_STEERING` signal.

When I try it with the `zb_zcl_put_reporting_info_from_req` call, I receive the initial value on my coordinator, but further changes don't trigger any reports.

xieqinan, 2023-09-22:
Hi @TJvV ,

> It seems to work now with the following configuration.

The configuration differs from the one above only in `info.u.send_info.delta.u32`, which appears to have a slight impact on attribute reporting. Have you made any other updates to this configuration?

Furthermore, if the issue has been resolved, please consider closing it.

TJvV, 2023-09-26:
Hi,

Reviewing everything, there was one other change: upgrading from version 0.9.0 to 0.9.3

I will close this issue for now and open a new one if something unexpected pops up :)

closed 2023-09-26 by TJvV