espressif/esp-zigbee-sdk #77: custom node descriptor

xmow49, 2023-08-20:
Hi,
I currently creating a ZigBee device on battery. I set up the power cluster like this:
```c
    esp_zb_attribute_list_t *esp_zb_power_cluster = esp_zb_power_config_cluster_create(&power_cfg);
    esp_zb_power_config_cluster_add_attr(esp_zb_power_cluster, ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_VOLTAGE_ID, &batteryVoltage);
    esp_zb_power_config_cluster_add_attr(esp_zb_power_cluster, ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_SIZE_ID, &batterySize);
    esp_zb_power_config_cluster_add_attr(esp_zb_power_cluster, ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_QUANTITY_ID, &batteryQuantity);
    esp_zb_power_config_cluster_add_attr(esp_zb_power_cluster, ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_RATED_VOLTAGE_ID, &batteryRatedVoltage);
    esp_zb_power_config_cluster_add_attr(esp_zb_power_cluster, ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_ALARM_MASK_ID, &batteryAlarmMask);
    esp_zb_power_config_cluster_add_attr(esp_zb_power_cluster, ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_VOLTAGE_MIN_THRESHOLD_ID, &batteryMinVoltage);
    esp_zb_power_config_cluster_add_attr(esp_zb_power_cluster, ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_A_HR_RATING_ID, &batteryAhrRating);
    esp_zb_power_config_cluster_add_attr(esp_zb_power_cluster, ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_PERCENTAGE_REMAINING_ID, &batteryPercentageRemaining);
``` 
And no problem, but in the node descriptor, the power source bit is set to 1:
[image]
And when I add my device to Home Assistant, I don't have a battery entity because of this flag.
More info on 2.3.2.3.6 MAC Capability Flags Field

Today, is there a way to customize the node descriptor?
Thanks

xieqinan, 2023-09-07:
Hello,

I think you can call `esp_zb_set_rx_on_when_idle(false)` to set power source to 0 for your device before the `esp_zb_start()`. We will provide the API for MAC Capability Flag later.

xmow49, 2023-09-20:
Its work.
Thanks

closed 2023-09-20 by xmow49

valdisz, 2023-12-12:
> esp_zb_set_rx_on_when_idle(false)

Is there already a better version of the SDK that supports MAC capability customization?

fernandorpardo, 2024-10-03:
Hi,
Thanks for the discussion and the information above.
A side effect of calling `esp_zb_set_rx_on_when_idle(false)` is that the signal  `ESP_ZB_COMMON_SIGNAL_CAN_SLEEP` is triggered continuously.
I notice that the performance of the code I am working is reduced so I have isolated the problem by adding the esp_zb_set_rx_on_when_idle(false) call to the HA_on_off_light example. The response to the light on/off command is delayed up to about a couple of seconds.
I tried adding 
```
    case ESP_ZB_COMMON_SIGNAL_CAN_SLEEP :
        esp_zb_sleep_now();
        break;
```
to `esp_zb_app_signal_handler()` but the issue is not improving.
How shall I handle the ESP_ZB_COMMON_SIGNAL_CAN_SLEEP signal to avoid the performance impact?

mdudek, 2024-10-28:
Hi, something new about missing API for MAC Capability Flag?

P-R-O-C-H-Y, 2024-10-30:
Hi @xieqinan, any updates on this? I am facing the same issue. Thanks

xieqinan, 2024-10-31:
Hi all,

There is a temporary API available for setting the MAC Capability Flag. You can use `extern void zb_set_ed_node_descriptor(bool power_src, bool rx_on_when_idle, bool alloc_addr);`. Then, call it before `esp_zb_start();`.

P-R-O-C-H-Y, 2024-10-31:
> Hi all,
> 
> There is a temporary API available for setting the MAC Capability Flag. You can use `extern void zb_set_ed_node_descriptor(bool power_src, bool rx_on_when_idle, bool alloc_addr);`. Then, call it before `esp_zb_start();`.

@xieqinan Do I need to include any specific file for that? I am getting undefined reference.

mdudek, 2024-11-01:
> > Hi all,
> > There is a temporary API available for setting the MAC Capability Flag. You can use `extern void zb_set_ed_node_descriptor(bool power_src, bool rx_on_when_idle, bool alloc_addr);`. Then, call it before `esp_zb_start();`.
> 
> @xieqinan Do I need to include any specific file for that? I am getting undefined reference.
> 
> EDIT: In Arduino its cpp code, when declared the function in `extern "C"` it works.

What did you do to make it working?  I got "error: 'zb_set_ed_node_descriptor' was not declared in this scope".

P-R-O-C-H-Y, 2024-11-01:
> What did you do to make it working? I got "error: 'zb_set_ed_node_descriptor' was not declared in this scope".

Hi @mdudek, I just added this line at the start of the file:
`extern "C" void zb_set_ed_node_descriptor(bool power_src, bool rx_on_when_idle, bool alloc_addr);`

and then just used the function in the code. Be aware it needs to be called **AFTER** Zigbee start. Otherwise it won't work.
This is the task which is running Zigbee for me:
```
static void esp_zb_task(void *pvParameters) {
  /* initialize Zigbee stack */
  ESP_ERROR_CHECK(esp_zb_start(false));

  //NOTE: This is a workaround to make battery powered devices to be discovered as battery powered
  if(((zigbee_role_t)Zigbee.getRole() == ZIGBEE_END_DEVICE) && edBatteryPowered ) {
    zb_set_ed_node_descriptor(0,0,0);
  }

  esp_zb_stack_main_loop();
}
```

theredone-1, 2024-11-30:
I've literally spent a week and a half trying to crack this issue, until I stumbled on this thread. Thank you so much, guys. I mean it.
Having said that, an addition to the SDK would be nice.