espressif/esp-zigbee-sdk #79: How to get rssi?

Tropaion, 2023-08-23:
Hello,
I'm not sure how it works, but is there a way to get the RSSI from my esp router/end device?
I created an router with an temperature and humidity sensor, from which I successfully get the LQI value but not RSSI, which my company would need for an field test.
Or maybe do I need the RSSI Location Cluster, which is not implemeted (yet), for this?

xieqinan, 2023-08-24:
Hello, 
Are you interested in obtaining the RSSI of an ESP Zigbee device?

Could you kindly share the method you employed to retrieve the LQI?

As of the current version, the RSSI API is not supported. However, there is a workaround available. You can utilize `esp_ieee802154_get_recent_rssi()` to access the RSSI value of the most recently received frame. It's important to note that this API only provides the RSSI for the latest received frame.

A more refined approach for obtaining RSSI will be introduced in an upcoming version.

Tropaion, 2023-08-24:
Hello @xieqinan,
I didn't directly retrieve it, Home Assistant shows the LQI of the device but not RSSI.

xieqinan, 2023-08-24:
OK, Let me know which kind way you expected to get  getting RSSI value.

Tropaion, 2023-08-24:
I want it to be displayed here.
[image]
HomeAssistant successfully displays LQI from the device but always says `Unknown` for RSSI.

BogatovT, 2023-08-25:
Yeah, it will be good to get RSSI in whole callback functions.

chshu, 2023-10-12:
@Tropaion @BogatovT Adding RSSI to all the callbacks doesn't help with this issue.

There should be some tools used to connect the Zigbee device and HomeAssistant APP, could you check where does the HomeAssistant read the RSSI from? Actually I also wondering whether the LQI value is eported by Zigbee device or not.

xieqinan, 2023-10-31:
@Tropaion,

> I want it to be displayed here. [image] HomeAssistant successfully displays LQI from the device but always says `Unknown` for RSSI.

Regarding this issue, I believe there are two suggestions.

Firstly, please ensure that the ZCL Identify cluster has been created in your endpoint, as the 'diagnose' button is meant for sending identify commands.

Secondly, please ensure the RSSI function is visible as follows:
[image]

As a result, the `Diagnose` control panel will be displayed as follows:
[image]

Tropaion, 2023-10-31:
@xieqinan Thank you. Did you test this? Do you have a code example? Because I did the basic implementation of the identify cluster, but I didn't get LQI and RSSI. Maybe there is some mistake in my [implementation](https://github.com/Tropaion/ZigBee_SmartMeter_Reader/blob/main/software/zigbee/components/zigbee_electricity_meter/src/zb_electricity_meter.c#L195)?

xieqinan, 2023-11-01:
Yes, I have already tested it. You just need to use the [HA_ON_OFF_LIGHT](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_light) as the end device to join Home Assistant, then diagnose it.
Please note to click the RSSI icon to make it visible.

Tropaion, 2023-11-01:
Ok, then I will try it again.

chshu, 2023-11-19:
@Tropaion Please reopen if you have any follow up question on this topic.

closed 2023-11-19 by chshu